Genesis-Embodied-AI/genesis-world #727: [Bug]: Strange performance when object gets pushed and slides on the surface

ZhuoyunZhong, 2025-02-13:
### Bug Description

My original goal was to use the robot to push an object on the table. But the object keeps jumping and wobbling strangely instead of sliding smoothly on the table (See videos). 

I guess it seems like the object will somehow go "inside" the table. I remember I saw this kind of strange behavior in other simulations too, but not this severe.

I quickly put up a simple script that could reproduce the problem.

### Steps to Reproduce

```python
import genesis as gs

# Init
gs.init(backend=gs.gpu)
scene = gs.Scene(show_viewer=True, show_FPS=False)

# Support
use_plane = False
if not use_plane:
    table = scene.add_entity(
        gs.morphs.Box(pos=(0.0, 0, -0.25), size=(5, 5, 0.5), fixed=True)
    )
else:
    plane = scene.add_entity(gs.morphs.Plane())
# Push objects
obj = scene.add_entity(
    gs.morphs.Box(pos=(0.0, 0, 0.05), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(1, 0, 0)),
)
obj2 = scene.add_entity(
    gs.morphs.Box(pos=(-0.4, 0, 0.05), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(1, 0, 0)),
)
scene.build()

# Execute
input("Start?")
for i in range(2000):
    if i < 300:
        obj2.set_dofs_velocity([0.6, 0, 0, 0, 0, 0])
    else:
        obj2.set_dofs_velocity([0, 0, 0, 0, 0, 0])
    scene.step()

```

### Expected Behavior

The objects won't slide smoothly but keep jumping and wobbling strangely.

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 22.04]
- GPU/CPU [RTX 4070 Ti Super]
- GPU-driver version (550.127.08)
- CUDA / CUDA-toolkit version (12.4)

### Release version or Commit ID

PIP genesis-world, 0.2.1

YilingQiao, 2025-02-14:
It's a bit unphysical and makes the simulation struggle when directly overwriting the velocity.

You can try this; it runs much smoother on my side.

```
obj2.control_dofs_velocity([0.6, 0, 0, 0, 0, 0])
```

Let me know if that’s still unsatisfactory. I think in the real world, things can also be shaky like that (imagine dragging a chair across the floor).

ZhuoyunZhong, 2025-02-14:
Thanks for the reply!

Well, using the obj2 to push obj is just to simulate using the robot end-effector to push obj. My key concern is still the movement of the pushed obj. Following your recommendation, l tried the following code
```python
import genesis as gs

# Init
gs.init(backend=gs.gpu)
scene = gs.Scene(
    show_viewer=True,
    show_FPS=False,
    rigid_options=gs.options.RigidOptions(dt=0.01),
)

# Support
use_plane = False
if not use_plane:
    table = scene.add_entity(
        gs.morphs.Box(pos=(0.0, 0, -0.25), size=(5, 5, 0.5), fixed=True)
    )
else:
    plane = scene.add_entity(gs.morphs.Plane())
# Push objects
obj = scene.add_entity(
    gs.morphs.Box(pos=(0, 0.02, 0.05), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(1, 0, 0)),
)
obj2 = scene.add_entity(
    gs.morphs.Box(pos=(-0.4, 0, 0.05), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(1, 0, 0)),
)
scene.build()

# Execute
input("Start?")
for i in range(2000):
    if i < 300:
        obj2.control_dofs_velocity([0.8, 0, 0, 0, 0, 0])
    else:
        obj2.control_dofs_velocity([0, 0, 0, 0, 0, 0])
    scene.step()
```

As you will see, even though the obj2 movement is smooth, the obj will still "go inside" the table and produce very undesired movements.

https://github.com/user-attachments/assets/166f0cd4-0828-4c0e-8ae5-0ab16dcfe85e

Kashu7100, 2025-02-14:
I think this is more like the friction between the table and block. You can try to reduce the friction to make it slide smoother.

ZhuoyunZhong, 2025-02-14:
Thanks for the suggestions. Here is the updated script that I tried

1, First I remove the gravity of blue `obj2` and make it float, then hit the red `obj`, to better simulate the situation of a robot end-effector pushing an object. (I also changed the color just to make it distinguishable.)
2, I reduced the friction of the `table` and `obj2` and tried different values

```python
import genesis as gs

# Init
gs.init(backend=gs.gpu)
scene = gs.Scene(
    show_viewer=True,
    show_FPS=False,
    rigid_options=gs.options.RigidOptions(dt=0.01),
)

# Support
friction = 0.1
table = scene.add_entity(
    gs.morphs.Box(pos=(0.0, 0, -0.25), size=(5, 5, 0.5), fixed=True),
    material=gs.materials.Rigid(friction=friction),
)
# Push objects
obj = scene.add_entity(
    gs.morphs.Box(pos=(0, 0.02, 0.05), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(1, 0, 0)),
    material=gs.materials.Rigid(friction=friction),
)
obj2 = scene.add_entity(
    gs.morphs.Box(pos=(-0.4, 0, 0.06), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(0, 0, 1)),
    material=gs.materials.Rigid(gravity_compensation=1.0),
)
scene.build()

# Execute
input("Start?")
for i in range(2000):
    if i < 300:
        obj2.control_dofs_velocity([0.8, 0, 0, 0, 0, 0])
    else:
        obj2.control_dofs_velocity([0, 0, 0, 0, 0, 0])
    scene.step()

```

friction = 1.0

https://github.com/user-attachments/assets/0e7ce5d3-ef16-4b6b-a812-59af084c4da5

friction = 0.5

https://github.com/user-attachments/assets/6ca64b21-1c47-4f94-94ae-89b3da00efb3

friction = 0.1

https://github.com/user-attachments/assets/211d05c7-0a44-42f5-9b4b-f41f65dff047

Friction lower than 0.1 will make things too unrealistic (like sliding on the ice). Any thoughts?

YilingQiao, 2025-02-20:
It will be jerky with high friction and fast speed. Maybe you can try to increase the simulation precision and it will help a bit
```
scene = gs.Scene(
    show_viewer=True,
    show_FPS=False,
    sim_options=gs.options.SimOptions(
        dt=0.01,
        substeps=8,
    ),
    rigid_options=gs.options.RigidOptions(
        constraint_solver=gs.constraint_solver.Newton,
    ),
)
```

ZhuoyunZhong, 2025-02-25:
Thank you for the reply!

Yes! So the thing actually helps is the `substeps`. Having it as 8 mitigates the weird collision problem and makes the sliding much smoother. Here is the latest code I have for whoever has this problem in the future.

```python
import genesis as gs

# Init
gs.init(backend=gs.gpu)
scene = gs.Scene(
    show_viewer=True,
    show_FPS=False,
    sim_options=gs.options.SimOptions(substeps=8),
)

# Support
friction = 0.2
table = scene.add_entity(
    gs.morphs.Box(pos=(0.0, 0, -0.25), size=(5, 5, 0.5), fixed=True),
    material=gs.materials.Rigid(friction=friction),
)
# Push objects
obj = scene.add_entity(
    gs.morphs.Box(pos=(0, 0.02, 0.05), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(1, 0, 0)),
    material=gs.materials.Rigid(friction=friction),
)
obj2 = scene.add_entity(
    gs.morphs.Box(pos=(-0.4, 0, 0.06), size=(0.1, 0.1, 0.1)),
    surface=gs.surfaces.Default(color=(0, 0, 1)),
    material=gs.materials.Rigid(gravity_compensation=1.0),
)
scene.build()

# Execute
input("Start?")
for i in range(1000):
    if i < 200:
        obj2.control_dofs_velocity([0.8, 0, 0, 0, 0, 0])
    else:
        obj2.control_dofs_velocity([0, 0, 0, 0, 0, 0])
    scene.step()
```

I will close this issue now.

https://github.com/user-attachments/assets/9e968b62-1694-478b-9f6b-2488987fd4eb

closed 2025-02-25 by ZhuoyunZhong