Genesis-Embodied-AI/genesis-world #733: [Feature]: Hope Genesis will support joint angles

BoruiHe, 2025-02-14:
### What feature or enhancement are you proposing?

The angle (either in radians or degrees) of each joint. I did not find any method relating to joint angles. Please correct me if I am wrong.

### Motivation

As a researcher working on robotic vision,
I want to further explore how proprioception cooperate with other sensibilities like vision,
Because a better model should take data from proprioception and perception for prediction potential risks.

### Potential Benefit

In robotics, position control refers to commanding robots to move to the target position. Relative positions, world coordinates and joint angles can all represent a specific position. Joint angles are straightforward and easy to read when we are dealing with motion constraints relating to rotation on joints.

### What is the expected outcome of the implementation work?

It should be a method of loaded entity that returns a tensor/array of the all joint angles. The length of the tensor/array should equal the number of joints.

Kashu7100, 2025-02-14:
Do you mean `dof_*`??

BoruiHe, 2025-02-15:
> Do you mean `dof_*`??

I assume you mean the dof (degrees of freedom) angles. That would be more accurate. A link should offer at least 1 dof on rotation. Then, a simulator can calculate angles on dofs. 

I also checked the `rigid_entity.py` and the closest method would be `get_dofs_position`, which returns positions. If so, a method that returns `dof angles` would be what I am talking about.

Correct me if misunderstood what you mean or miss anything.

Kashu7100, 2025-02-15:
Yes, you can use `get_dofs_position` to obtain the joint angle.

BoruiHe, 2025-02-15:
Thank you for your reply.

I have some questions:

1. I found a joint that has 6 dofs (dofs_index from 0 to 5), which has never been seen in the previous simulator. I would like to record these values in case I may need them one day. I wonder what order are these 6 dofs following? (The meaning of this these 6 dofs) I guess [x,y,z,raw,pitch,yaw]. Am I correct?
2. All angles fall in the range of [0, 2*pi), i.e., in radians, due to the periodicity. Right?
3. Is that possible for Genesis to load fixed joints? (I am using URDF files.)
I have a fake joint, head_cam, for my robot, and would like to mount a follower camera at this position. However, Genesis does not load any fixed joints.
`</joint>
	<joint name="head_cam" type="fixed">
    		<origin xyz="0 .075 .025" rpy="0 0 0"></origin>
		<parent link="head"></parent>
		<child link="camera"></child>
		<axis xyz="1 0 0"></axis>
</joint>`

lupinjia, 2025-02-19:
> Thank you for your reply.
>
> I have some questions:
>
> 1. I found a joint that has 6 dofs (dofs_index from 0 to 5), which has never been seen in the previous simulator. I would like to record these [...]

1. not sure what's the meaning of "what order are these 6 dofs following", [x,y,z,**roll**,pitch,yaw] should be pose info, `get_dofs_position` api can return all the angles of the joints, in the order of joints loaded into genesis.

3. to keep fixed joints, just specify links_to_keep when loading URDF. Please refer to https://genesis-world.readthedocs.io/en/latest/api_reference/options/morph/file_morph/urdf.html

closed 2025-02-19 by BoruiHe

BoruiHe, 2025-02-19:
Thank you. I can refer to this API reference for details.