Genesis-Embodied-AI/genesis-world #740: [Bug]: Recent update not compatible with Mac

mi-robotics, 2025-02-17:
### Bug Description

M3 

error:

dlsym(0x3e624c0b0, glDrawElementsInstancedBaseInstance): symbol not found

glDrawElementsInstancedBaseInstance is supported with OpenGL min version 4.2: https://registry.khronos.org/OpenGL-Refpages/gl4/html/glDrawElementsInstancedBaseInstance.xhtml

Apple silicon only supports up to OpenGL 4.1

### Steps to Reproduce

python ./examples/render_on_macos.py

### Expected Behavior

AttributeError: dlsym(0x3e624c0b0, glDrawElementsInstancedBaseInstance): symbol not found

### Relevant log output

```shell

```

### Environment

[Genesis] [11:22:29] [INFO] Running on [Apple M3 Pro] with backend gs.cpu. Device memory: 36.00 GB.
[Genesis] [11:22:29] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.

### Release version or Commit ID

bf3bc101630bea36b1835f9d9cbbbf8e65596760

hadamove, 2025-02-23:
Looks like it has been caused by https://github.com/Genesis-Embodied-AI/Genesis/pull/545

duburcqa, 2025-02-24:
I can confirm that it is related to #545. OpenGL on MacOS is limited to version 4.1 while `glDrawElementsInstancedBaseInstance` has been introduced with version 4.2.

duburcqa, 2025-02-24:
@mi-robotics Did it work for you before the update ? Based on my experiment, it is not working either, because `glGenTextures` behaves differently on MacOS.

mi-robotics, 2025-02-24:
If you need a quick fix:
```
class GLWrapper:
    def __init__(self):
        self.gl_funcs = {}

        load_func = self.load_func
        load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
        load_func("glUniformMatrix4fv", GLvoid, GLint, GLsizei, GLboolean, GLvoidp)
        load_func("glUniform1i", GLvoid, GLint, GLint)
        load_func("glUniform1f", GLvoid, GLint, GLfloat)
        load_func("glUniform2f", GLvoid, GLint, GLfloat, GLfloat)
        load_func("glUniform3fv", GLvoid, GLint, GLsizei, GLvoidp)
        load_func("glUniform4fv", GLvoid, GLint, GLsizei, GLvoidp)
        load_func("glBindVertexArray", GLvoid, GLuint)
        load_func("glActiveTexture", GLvoid, GLenum)
        load_func("glBindTexture", GLvoid, GLenum, GLuint)
        load_func("glEnable", GLvoid, GLenum)
        load_func("glDisable", GLvoid, GLenum)
        load_func("glBlendFunc", GLvoid, GLenum, GLenum)
        load_func("glPolygonMode", GLvoid, GLenum, GLenum)
        load_func("glCullFace", GLvoid, GLenum)
        load_func("glDrawElementsInstanced", GLvoid, GLenum, GLsizei, GLenum, GLvoidp, GLsizei)
        load_func("glDrawArraysInstanced", GLvoid, GLenum, GLint, GLsizei, GLsizei)
        if gs.platform != 'macOS':
            load_func("glDrawElementsInstancedBaseInstance", GLvoid, GLenum, GLsizei, GLenum, GLvoidp, GLsizei, GLuint)
            load_func("glDrawArraysInstancedBaseInstance", GLvoid, GLenum, GLint, GLsizei, GLsizei, GLuint)
        load_func("glUseProgram", GLvoid, GLuint)
        load_func("glFlush", GLvoid)
        load_func("glReadPixels", GLvoid, GLint, GLint, GLsizei, GLsizei, GLenum, GLenum, GLvoidp)
        load_func("glBindBuffer", GLvoid, GLenum, GLuint)
        load_func("glBufferData", GLvoid, GLenum, GLsizeiptr, GLvoidp, GLenum)
        load_func("glBufferSubData", GLvoid, GLenum, GLintptr, GLsizeiptr, GLvoidp)

        self.build_wrapper()
```

Inside the forward_pass, shadow_mapping_pass, point_shadow_mapping_pass functions of JitRender make this change:
```
# if render_flags[id, 6] or env_idx == -1:
              

  if n_indices[id] > 0:
                    gl.glDrawElementsInstanced(
                        mode[id], n_indices[id], GL_UNSIGNED_INT, address_to_ptr(0), n_instances[id]
                    )
                else:
                    gl.glDrawArraysInstanced(mode[id], 0, -n_indices[id], n_instances[id])
              
                # if n_indices[id] > 0:
                #     gl.glDrawElementsInstancedBaseInstance(
                #         mode[id], n_indices[id], GL_UNSIGNED_INT, address_to_ptr(0), 1, env_idx
                #     )
                # else:
                #     gl.glDrawA

rraysInstancedBaseInstance(mode[id], 0, -n_indices[id], 1, env_idx)
```

duburcqa, 2025-02-24:
Yes, I'm working on a more maintainable fix. I worked around the issues you are addressing, but it is not sufficient to make it work on MacBook Pro M3. I'm running into the following issue that I'm retrying to address right now:
```python
File ~/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/ext/pyrender/texture.py:180, in Texture._add_to_context(self)
    177     fmt = GL_RGBA
    179 # Generate the OpenGL texture
--> 180 self._texid = glGenTextures(1)
    181 glBindTexture(self.tex_type, self._texid)
    183 # Flip data for OpenGL buffer

[...]

ArgumentError: ("argument 2: TypeError: No array-type handler for type _ctypes.type (value: <cparam 'P' (0x3de005418)>) registered", (1, <cparam 'P' (0x3de005418)>))
```

mi-robotics, 2025-02-24:
I see, I did not encounter this issue, also on M3

duburcqa, 2025-02-24:
Interesting... Which version of `pyopengl` are you using ? 

What is the output of this snippet on your side ?
```python
from OpenGL.GL import *
print(glGenTextures(1))
```

duburcqa, 2025-02-24:
Ok I think I fixed the various issues, it should be fine now !

closed 2025-02-25 by YilingQiao