Genesis-Embodied-AI/genesis-world #748: [Bug]: ompl installation problem

milong26, 2025-02-18:
### Bug Description

When I follow doc to install ompl, I met this problem:
```
$ pip install ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl 

ERROR: ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.
```

python and os are correct

### Steps to Reproduce

1. wget https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl
2. pip install ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl 

then ERROR: ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.

### Expected Behavior

I can install OMPL for motion planning

### Relevant log output

```shell
I searched in OMPL's repo one coders raised same problem the owner of repo said(vert long)

<details>

WARNING: This command is only meant for debugging. Do not use this with automation for parsing and getting these details, since the output and options of this command may change without notice.

pip version: pip 25.0 from ../genesis/lib/python3.10/site-packages/pip (python 3.10)

sys.version: 3.10.0 (default, Mar  3 2022, 09:58:08) [GCC 7.5.0]
sys.executable: /data/zly/application/anaconda3/envs/genesis/bin/python
sys.getdefaultencoding: utf-8
sys.getfilesystemencoding: utf-8
locale.getpreferredencoding: UTF-8
sys.platform: linux
sys.implementation:
  name: cpython
'cert' config value: Not specified
REQUESTS_CA_BUNDLE: None
CURL_CA_BUNDLE: None
pip._vendor.certifi.where(): /data/zly/application/anaconda3/envs/genesis/lib/python3.10/site-packages/pip/_vendor/certifi/cacert.pem
pip._vendor.DEBUNDLED: False
vendored library versions:
  CacheControl==0.14.1
  distlib==0.3.9
  distro==1.9.0
  msgpack==1.1.0
  packaging==24.2
  platformdirs==4.3.6
  pyproject-hooks==1.2.0
  requests==2.32.3
  certifi==2024.08.30
  idna==3.10
  urllib3==1.26.20
  rich==13.9.4 (Unable to locate actual module version, using vendor.txt specified version)
  pygments==2.18.0
  typing_extensions==4.12.2 (Unable to locate actual module version, using vendor.txt specified version)
  resolvelib==1.0.1
  setuptools==70.3.0 (Unable to locate actual module version, using vendor.txt specified version)
  tomli==2.2.1
  truststore==0.10.0
Compatible tags: 634
  cp310-cp310-manylinux_2_27_x86_64
  cp310-cp310-manylinux_2_26_x86_64
  cp310-cp310-manylinux_2_25_x86_64
  cp310-cp310-manylinux_2_24_x86_64
[... 616 lines cut ...]
  py30-none-linux_x86_64
  cp310-none-any
  py310-none-any
  py3-none-any
  py39-none-any
  py38-none-any
  py37-none-any
  py36-none-any
  py35-none-any
  py34-none-any
  py33-none-any
  py32-none-any
  py31-none-any
  py30-none-any
</details>
```

### Environment

- python vesion=3.10.0.
- OS=Linux hmc35 4.15.0-208-generic # 220-Ubuntu SMP Mon Mar 20 14:27:01 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Release version or Commit ID

Although I don't think it's about version but `Commit abcc40f`

bxtbold, 2025-02-19:
Which Ubuntu are you using? It seems like you're using 18.04 based on from the kernel you're using.
If that's the case, try testing on newer releases like 22.04 or 24.04.

milong26, 2025-02-19:
> Which Ubuntu are you using? It seems like you're using 18.04 based on from the kernel you're using. If that's the case, try testing on newer releases like 22.04 or 24.04.

I got answer from ompl repo, said OMPL wheels are built only for glibc >=2.28 while mine glibc=2.27. So I will try build myself :) thanks!

closed 2025-02-19 by milong26