Genesis-Embodied-AI/genesis-world #749: [Bug]: IK example error on Intel MacOS

trubby22, 2025-02-18:
### Bug Description

When running the [🦾 Inverse Kinematics & Motion Planning example from the user guide](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/inverse_kinematics_motion_planning.html), I get `std::bad_cast` thrown by the solver.

### Steps to Reproduce

main.py:
```
import sys
import genesis as gs
import numpy as np
from tqdm import tqdm

def main():
    gs.init(backend=gs.cpu)
    scene = gs.Scene(
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(3, -1, 1.5),
            camera_lookat=(0.0, 0.0, 0.5),
            camera_fov=30,
            max_FPS=60,
        ),
        sim_options=gs.options.SimOptions(
            dt=0.01,
        ),
        show_viewer=True,
        show_FPS=False,
    )
    plane = scene.add_entity(gs.morphs.Plane())
    cube = scene.add_entity(
        gs.morphs.Box(
            size=(0.04, 0.04, 0.04),
            pos=(0.65, 0.0, 0.02),
        )
    )
    franka = scene.add_entity(
        gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
    )
    scene.build()
    if sys.platform == "darwin":
        scene._visualizer._viewer._pyrender_viewer._renderer.dpscale = 1
    gs.tools.run_in_another_thread(fn=run_sim, args=(scene, franka))
    scene.viewer.start()

def run_sim(scene, franka):
    motors_dof = np.arange(7)
    fingers_dof = np.arange(7, 9)
    franka.set_dofs_kp(
        np.array([4500, 4500, 3500, 3500, 2000, 2000, 2000, 100, 100]),
    )
    franka.set_dofs_kv(
        np.array([450, 450, 350, 350, 200, 200, 200, 10, 10]),
    )
    franka.set_dofs_force_range(
        np.array([-87, -87, -87, -87, -12, -12, -12, -100, -100]),
        np.array([87, 87, 87, 87, 12, 12, 12, 100, 100]),
    )
    end_effector = franka.get_link("hand")
    # Move to pre-grasp pose
    qpos_pre_grasp = franka.inverse_kinematics(
        link=end_effector,
        pos=np.array([0.65, 0.0, 0.25]),
        quat=np.array([0, 1, 0, 0]),
    )
    qpos_pre_grasp[-2:] = 0.04
    print(type(qpos_pre_grasp))
    print(qpos_pre_grasp.shape)
    print(qpos_pre_grasp)
    path = franka.plan_path(
        qpos_goal=qpos_pre_grasp,
[... 194 lines cut ...]
tensorboard @ file:///home/conda/feedstock_root/build_artifacts/tensorboard_1733820672324/work/tensorboard-2.18.0-py3-none-any.whl#sha256=107ca4821745f73e2aefa02c50ff70a9b694f39f790b11e6f682f7d326745eab
tensorboard_data_server @ file:///Users/runner/miniforge3/conda-bld/tensorboard-data-server_1728639744148/work/tensorboard_data_server-0.7.0-py3-none-macosx_10_9_x86_64.whl#sha256=5900110b293d873154e0dd7868dc4ec2d0a3b669f3ef4570482e562fc99edcb8
tetgen==0.6.4
tifffile==2025.1.10
torch==2.2.2
torchaudio==2.2.2
torchvision==0.17.2
tornado @ file:///Users/runner/miniforge3/conda-bld/tornado_1732615926389/work
tqdm==4.67.1
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1733367359838/work
typing-inspect==0.9.0
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1733188668063/work
urllib3==2.3.0
vtk==9.3.1
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1733231326287/work
Werkzeug @ file:///home/conda/feedstock_root/build_artifacts/werkzeug_1733160440960/work
wrapt==1.17.2
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1732827521216/work
```

`pip install -r requirements.txt`

### Release version or Commit ID

release version: 0.2.0

wangyian-me, 2025-03-20:
Seems that you are using `ompl-1.6.0-cp311-cp311-macosx_14_0_x86_64.whl`, but your system is macOS 15.
Can you try to use the [ompl-1.6.0-cp311-cp311-macosx_15_0_x86_64.whl](https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp311-cp311-macosx_15_0_x86_64.whl)?

trubby22, 2025-03-21:
Thanks for suggesting how to fix the bug!

I have now switched from MacOS to Linux given that Genesis offers better support for the latter. Hence, I no longer require assistance in this issue and won't be testing any potential bug fixes on my MacOS machine.

closed 2025-03-21 by trubby22

duburcqa, 2025-03-22:
Not sure if we can do anything about this issue because reproducing it would be difficult. It is definitely related to ompl that is not working properly on your machine for some reason.