Genesis-Embodied-AI/genesis-world #750: [Bug]: RL locomotion example error on Intel MacOS

trubby22, 2025-02-18:
### Bug Description

When running the [🦿 Training Locomotion Policies with RL example from the user guide](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/locomotion.html) (which corresponds to the following folder: [examples/locomotion](https://github.com/trubby22/genesis-fork/tree/main/examples/locomotion)), I get "ValueError: too many values to unpack (expected 2)" thrown by the rsl_rl library. I modified the code with respect to the original example slightly to enable running on my MacOS machine (use run_sim function, run on CPU).

### Steps to Reproduce

`pip install rsl-rl-lib`
`python examples/locomotion/go2_train.py`

### Expected Behavior

I would expect no errors when running this example.

### Relevant log output

```shell
Traceback (most recent call last):      

 File "/Users/foo/Documents/Genesis/examples/locomotion/go2_train.py", line 170, in <module>

  main()        

 File "/Users/foo/Documents/Genesis/examples/locomotion/go2_train.py", line 159, in main

  runner = OnPolicyRunner(env, train_cfg, log_dir, device="cpu")

       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

 File "/opt/miniconda3/envs/genesis-sim/lib/python3.11/site-packages/rsl_rl/runners/on_policy_runner.py", line 32, in __init__

  obs, extras = self.env.get_observations()                              

  ^^^^^^^^^^^

ValueError: too many values to unpack (expected 2)
```

### Environment

- OS: [e.g. MacOS 15.1.1 (24B91)]
- CPU [2.3 GHz 8-Core Intel Core i9]
- GPU-driver version: N/A
- CUDA / CUDA-toolkit version: N/A

requirements.txt:
```
absl-py @ file:///home/conda/feedstock_root/build_artifacts/absl-py_1733730548347/work
annotated-types==0.7.0
appnope @ file:///home/conda/feedstock_root/build_artifacts/appnope_1733332318622/work
asttokens @ file:///home/conda/feedstock_root/build_artifacts/asttokens_1733250440834/work
certifi @ file:///home/conda/feedstock_root/build_artifacts/certifi_1739515848642/work/certifi
charset-normalizer==3.4.1
coacd==1.0.5
colorama==0.4.6
comm @ file:///home/conda/feedstock_root/build_artifacts/comm_1733502965406/work
contourpy==1.3.1
cycler==0.12.1
Cython==3.0.12
dataclasses-json==0.6.7
debugpy @ file:///Users/runner/miniforge3/conda-bld/debugpy_1737269781257/work
decorator @ file:///home/conda/feedstock_root/build_artifacts/decorator_1733236420667/work
Deprecated==1.2.18
dill==0.3.9
etils==1.12.0
exceptiongroup @ file:///home/conda/feedstock_root/build_artifacts/exceptiongroup_1733208806608/work
executing @ file:///home/conda/feedstock_root/build_artifacts/executing_1733569351617/work
filelock==3.17.0
fonttools==4.56.0
freetype-py==2.5.1
fsspec==2025.2.0
genesis-world==0.2.0
gitdb==4.0.12
GitPython==3.1.44
glfw==2.8.0
grpcio @ file:///Users/runner/miniforge3/conda-bld/grpc-split_1735582802685/work
idna==3.10
imageio==2.37.0
imageio-ffmpeg==0.6.0
importlib_metadata @ file:///home/conda/feedstock_root/build_artifacts/importlib-metadata_1737420181517/work
importlib_resources==6.5.2
ipykernel @ file:///Users/runner/miniforge3/conda-bld/ipykernel_1719845458456/work
ipython @ file:///home/conda/feedstock_root/build_artifacts/bld/rattler-build_ipython_1738421264/work
jedi @ file:///home/conda/feedstock_root/build_artifacts/jedi_1733300866624/work
Jinja2==3.1.5
jupyter_client @ file:///home/conda/feedstock_root/build_artifacts/jupyter_client_1733440914442/work
jupyter_core @ file:///home/conda/feedstock_root/build_artifacts/jupyter_core_1727163409502/work
kiwisolver==1.4.8
lazy_loader==0.4
libigl==2.5.1
llvmlite==0.44.0
lxml==5.3.1
Markdown @ file:///home/conda/feedstock_root/build_artifacts/markdown_1710435156458/work
markdown-it-py==3.0.0
MarkupSafe @ file:///Users/runner/miniforge3/conda-bld/markupsafe_1733219683650/work
marshmallow==3.26.1
matplotlib==3.10.0
matplotlib-inline @ file:///home/conda/feedstock_root/build_artifacts/matplotlib-inline_1733416936468/work
mdurl==0.1.2
moviepy==2.1.2
mpmath==1.3.0
mujoco==3.2.5
mypy-extensions==1.0.0
narwhals==1.26.0
nest_asyncio @ file:///home/conda/feedstock_root/build_artifacts/nest-asyncio_1733325553580/work
networkx==3.4.2
numba==0.61.0
numpy==1.26.4
ompl @ file:///Users/piotrblaszyk/Downloads/ompl-1.6.0-cp311-cp311-macosx_14_0_x86_64.whl#sha256=ecc71dd889b96c5499f26fc08760bf202b322fc6a401c4b77355b86b3a566ecd
onnx==1.17.0
opencv-python==4.11.0.86
OpenEXR==3.3.2
packaging @ file:///home/conda/feedstock_root/build_artifacts/packaging_1733203243479/work
parso @ file:///home/conda/feedstock_root/build_artifacts/parso_1733271261340/work
pexpect @ file:///home/conda/feedstock_root/build_artifacts/pexpect_1733301927746/work
pickleshare @ file:///home/conda/feedstock_root/build_artifacts/pickleshare_1733327343728/work
pillow==10.4.0
platformdirs @ file:///home/conda/feedstock_root/build_artifacts/platformdirs_1733232627818/work
plotly==6.0.0
pooch==1.8.2
proglog==0.1.10
prompt_toolkit @ file:///home/conda/feedstock_root/build_artifacts/prompt-toolkit_1737453357274/work
protobuf @ file:///Users/runner/miniforge3/conda-bld/protobuf_1731365486807/work/bazel-bin/python/dist/protobuf-5.28.3-cp311-abi3-macosx_10_13_x86_64.whl#sha256=da151a2c36d396a7df43a0a344dcfec3d7a99efd2e0f36942b8d81f78e837710
psutil @ file:///Users/runner/miniforge3/conda-bld/psutil_1735327364162/work
ptyprocess @ file:///home/conda/feedstock_root/build_artifacts/ptyprocess_1733302279685/work/dist/ptyprocess-0.7.0-py2.py3-none-any.whl#sha256=92c32ff62b5fd8cf325bec5ab90d7be3d2a8ca8c8a3813ff487a8d2002630d1f
pure_eval @ file:///home/conda/feedstock_root/build_artifacts/pure_eval_1733569405015/work
pycollada==0.8
pydantic==2.7.1
pydantic_core==2.18.2
PyGEL3D==0.5.2
pyglet==2.1.2
pygltflib==1.16.0
Pygments @ file:///home/conda/feedstock_root/build_artifacts/pygments_1736243443484/work
pymeshlab==2023.12.post2
pyobjc-core==11.0
pyobjc-framework-Cocoa==11.0
PyOpenGL==3.1.9
pyparsing==3.2.1
python-dateutil @ file:///home/conda/feedstock_root/build_artifacts/python-dateutil_1733215673016/work
python-dotenv==1.0.1
pyvista==0.44.2
pyzmq @ file:///Users/runner/miniforge3/conda-bld/pyzmq_1738271005832/work
requests==2.32.3
rich==13.9.4
rsl-rl-lib==2.2.1
scikit-image==0.25.1
scipy==1.15.1
scooby==0.10.0
screeninfo==0.8.1
six @ file:///home/conda/feedstock_root/build_artifacts/six_1733380938961/work
smmap==5.0.2
stack_data @ file:///home/conda/feedstock_root/build_artifacts/stack_data_1733569443808/work
sympy==1.13.3
taichi==1.7.2
tensorboard @ file:///home/conda/feedstock_root/build_artifacts/tensorboard_1733820672324/work/tensorboard-2.18.0-py3-none-any.whl#sha256=107ca4821745f73e2aefa02c50ff70a9b694f39f790b11e6f682f7d326745eab
tensorboard_data_server @ file:///Users/runner/miniforge3/conda-bld/tensorboard-data-server_1728639744148/work/tensorboard_data_server-0.7.0-py3-none-macosx_10_9_x86_64.whl#sha256=5900110b293d873154e0dd7868dc4ec2d0a3b669f3ef4570482e562fc99edcb8
tetgen==0.6.4
tifffile==2025.1.10
torch==2.2.2
torchaudio==2.2.2
torchvision==0.17.2
tornado @ file:///Users/runner/miniforge3/conda-bld/tornado_1732615926389/work
tqdm==4.67.1
traitlets @ file:///home/conda/feedstock_root/build_artifacts/traitlets_1733367359838/work
typing-inspect==0.9.0
typing_extensions @ file:///home/conda/feedstock_root/build_artifacts/typing_extensions_1733188668063/work
urllib3==2.3.0
vtk==9.3.1
wcwidth @ file:///home/conda/feedstock_root/build_artifacts/wcwidth_1733231326287/work
Werkzeug @ file:///home/conda/feedstock_root/build_artifacts/werkzeug_1733160440960/work
wrapt==1.17.2
zipp @ file:///home/conda/feedstock_root/build_artifacts/zipp_1732827521216/work
```

`pip install -r requirements.txt`

### Release version or Commit ID

release version: 0.2.0

Kashu7100, 2025-02-18:
I think you need to use the older version of RSL RL (https://github.com/leggedrobotics/rsl_rl/releases/tag/v1.0.2)

trubby22, 2025-02-19:
I tried with the older version of RSL RL (v 1.0.2) but I'm getting the same error. I updated the bug description above to reflect this.

stillonearth, 2025-02-19:
@trubby2 it seems the tagged version uses new gym environment signature. The correct commit of rsl_rl is 8aef3f3e9422bfa89a404f21150c54b4067cccd7

trubby22, 2025-02-20:
This fixed my issue. Thanks!

closed 2025-02-20 by trubby22

Pandakingxbc, 2025-02-24:
Bro, I meet the same problem like you. I fixed it using the method as below:

Actually, I installed it twice. The first time: pip install rsl_rl, which means the latest. Even though i delete it already using pip uninstall rsl_rl. But the rsl_rl_lib in python3 lib won't be deleted. I delete it in /home/yangzhi/.local/lib/python3.10. After that, I using the required version1.0.2. 

Then, everything goes right