Genesis-Embodied-AI/genesis-world #754: [Feature]: Luisa Render upgrade to support aarch64

johnnynunez, 2025-02-19:
### What feature or enhancement are you proposing?

I wonder if you planned to use new luisarender commits, because now it's compatible with cuda + arm (jetson and grace)

### Motivation

Coming devices, like digits, jetson thor etc from Nvidia

### Potential Benefit

CUDA + ARM support devices

### What is the expected outcome of the implementation work?

Upgrade fork repository to last commit

johnnynunez, 2025-03-03:
@zhenjia-xu @duburcqa 
Is it possible to prioritise this? That's how I have everything working in aarch64

johnnynunez, 2025-03-06:
@Kashu7100 could you move this forward?

johnnynunez, 2025-03-08:
@zhenjia-xu @duburcqa @Alif-01 @Kashu7100  any news?

johnnynunez, 2025-03-09:
@ACMLCZH I send you an email

ACMLCZH, 2025-03-10:
Thanks. I will update LuisaRender soon.

johnnynunez, 2025-03-10:
> Thanks. I will update LuisaRender soon.

Could be before gtc?
If you upgrade to last commit is compatible with:
ARM
WINDOWS 2025
macos-15
Also comes with nvcomp for sbsa (grace)
Cuda 12.8.1

johnnynunez, 2025-03-10:
also take account optix for arm

[image] @ACMLCZH

johnnynunez, 2025-03-10:
@ACMLCZH @zhenjia-xu @duburcqa @Alif-01 @Kashu7100 FYI:
https://github.com/LuisaGroup/LuisaCompute/issues/173#issuecomment-2710438992

johnnynunez, 2025-03-14:
@ACMLCZH @zhenjia-xu @duburcqa @Alif-01 @Kashu7100 FYI:
I added support on LuisaRender for OIDN https://github.com/RenderKit/oidn/pull/254

ACMLCZH, 2025-03-17:
@johnnynunez Thank you for your work on LuisaCompute and LuisaRender! I was updating LuisaRender and LuisaCompute in Genesis recently and encountered some issues with handling `nvcomp`.
1. **Download Issue**: The download for `nvcomp` version [4.2.0.11](https://github.com/LuisaGroup/LuisaCompute/blob/8564c2f29d4c2866e01c5127b36f740f4122e0aa/src/backends/cuda/CMakeLists.txt#L101) failed because the URL provided by NVIDIA has changed since version `4.0.0`. I updated it to the [correct URL](https://github.com/ACMLCZH/LuisaCompute/blob/499096274c21b7384dffa69331226e8ae8e97505/src/backends/cuda/CMakeLists.txt#L88). (Also note that version `4.0.0` uses another different link pattern compared to both versions before and after `4.0.0`. Should we consider this in our updates?)
2. **Target Consolidation**: After version `4.0.0`, `nvcomp` appears to combine the targets into a single file `libnvcomp.so`. Therefore, the targets `nvcomp_bitcomp` and `nvcomp_gdeflate` should be [removed](https://github.com/ACMLCZH/LuisaCompute/blob/499096274c21b7384dffa69331226e8ae8e97505/src/backends/cuda/CMakeLists.txt#L140).
3. **Library Mismatch**: When running the code, it reports an error indicating that `liblc-backend-cuda` is looking for `libnvcomp.so.4`, but the dynamic library copied to the build folder is `libnvcomp.so.4.2.0.11`, causing a mismatch. Manually renaming `libnvcomp.so.4.2.0.11` to `libnvcomp.so.4` resolves the issue, but is there a way to automate this renaming process?

[image]

Would you please check these issues? Are my modifications correct?

ACMLCZH, 2025-03-18:
I created a PR #886 to update LuisaRender. It is expected to support aarch64 now.

duburcqa, 2025-03-22:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/issues/854

closed 2025-03-22 by duburcqa