Genesis-Embodied-AI/genesis-world #762: [Bug]: TypeError: not a sequence when loading MJCF

HaoxiangYou, 2025-02-20:
### Bug Description

I want to implement several RL benchmark with Genesis. 

However, when I try to open(via gs view) xml from [deepmind mujoco](https://github.com/google-deepmind/mujoco/blob/main/model/humanoid/humanoid.xml) or [brax](https://github.com/google/brax/blob/main/brax/envs/assets/humanoid.xml),
it report  
```
 File "/media/haoxiangyou/6c569641-7e1c-4595-b053-7021a7dbe092/Documents/Yale/Research/Genesis/genesis/engine/entities/rigid_entity/rigid_entity.py", line 1825, in _get_dofs_idx
    return self._get_dofs_idx_local(dofs_idx_local) + self._dof_start
  File "/media/haoxiangyou/6c569641-7e1c-4595-b053-7021a7dbe092/Documents/Yale/Research/Genesis/genesis/engine/entities/rigid_entity/rigid_entity.py", line 1819, in _get_dofs_idx_local
    dofs_idx_local = torch.as_tensor(dofs_idx_local, dtype=gs.tc_int, device=gs.device)
TypeError: not a sequence
```

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

```
gs.init(backend=gs.cpu)

scene = gs.Scene(sim_options=gs.options.SimOptions(dt=dt))

humanoid = scene.add_entity(
    gs.morphs.MJCF(file=os.path.join(dir,"asset/humanoid.xml")),
)

dofs_names = [
    # "abdomen_y",
    "abdomen_z",
    "abdomen_x"
]

dofs_idx = [humanoid.get_joint(name).dof_idx_local for name in dofs_names]

print(dofs_idx)
```
I saw the dofs_idx = [[6,7], 12].
I also cannot retrieve joint "abdomen_y" via get_joint(name) althrough it is in xml file

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 22.04

### Release version or Commit ID

commit ID: e7f98d9360e6ab0d60f6cffcb902ae583f31b72c

Kashu7100, 2025-02-21:
Some joints in `dofs_idx_local` have multiple dofs, and we need to work on such cases. 
@YilingQiao I think this is also related to the ball joint problem.

YilingQiao, 2025-02-26:
Can you share us a minimal example?

This script works fine for me
```
import genesis as gs
import time
import numpy as np

gs.init(backend=gs.gpu)

scene = gs.Scene(
    show_viewer=True,
)

robot = scene.add_entity(
    gs.morphs.MJCF(
        file="./humanoid/humanoid.xml", # https://github.com/google-deepmind/mujoco/tree/main/model/humanoid
    ),
    vis_mode="collision",
)

scene.build()
rigid = scene.sim.rigid_solver

robot.set_dofs_kp(rigid.dofs_info.kp.to_numpy()[:3], np.array([2, 3, 7]))
robot._get_dofs_idx(1)
robot._get_dofs_idx(np.array([1, 2, 3]))

for i in range(1000):
    scene.step()
```

HaoxiangYou, 2025-02-26:
I wonder

HaoxiangYou, 2025-02-26:
I wonder whether this is designed the feature several joints in the xml files seems to be merged into one joint. Below is an example:
```import genesis as gs
import numpy as np

gs.init(backend=gs.gpu)

scene = gs.Scene(
    show_viewer=True,
)

robot = scene.add_entity(
    gs.morphs.MJCF(
        file="./humanoid/humanoid.xml", # https://github.com/google-deepmind/mujoco/tree/main/model/humanoid
    ),
    vis_mode="collision",
)

scene.build()
rigid = scene.sim.rigid_solver

for joint in robot.joints:
    print(joint.name)

print(robot.get_joint("abdomen_z"))
```
Here, the xml file in humanoid contains two joint at same location:
```
<joint limited="true" name="abdomen_z" pos="0 0 .065" axis="0 0 1" range="-45 45" damping="5" armature=".02"/>
<joint limited="true" name="abdomen_y" pos="0 0 .065" axis="0 1 0" range="-75 30" damping="5" armature=".01"/>
```
However, ```print(joint.name)``` only contains the following joints without "abdomen_y": 
```
root
head_joint
abdomen_z
shoulder1_right
shoulder1_left
abdomen_x
elbow_right
elbow_left
hip_x_right
hip_x_left
hand_right_joint
hand_left_joint
knee_right
knee_left
ankle_y_right
ankle_y_left
world_joint
```
Moreover, the output of ```print(robot.get_joint("abdomen_z"))``` 
``` 
'n_qs': <int>: 2
          'n_dofs': <int>: 2
           'q_end': <int>: 9
           'q_idx': <list>: [7, 8]
```
seems suggest the dof of abdomen_y is merged to the abdomen_z.

I wonder in this case, how should I control the desired joints, or how should I associated the dofs_idx to the desired joints.

Thanks!

YilingQiao, 2025-02-27:
yeah, at this moment we merge joints within the same link to one joint. Would it help if we change the name into `abdomen_x,abdomen_y,abdomen_z` instead of using `abdomen_z`?

HaoxiangYou, 2025-02-28:
I think that helps a lot! Thank you!

closed 2025-03-20 by duburcqa