Genesis-Embodied-AI/genesis-world #764: [Feature]: Could you please let us know when the team will be able to provide collision detection interfaces between rigid bodies and methods for coupling multiple solvers?

Tryoe, 2025-02-21:
### What feature or enhancement are you proposing?

****Could you please let us know when the team will be able to provide collision detection interfaces between rigid bodies and methods for coupling multiple solvers?

### Motivation

As a [e.g. civil engineering researcher working on flood simulation]
I want to [e.g. be able to simulate city-scale fluid / terrain simulations]
Because [e.g. this enables the efficient design and validation of flood protection structures].

### Potential Benefit

Could you please let us know when the team will be able to provide collision detection interfaces between rigid bodies and methods for coupling multiple solvers?

### What is the expected outcome of the implementation work?

- [ ]Could you please let us know when the team will be able to provide collision detection interfaces between rigid bodies and methods for coupling multiple solvers?

### Additional information

Could you please let us know when the team will be able to provide collision detection interfaces between rigid bodies and methods for coupling multiple solvers?

bxtbold, 2025-02-22:
For collision detection, it is already available at [`RigidEntity.collision_detection()`](https://github.com/Genesis-Embodied-AI/Genesis/blob/fa388580a38f17991a8fca96b301db1cc6bf05b5/genesis/engine/entities/rigid_entity/rigid_entity.py#L2300), which will return indices of collided pairs.

Genesis-Embodied-AI, 2025-02-23:
Inter-solver coupling is already implemented. Any idea what kind of api you are looking for?

Tryoe, 2025-02-24:
Thank you for your response. I'm specifically looking for APIs that can simulate localized forces acting on rigid bodies (represented as meshes) and their subsequent fracturing with fragment trajectories. Do you have any recommendations for such tools? I appreciate your help once again!

duburcqa, 2025-09-27:
What you are trying to achieve is not clear to me. I have no other choice than closing for now, unless you can provide additional details.

closed 2025-09-27 by duburcqa