Genesis-Embodied-AI/genesis-world #771: [Feature]: Different control abstraction for the drone environment

atahirkarasahin, 2025-02-23:
### What feature or enhancement are you proposing?

The current drone environment use an one of control abstraction layer such as rpm. However, I think that collective thrust and body rate (CTBR) is important especially real-world experiment. So, could you please add different control abstraction layer especially CTBR for the drone environment?

### Motivation

As a aerial robotics researcher working on reinforcement learning, I want to use RL policy in real-world experiment. For this reason, I think that the genesis environment being a different control abstraction for sim2real scenarios will support the development of this environment.

### Potential Benefit

If the genesis environment offers a different layer of control abstraction, it will help researchers to implement the RL policy in a real-world experiment with fewer sim2real problems.

### What is the expected outcome of the implementation work?

- Fewer sim2real problems during the evaluation of RL policy

KafuuChikai, 2025-03-28:
You can reference the PID controller implementation in `Genesis/examples/drone/quadcopter_controller.py` to develop your external controller.

From our understanding, the development team may not be prioritizing the integration of this control abstraction layer directly into the core engine at this time. 

You can consider developing a custom CTBR controller as an extension module. This wouldn't require modifying the core engine while still meeting your research requirements.

atahirkarasahin, 2025-03-31:
Thank you for your suggestion. I am comparing options high-parallel simulation environment to use in the real-world experiment. So, CTBR control abstraction is a criteria for these options.

KafuuChikai, 2025-03-31:
I'm also considering developing a general external module for similar applications (can be used in [GenesisDroneEnv](https://github.com/KafuuChikai/GenesisDroneEnv) or [Dashing-for-the-Golden-Snitch-Multi-Drone-RL](https://github.com/KafuuChikai/Dashing-for-the-Golden-Snitch-Multi-Drone-RL)) as well, but I'll need to complete organizing my current open-source project first.

atahirkarasahin, 2025-03-31:
Thank you for your sharing. I am investigate it.