Genesis-Embodied-AI/genesis-world #775: [Bug]: 设置damping会导致整个entity受力而非关节阻尼

Albusgive, 2025-02-24:
### Bug Description

如下初始化会导致entity受到一个力,并非dof增加阻尼

### Steps to Reproduce

  damping = np.full((self.num_envs, self.num_actions), self.env_cfg["damping"])
        self.robot.set_dofs_damping(damping=damping, 
                                   dofs_idx_local=np.arange(0, self.num_actions),)

### Expected Behavior

entity不受外部作用力,dof增加damping

### Screenshots/Videos

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 24.04]
- GPU/CPU [RTX 4070, M3pr, Intel I5-12400f] (N/A if no GPU/CPU)
- GPU-driver version NV550
- CUDA / CUDA-toolkit version 12.4

### Release version or Commit ID

main

Albusgive, 2025-02-24:
[wheel_legged_.zip](https://github.com/user-attachments/files/18943502/wheel_legged_.zip)  
bug实现代码

YilingQiao, 2025-02-25:
The fisrt 6 dofs are for free joint (i.e. for the entire entity). You should leave them 0.

closed 2025-02-26 by YilingQiao