Genesis-Embodied-AI/genesis-world #776: [Feature]: How do I apply external forces to a rigid body?

Tryoe, 2025-02-24:
### What feature or enhancement are you proposing?

How do I apply external forces to a rigid body?

### Motivation

As a [e.g. civil engineering researcher working on flood simulation]
I want to [e.g. be able to simulate city-scale fluid / terrain simulations]
Because [e.g. this enables the efficient design and validation of flood protection structures].

### Potential Benefit

How do I apply external forces to a rigid body?

### What is the expected outcome of the implementation work?

- [ ]

Kashu7100, 2025-02-24:
You can apply via RigidSolver

```python
for solver in scene.sim.solvers:
        if not isinstance(solver, RigidSolver):
            continue
        rigid_solver = solver

rigid_solver.apply_links_external_force(
            force=np.array([[1,0,0]]),
            links_idx=[1], # global link idx
        )
```

Tryoe, 2025-02-25:
Thank you for your feedback! Regarding your question: Does Genesis have custom configuration options for the maximum stiffness of rigid bodies, and a solver capable of applying different forces to distinct regions of the model?

Kashu7100, 2025-02-26:
yes you can apply forces to each link in the model. you can `set_dofs_stiffness` if this is what you mean.

Tryoe, 2025-03-04:
> 是的，您可以对模型中的每个链接施加力。如果这是您的意思，您可以。`set_dofs_stiffness`

Thanks, my problem is that different forces need to be set for a mesh object to be discretized into different parts.

JunyinLi7758, 2025-03-05:
BTW, Is there any possible to apply contraint or force on a FEM entity or MPM entity?

closed 2025-05-22 by duburcqa