Genesis-Embodied-AI/genesis-world #780: [Bug]: stiffness存在类似damping问题

Albusgive, 2025-02-25:
### Bug Description

设置stiffness存在外部作用力

### Steps to Reproduce

        stiffness = np.full((self.num_envs, self.robot.n_dofs), self.env_cfg["stiffness"])
        self.robot.set_dofs_stiffness(stiffness=stiffness, 
                                   dofs_idx_local=np.arange(0, self.robot.n_dofs))

### Expected Behavior

无外部作用力

### Screenshots/Videos

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 22.04]
- GPU/CPU [RTX 4070 Intel I9-12400f] (N/A if no GPU/CPU)
- GPU-driver version NV550
- CUDA / CUDA-toolkit version cuda12.4

### Release version or Commit ID

main

duburcqa, 2025-02-25:
Could you provide a complete reproduction script that is self contained ? May be on the Franka robot arm of the tutorial. It would be a good way to check if the issue is related to your robot or more general.

Albusgive, 2025-02-25:
> 您能否提供一个完整的自包含复制脚本？可能在 Franka 机械臂上的教程。这将是检查问题是否与您的机器人或更普遍的机器人有关的好方法。

https://github.com/Albusgive/wheel_legged_genesis
In this code, when setting dapming and stiffness, the base will be subjected to an external force

closed 2025-02-26 by YilingQiao