Genesis-Embodied-AI/genesis-world #786: [Bug]:

Mengmouxu, 2025-02-26:
### Bug Description

After compiling the LuisaRender and copy the build folder in the conda env, the rendering/demo.py report the error as above. How could I fix them?

### Steps to Reproduce

python examples/rendering/demo.py

### Expected Behavior

Correctly run this demo

### Relevant log output

```shell
[Genesis] [15:15:53] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [15:15:53] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [15:15:53] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [15:15:53] [INFO] Running on [NVIDIA GeForce RTX 3090] with backend gs.cuda. Device memory: 23.60 GB.
[Genesis] [15:15:54] [DEBUG] [Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.10.14
[Genesis] [15:15:54] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [15:15:54] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/GPFS/data/yanxumeng/yanxumeng/backup/Genesis/examples/rendering/demo.py", line 168, in <module>
    main()
  File "/GPFS/data/yanxumeng/yanxumeng/backup/Genesis/examples/rendering/demo.py", line 11, in main
    scene = gs.Scene(
  File "/GPFS/rhome/yanxumeng/anaconda3/envs/genesis_new/lib/python3.10/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/GPFS/rhome/yanxumeng/anaconda3/envs/genesis_new/lib/python3.10/site-packages/genesis/engine/scene.py", line 147, in __init__
    self._visualizer = Visualizer(
  File "/GPFS/rhome/yanxumeng/anaconda3/envs/genesis_new/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 62, in __init__
    self._renderer = self._raytracer = Raytracer(renderer, vis_options)
  File "/GPFS/rhome/yanxumeng/anaconda3/envs/genesis_new/lib/python3.10/site-packages/genesis/vis/raytracer.py", line 174, in __init__
    LuisaRenderPy.init(
TypeError: init(): incompatible function arguments. The following argument types are supported:
    1. (context_path: str, cuda_device: int = 0, log_level: LuisaRenderPy.LogLevel = <LogLevel.WARNING: 2>, integrator_options: LuisaRenderPy.Integrator = <LuisaRenderPy.Integrator object at 0x7f694075da30>, spectrum_options: LuisaRenderPy.Spectrum = <LuisaRenderPy.Spectrum object at 0x7f694075d9f0>, clamp_normal: float = 0.0) -> None

Invoked with: kwargs: context_path='/GPFS/rhome/yanxumeng/anaconda3/envs/genesis_new/lib/python3.10/site-packages/genesis/ext/LuisaRender/build/bin', context_id='535ba51e79df4da08309c9dd27697a7d', cuda_device=0, log_level=<LogLevel.WARNING: 2>
[Genesis] [15:16:02] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

### Environment

CUDA 12.1 python=3.10

### Release version or Commit ID

genesis 0.2.1

closed 2025-02-27 by Mengmouxu

yamanoko-do, 2025-06-21:
how to fix this problem?

yamanoko-do, 2025-06-21:
May I ask how you solved this problem?