Genesis-Embodied-AI/genesis-world #788: [Bug]: Build Fails for Python < 3.12

bxtbold, 2025-02-26:
### Bug Description

The build fails for Python<=3.11 in `genesis/ext/fast_simplification/_simplify.cpp` with errors like `PyDataType_ELSIZE` was not declared. This seems to be a numpy API mismatch in the cython extensions, possibly from changes in #711

### Steps to Reproduce

Try building from the source.

### Expected Behavior

Installing work for .

### Relevant log output

```shell
python3.10

      Compiling genesis/ext/fast_simplification/_simplify.pyx because it depends on /tmp/pip-build-env-yhwkmbad/overlay/lib/python3.10/site-packages/Cython/Includes/libc/string.pxd.
      [1/1] Cythonizing genesis/ext/fast_simplification/_simplify.pyx
      building 'genesis.ext.fast_simplification._simplify' extension
      x86_64-linux-gnu-g++ -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DIS64BITPLATFORM -I/home/bat/ws/venvs/genesis_test/.venv/include -I/usr/include/python3.10 -I/tmp/pip-build-env-yhwkmbad/overlay/lib/python3.10/site-packages/numpy/_core/include -c genesis/ext/fast_simplification/_simplify.cpp -o build/temp.linux-x86_64-cpython-310/genesis/ext/fast_simplification/_simplify.o -std=gnu++11 -O3 -w
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘npy_intp __pyx_f_5numpy_5dtype_8itemsize_itemsize(PyArray_Descr*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18234:13: error: ‘PyDataType_ELSIZE’ was not declared in this scope; did you mean ‘PyDataType_ISUNSIZED’?
      18234 |   __pyx_r = PyDataType_ELSIZE(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~
            |             PyDataType_ISUNSIZED
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘npy_intp __pyx_f_5numpy_5dtype_9alignment_alignment(PyArray_Descr*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18268:13: error: ‘PyDataType_ALIGNMENT’ was not declared in this scope; did you mean ‘PyDataType_ISSIGNED’?
      18268 |   __pyx_r = PyDataType_ALIGNMENT(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~~
            |             PyDataType_ISSIGNED
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘PyObject* __pyx_f_5numpy_5dtype_6fields_fields(PyArray_Descr*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18306:15: error: ‘PyDataType_FIELDS’ was not declared in this scope; did you mean ‘PyDataType_HASFIELDS’?
      18306 |   __pyx_t_1 = PyDataType_FIELDS(__pyx_v_self);
            |               ^~~~~~~~~~~~~~~~~
            |               PyDataType_HASFIELDS
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘PyObject* __pyx_f_5numpy_5dtype_5names_names(PyArray_Descr*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18348:15: error: ‘PyDataType_NAMES’ was not declared in this scope; did you mean ‘PyDataType_ISNUMBER’?
      18348 |   __pyx_t_1 = PyDataType_NAMES(__pyx_v_self);
            |               ^~~~~~~~~~~~~~~~
            |               PyDataType_ISNUMBER
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘PyArray_ArrayDescr* __pyx_f_5numpy_5dtype_8subarray_subarray(PyArray_Descr*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18386:13: error: ‘PyDataType_SUBARRAY’ was not declared in this scope; did you mean ‘PyDataType_HASSUBARRAY’?
      18386 |   __pyx_r = PyDataType_SUBARRAY(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~
            |             PyDataType_HASSUBARRAY
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘npy_uint64 __pyx_f_5numpy_5dtype_5flags_flags(PyArray_Descr*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18420:13: error: ‘PyDataType_FLAGS’ was not declared in this scope; did you mean ‘PyDataType_FLAGCHK’?
      18420 |   __pyx_r = PyDataType_FLAGS(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~
            |             PyDataType_FLAGCHK
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘int __pyx_f_5numpy_9broadcast_7numiter_numiter(PyArrayMultiIterObject*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18454:13: error: ‘PyArray_MultiIter_NUMITER’ was not declared in this scope; did you mean ‘PyArray_MultiIter_NEXTi’?
      18454 |   __pyx_r = PyArray_MultiIter_NUMITER(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~~~~~~~
            |             PyArray_MultiIter_NEXTi
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘npy_intp __pyx_f_5numpy_9broadcast_4size_size(PyArrayMultiIterObject*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18488:13: error: ‘PyArray_MultiIter_SIZE’ was not declared in this scope; did you mean ‘PyArray_MultiIter_GOTO’?
      18488 |   __pyx_r = PyArray_MultiIter_SIZE(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~~~~
            |             PyArray_MultiIter_GOTO
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘npy_intp __pyx_f_5numpy_9broadcast_5index_index(PyArrayMultiIterObject*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18522:13: error: ‘PyArray_MultiIter_INDEX’ was not declared in this scope; did you mean ‘PyArray_MultiIter_NEXT’?
      18522 |   __pyx_r = PyArray_MultiIter_INDEX(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~~~~~
            |             PyArray_MultiIter_NEXT
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘int __pyx_f_5numpy_9broadcast_2nd_nd(PyArrayMultiIterObject*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18556:13: error: ‘PyArray_MultiIter_NDIM’ was not declared in this scope; did you mean ‘PyArray_MultiIter_NEXT’?
      18556 |   __pyx_r = PyArray_MultiIter_NDIM(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~~~~
            |             PyArray_MultiIter_NEXT
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘npy_intp* __pyx_f_5numpy_9broadcast_10dimensions_dimensions(PyArrayMultiIterObject*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18590:13: error: ‘PyArray_MultiIter_DIMS’ was not declared in this scope; did you mean ‘PyArray_MultiIter_DATA’?
      18590 |   __pyx_r = PyArray_MultiIter_DIMS(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~~~~
            |             PyArray_MultiIter_DATA
      genesis/ext/fast_simplification/_simplify.cpp: In function ‘void** __pyx_f_5numpy_9broadcast_5iters_iters(PyArrayMultiIterObject*)’:
      genesis/ext/fast_simplification/_simplify.cpp:18624:13: error: ‘PyArray_MultiIter_ITERS’ was not declared in this scope; did you mean ‘PyArray_MultiIter_NEXT’?
      18624 |   __pyx_r = PyArray_MultiIter_ITERS(__pyx_v_self);
            |             ^~~~~~~~~~~~~~~~~~~~~~~
            |             PyArray_MultiIter_NEXT
      error: command '/usr/bin/x86_64-linux-gnu-g++' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for genesis-world
Failed to build genesis-world
ERROR: Could not build wheels for genesis-world, which is required to install pyproject.toml-based projects

python3.11

      [1/1] Cythonizing genesis/ext/fast_simplification/_simplify.pyx
      building 'genesis.ext.fast_simplification._simplify' extension
      x86_64-linux-gnu-g++ -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DIS64BITPLATFORM -I/home/bat/ws/venvs/test_genesis_2/.venv/include -I/usr/include/python3.11 -I/tmp/pip-build-env-w_wp88il/overlay/lib/python3.11/site-packages/numpy/_core/include -c genesis/ext/fast_simplification/_simplify.cpp -o build/temp.linux-x86_64-cpython-311/genesis/ext/fast_simplification/_simplify.o -std=gnu++11 -O3 -w
      genesis/ext/fast_simplification/_simplify.cpp:46:10: fatal error: Python.h: No such file or directory
         46 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/x86_64-linux-gnu-g++' failed with exit code 1
      [end of output]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for genesis-world
  Building wheel for pygltflib (setup.py) ... done
  Created wheel for pygltflib: filename=pygltflib-1.16.0-py3-none-any.whl size=26754 sha256=73deccb4b8f0894b7827905a2b52a1f7c3e756276920dcdbe5936e486860481a
  Stored in directory: /home/bat/.cache/pip/wheels/11/5b/3f/80645f53862a37d1134843a9727109bb3e81fc04214ee7d5ec
Successfully built pygltflib
Failed to build genesis-world
ERROR: Could not build wheels for genesis-world, which is required to install pyproject.toml-based projects
```

### Environment

- OS: Ubuntu 22.04
- Python 3.10

### Release version or Commit ID

245f49c2141ca978ea282a53904939a6b6bde1a8

YilingQiao, 2025-02-26:
#711  changes `numpy == 1.26.4` to `numpy >= 1.26.4`
So you could probably still pip install `numpy == 1.26.4`. Would that work?

Maybe I'll revert it to `numpy == 1.26.4` at this moment.

duburcqa, 2025-02-26:
@YilingQiao Wait a minute, I will check if I can reproduce the issue with numpy 2.X, and if so, I will provide a fix. It should be straightforward.

johnnynunez, 2025-02-26:
> [@YilingQiao](https://github.com/YilingQiao) Wait a minute, I will check if I can reproduce the issue with numpy 2.X, and if so, I will provide a fix. It should be straightforward.

I don’t have this error.
Maybe you install numpy 2.2.0 that has more changes

https://www.jetson-ai-lab.com/genesis.html

johnnynunez, 2025-02-26:
> [@johnnynunez](https://github.com/johnnynunez) Yes, from what I understand, the issue is unrelated. Still investigating.

In my cases, I have problems with outdated ompl and luisa render.
Ompl uses castxml and pygccxml.
I’m still working to fix pygccxml that it’s bugged from 2 years.

Now it’s starting working again because I fixed castxml builds to latest version 0.6.11. But pygccxml is in WIP.

On the luisa render side. The problems was fixed from last week, and official trimesh repository supports numpy 2.2.0 and arm. The problem is that genesis uses luisa render and trimesh modified.

I would like if the owners can updated that and has it on top list requirements. I love genesis but is still bugged if you uses de latest packages

duburcqa, 2025-02-26:
Yes, I will have a look at it in the following days.

bxtbold, 2025-02-27:
Thank you for quick replies despite latter comments are unrelated to the issue.

> [#711](https://github.com/Genesis-Embodied-AI/Genesis/pull/711) changes `numpy == 1.26.4` to `numpy >= 1.26.4` So you could probably still pip install `numpy == 1.26.4`. Would that work?
> 
> Maybe I'll revert it to `numpy == 1.26.4` at this moment.

I have been working on the issue using clean python venvs and found a quick(minimal?) fix by adding numpy header for extensions in setup.py.

I have opened a PR on #794. Could you take a look at it? @YilingQiao

closed 2025-02-27 by duburcqa