Genesis-Embodied-AI/genesis-world #789: [Bug]: ValueError: Unable to avoid copy while creating an array as requested.

alex1600, 2025-02-26:
### Bug Description

Bug appears when dragging to move the camera in the render with the left click.

### Steps to Reproduce

```python
import genesis as gs
gs.init(backend=gs.cuda)

scene = gs.Scene(show_viewer=True,show_FPS=False)
plane = scene.add_entity(gs.morphs.Plane())

scene.build()

for i in range(5000):
    scene.step()
```

### Expected Behavior

Should not display an exception and enable to rotate the camera

### Relevant log output

```shell
Exception ignored on calling ctypes callback function: <function Win32Window._get_window_proc.<locals>.f at 0x00000286D65B20E0>
Traceback (most recent call last):
  File "c:\Users\xxxxxxx\AppData\Local\pypoetry\Cache\virtualenvs\genesis-project-2i1rbOal-py3.10\lib\site-packages\pyglet\window\win32\__init__.py", line 839, in f
    result = event_handler(msg, wParam, lParam)
  File "c:\Users\xxxxxxx\AppData\Local\pypoetry\Cache\virtualenvs\genesis-project-2i1rbOal-py3.10\lib\site-packages\pyglet\window\win32\__init__.py", line 1073, in _event_mousemove       
    self.dispatch_event('on_mouse_drag',
  File "c:\Users\xxxxxxx\AppData\Local\pypoetry\Cache\virtualenvs\genesis-project-2i1rbOal-py3.10\lib\site-packages\pyglet\window\__init__.py", line 672, in dispatch_event
    super().dispatch_event(*args)
  File "c:\Users\xxxxxxx\AppData\Local\pypoetry\Cache\virtualenvs\genesis-project-2i1rbOal-py3.10\lib\site-packages\pyglet\event.py", line 368, in dispatch_event
    if getattr(self, event_type)(*args):
  File "c:\Users\xxxxxxx\AppData\Local\pypoetry\Cache\virtualenvs\genesis-project-2i1rbOal-py3.10\lib\site-packages\genesis\ext\pyrender\viewer.py", line 737, in on_mouse_drag
    self._trackball.drag(np.array([x, y]))
  File "c:\Users\xxxxxxx\AppData\Local\pypoetry\Cache\virtualenvs\genesis-project-2i1rbOal-py3.10\lib\site-packages\genesis\ext\pyrender\trackball.py", line 112, in drag
    x_rot_mat = transformations.rotation_matrix(x_angle, y_axis, target)
  File "c:\Users\xxxxxxx\AppData\Local\pypoetry\Cache\virtualenvs\genesis-project-2i1rbOal-py3.10\lib\site-packages\genesis\ext\trimesh\transformations.py", line 379, in rotation_matrix  
    point = np.array(point[:3], dtype=np.float64, copy=False)
ValueError: Unable to avoid copy while creating an array as requested.
If using `np.array(obj, copy=False)` replace it with `np.asarray(obj)` to allow a copy when needed (no behavior change in NumPy 1.x).
For more details, see https://numpy.org/devdocs/numpy_2_0_migration_guide.html#adapting-to-changes-in-the-copy-keyword.
```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

commit hash: 245f49c2141ca978ea282a53904939a6b6bde1a8

### Additional Context

OS: Windows
GPU: [NVIDIA RTX A2000 Laptop GPU]

alex1600, 2025-02-26:
I confirm if i downgrade numpy to 1.26.4 the issue do not appears. 
Based on https://numpy.org/devdocs/release/2.0.0-notes.html#copy-keyword-changes-2-0 
The difference is in numpy 2.0 the argument copy=False for np.array() will trigger an exceptions in case it requires a copy.

duburcqa, 2025-02-26:
Yes, switch to `np.asarray(obj)` in this case is the way to go.

duburcqa, 2025-02-26:
While `numpy>=2.0` is working fine on Windows OS, the robot is not displayed on Mac OS (while the scene is properly rendered). I'm investigating the issue. Probably related to some memory lifetime issues.

closed 2025-02-28 by duburcqa