Genesis-Embodied-AI/genesis-world #797: [Bug]: Instability when simulating Open-Loop Robots

lgleim, 2025-02-27:
### Bug Description

When simulating closed-loop robots (based on https://github.com/Genesis-Embodied-AI/Genesis/pull/636), such as the [robotiq gripper](https://github.com/google-deepmind/mujoco_menagerie/tree/main/robotiq_2f85_v4), the simulation is highly unstable and results in rapid energy accumulation & eventually explosion (not shown in the video below)

https://github.com/user-attachments/assets/c11289af-a53a-40af-9434-01a342a94152

### Steps to Reproduce

```python
import time
import genesis as gs

########################## init ##########################
gs.init(backend=gs.gpu, precision="32")
########################## create a scene ##########################
scene = gs.Scene(
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(3, -1, 1.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=30,
        res=(960, 640),
        max_FPS=60,
    ),
    sim_options=gs.options.SimOptions(
        dt=0.01,
    ),
    rigid_options=gs.options.RigidOptions(
        box_box_detection=True,
    ),
    show_viewer=True,
)

########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)
robot = scene.add_entity(
    gs.morphs.MJCF(file="xml/robotiq_2f85_v4/mjx_2f85.xml"),
)

cube = scene.add_entity(
    gs.morphs.Box(
        size=(0.04, 0.04, 0.04),
        pos=(0.65, 0.0, 0.02),
    )
)
########################## build ##########################
scene.build()

for i in range(10):
    time.sleep(1)
    scene.step()
```

### Expected Behavior

Stable simulation out of the box

### Relevant log output

```shell
[Genesis] [11:31:23] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [11:31:23] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [11:31:23] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [11:31:23] [INFO] Running on [NVIDIA GeForce RTX 4090] with backend gs.cuda. Device memory: 23.49 GB.
[Genesis] [11:31:24] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [11:31:24] [INFO] Scene <ad6806e> created.
[Genesis] [11:31:24] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <3ddef28>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [11:31:24] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <b52a0da>, morph: <gs.morphs.MJCF(file='/home/lars/workspace/genesis/genesis/assets/xml/robotiq_2f85_v4/mjx_2f85.xml')>, material: <gs.materials.Rigid>.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [11:31:24] [INFO] Adding <gs.RigidEntity>. idx: 2, uid: <438d3b2>, morph: <gs.morphs.Box>, material: <gs.materials.Rigid>.
[Genesis] [11:31:24] [INFO] Building scene <ad6806e>...
[Genesis] [11:31:26] [INFO] Compiling simulation kernels...
[Genesis] [11:31:30] [INFO] Building visualizer...
[Genesis] [11:31:31] [INFO] Viewer created. Resolution: 960×640, max_FPS: 60.
[Genesis] [11:31:42] [INFO] Running at 1.81 FPS.
[Genesis] [11:31:43] [INFO] Running at 1.74 FPS.
[Genesis] [11:31:44] [INFO] Running at 1.67 FPS.
[Genesis] [11:31:45] [INFO] Running at 1.62 FPS.
[Genesis] [11:31:46] [INFO] Running at 1.57 FPS.
[Genesis] [11:31:47] [INFO] Running at 1.53 FPS.
[Genesis] [11:31:48] [INFO] Running at 1.49 FPS.
[Genesis] [11:31:49] [INFO] Running at 1.45 FPS.
[Genesis] [11:31:50] [INFO] Running at 1.42 FPS.
[Genesis] [11:31:51] [INFO] Running at 1.39 FPS.
[Genesis] [11:31:51] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

### Environment

- OS: Ubuntu 22.04
- GPU/CPU: RTX 4090
- GPU-driver version: 570
- CUDA / CUDA-toolkit 12.7

### Release version or Commit ID

2235ce0bf2c684b3aa05e16a6a6c4984ce783302

Kashu7100, 2025-02-28:
I think you need to add this RigidOptions (especially `dt`):

```python
rigid_options=gs.options.RigidOptions(
            dt=0.0025, # some dt like this will be fine
            # contact_resolve_time=0.02,
),
```

lgleim, 2025-03-04:
Reducing `dt` indeed mitigates the instability (as also noted in the original issue report) but it still remains an instability. 

Other physics engines, e.g. PhysX, are in my experience able to simulate the same setup stably at much larger `dt`. This may be related to default parameter settings (e.g. damping factors). 

Either way, the default settings on Gensis are instable for open-loop robotics and should imho either be changed (e.g. to `dt=0.002` as in MJX or by setting larger default damping values) or a dedicated section on parameter tuning should be added to the user guide.

YilingQiao, 2025-03-07:
Yeah, the default `dt` for this robot should be 0.002. Maybe we should add a user guide or troubleshooting resources on this matter.

```
import mujoco
mj_model = mujoco.MjModel.from_xml_path("mujoco_menagerie/robotiq_2f85/2f85.xml")
print(mj_model.opt.timestep) # 0.002
```

closed 2025-03-07 by YilingQiao