Genesis-Embodied-AI/genesis-world #803: [Bug]: No display detected

Kashu7100, 2025-02-27:
### Bug Description

@duburcqa After your commit at `79f12f00c3b6973873f00a86448a2bbe7bb122c8`, I am getting the following error.

```bash
(genesis) kashu@uranus:~/research/Genesis ((79f12f0...))$ python tmp/mjcf/ball_joint.py 
[Genesis] [17:39:08] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [17:39:08] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [17:39:08] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [17:39:08] [INFO] Running on [NVIDIA GeForce RTX 3080] with backend gs.cuda. Device memory: 9.77 GB.
[Genesis] [17:39:08] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/home/kashu/research/Genesis/tmp/mjcf/ball_joint.py", line 26, in <module>
    scene = gs.Scene(
  File "/home/kashu/research/Genesis/genesis/utils/misc.py", line 28, in new_init
    original_init(self, *args, **kwargs)
  File "/home/kashu/research/Genesis/genesis/engine/scene.py", line 147, in __init__
    self._visualizer = Visualizer(
  File "/home/kashu/research/Genesis/genesis/vis/visualizer.py", line 45, in __init__
    gs.raise_exception("No display detected. Use `show_viewer=False` for headless mode.")

[Genesis] [17:39:09] [ERROR] No display detected. Use `show_viewer=False` for headless mode.
[Genesis] [17:39:09] [INFO] 💤 Exiting Genesis and caching compiled kernels...
(genesis) kashu@uranus:~/research/Genesis ((79f12f0...))$ 
```

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
import genesis as gs
gs.init()

viewer_options = gs.options.ViewerOptions(
    camera_pos=(0, -3.5, 2.5),
    camera_lookat=(0.0, 0.0, 0.5),
    camera_fov=40,
    max_FPS=200,
)
scene = gs.Scene(
    viewer_options=viewer_options,
    rigid_options=gs.options.RigidOptions(
        dt=0.001,
        contact_resolve_time=0.02,
    ),
    show_viewer=True,
    show_FPS=False
)

scene.build()
```

### Expected Behavior

Prior hash `d1c7831543479595148ad32b8d98b12dd84f5f78` works fine.

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 24.04, 
- GPU/CPU RTX 3080,
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

79f12f00c3b6973873f00a86448a2bbe7bb122c8

Kashu7100, 2025-02-27:
@YilingQiao Do you think its safe to revert to `d1c7831543479595148ad32b8d98b12dd84f5f78` for now?

duburcqa, 2025-02-27:
Hum I’m a bit surprised. I’m only using pyglet to get the screen size. I’m not creating any rendering context so it should work whatever you plan to use egl or not. 

Yes I can revert if you want for now. Anyway I’m working to improve rendering support on all kind of machines so I will have time to fix it later.

Would you mind running locally the commands I’m using to get the screen object and put the backtrace exception in this thread ? It would be very helpful.

Kashu7100, 2025-02-27:
Ok, maybe this is the pyglet version issue. 

```bash
(genesis) kashu@uranus:~/research/Genesis (main)$ pip list | grep pyglet
pyglet                       1.5.31
(genesis) kashu@uranus:~/research/Genesis (main)$ python
Python 3.10.16 | packaged by conda-forge | (main, Dec  5 2024, 14:16:10) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyglet
>>> display = pyglet.display.get_display()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pyglet' has no attribute 'display'
```

duburcqa, 2025-02-28:
Ah yes indeed ! This is the issue, I know that the API of pyglet changed a lot with the release of 2.0. I will bump the minimal version required asap

duburcqa, 2025-02-28:
@Kashu7100  PR ready. I checked locally that it was working, but it would be nice if you could confirm that it works well now :)

closed 2025-02-28 by duburcqa