Genesis-Embodied-AI/genesis-world #817: [Bug]: something wrong with get_AABB

Kashu7100, 2025-03-03:
### Bug Description

When `morphs.Terrain` is added to the scene, except the first entity added, `entity.get_AABB()` fails to obtain reasonable values. I think some parts of `entity.get_verts()` or its underlying function potentially has bug with Terrain. 

### Steps to Reproduce

```python
import argparse
import time

import numpy as np
import torch

import genesis as gs

def main():

    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    parser.add_argument("-c", "--cpu", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(seed=0, backend=gs.cpu if args.cpu else gs.gpu)

    ########################## create a scene ##########################

    scene = gs.Scene(
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(-5.0, -5.0, 10.0),
            camera_lookat=(5.0, 5.0, 0.0),
            camera_fov=40,
        ),
        show_viewer=args.vis,
        rigid_options=gs.options.RigidOptions(
            dt=0.01,
            constraint_solver=gs.constraint_solver.Newton,
        ),
        vis_options=gs.options.VisOptions(
            # geom_type='sdf',
        ),
    )

    # load duck
    obj = scene.add_entity(
        morph=gs.morphs.Mesh(
            file="meshes/duck.obj",
            scale=0.1,
            pos=(0.5, 0.5, 0.5),
            euler=(0.0, 0.0, 0.0),
        ),
    )
    

    terrain = scene.add_entity(
        morph=gs.morphs.Terrain(
            n_subterrains=(1,1),
            horizontal_scale=0.25,
            vertical_scale=0.005,
            subterrain_size=(50.0, 50.0),
            subterrain_types="flat_terrain",
            name="debug",
        ),
    )  

    cam = scene.add_camera(
        res = (640,480),
        pos=(3,0,3),
        lookat=(0,0,1),
        fov=50
    )

    scene.build()

    print(obj.get_AABB())   
    print(terrain.get_AABB())

if __name__ == "__main__":
    main()
```
The code above will result in:
```
tensor([[0.3782, 0.4964, 0.2997],
        [0.6220, 0.7492, 0.6992]], device='cuda:0')
tensor([[0., 0., 0.],
        [0., 0., 0.]], device='cuda:0') # <-- something wrong
```

```python
import argparse
import time

import numpy as np
import torch

import genesis as gs

def main():

    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    parser.add_argument("-c", "--cpu", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(seed=0, backend=gs.cpu if args.cpu else gs.gpu)

    ########################## create a scene ##########################

    scene = gs.Scene(
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(-5.0, -5.0, 10.0),
            camera_lookat=(5.0, 5.0, 0.0),
            camera_fov=40,
        ),
        show_viewer=args.vis,
        rigid_options=gs.options.RigidOptions(
            dt=0.01,
            constraint_solver=gs.constraint_solver.Newton,
        ),
        vis_options=gs.options.VisOptions(
            # geom_type='sdf',
        ),
    )

    terrain = scene.add_entity(
        morph=gs.morphs.Terrain(
            n_subterrains=(1,1),
            horizontal_scale=0.25,
            vertical_scale=0.005,
            subterrain_size=(50.0, 50.0),
            subterrain_types="flat_terrain",
            name="debug",
        ),
    )  

    # load duck
    obj = scene.add_entity(
        morph=gs.morphs.Mesh(
            file="meshes/duck.obj",
            scale=0.1,
            pos=(0.5, 0.5, 0.5),
            euler=(0.0, 0.0, 0.0),
        ),
    )

    cam = scene.add_camera(
        res = (640,480),
        pos=(3,0,3),
        lookat=(0,0,1),
        fov=50
    )

    scene.build()

    print(obj.get_AABB())   
    print(terrain.get_AABB())

if __name__ == "__main__":
    main()
```
This version with flipped order will result in:
```
tensor([[0.0000, 0.0000, 0.0000],
        [0.6220, 0.7486, 0.6992]], device='cuda:0') # <-- something is messed up
tensor([[ 0.0000,  0.0000, -1.0000],
        [49.7500, 49.7500,  0.0000]], device='cuda:0')
```

### Expected Behavior

We can get the correct AABB regardless of order, combination.

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 24.04]
- GPU/CPU [RTX 3080]
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

[98b29b4](https://github.com/Genesis-Embodied-AI/Genesis/commit/98b29b4a4e56bdaa55ca1d8a359c5935eb24f26c)

Kashu7100, 2025-03-04:
@duburcqa if you have some time, could you also check this?

YilingQiao, 2025-03-04:
I'm on my way to moving to CA. I'll test it tomorrow.

Kashu7100, 2025-03-04:
Have a safe trip!

YilingQiao, 2025-03-06:
Caused by #673 
`_func_update_all_verts` function is not right

PR #830 (2025-03-06): [BUG FIX] fix verts_state_idx

## Description

Fix #817 

There are two bugs.
1. should use `verts_state_idx` instead of `i_v` in _func_update_verts_for_geom
2. should not +1 when using np.arange

## Related Issue
closed 2025-03-06 by Kashu7100