Genesis-Embodied-AI/genesis-world #826: [Bug]: should _kernel_forward_kinematics_links_geoms take envs_idx?

Kashu7100, 2025-03-04:
### Bug Description

Should `_kernel_forward_kinematics_links_geoms()` in rigid_solver take envs_idx as input?
They are called in multiple places using envs_idx, but `_kernel_forward_kinematics_links_geoms()` seems to running on all of the batched envs.

e.g.
```python
def set_dofs_position(self, position, dofs_idx, envs_idx=None):
        position, dofs_idx, envs_idx = self._validate_1D_io_variables(position, dofs_idx, envs_idx)
        self._kernel_set_dofs_position(
            position,
            dofs_idx,
            envs_idx,
        )
        self._kernel_forward_kinematics_links_geoms() # <- should this be self._kernel_forward_kinematics_links_geoms(envs_idx) ??
```

### Steps to Reproduce

```python
```

### Expected Behavior

@YilingQiao could you check if this is intended? I can fix if not. 

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

[9a87bd3](https://github.com/Genesis-Embodied-AI/Genesis/commit/9a87bd3d59875bfba8c3e5fd5a955ec244c8e9d7)

YilingQiao, 2025-03-06:
yes, it should also run with `envs_idx`. Can you please implement that? Thank you!

Kashu7100, 2025-03-06:
ok, will work on it

closed 2025-03-11 by YilingQiao