Genesis-Embodied-AI/genesis-world #861: [Bug]: ValueError: Buffer model does not exist

lalalala999, 2025-03-13:
### Bug Description

Hello, I'm getting the following error when running locomotion/examples/go2_eval.py. What caused this error and how can I resolve it? 

### Steps to Reproduce

(Genesis) ylxj@ylxj:~/anaconda3/envs/Genesis/Genesis/examples/locomotion$ __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia python3 go2_eval.py 
[Genesis] [22:42:04] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [22:42:04] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [22:42:04] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [22:42:04] [INFO] Running on [NVIDIA GeForce GTX 1060] with backend gs.cuda. Device memory: 5.92 GB.
[Genesis] [22:42:04] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [22:42:06] [INFO] Scene <5e1339b> created.
[Genesis] [22:42:06] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <007d926>, morph: <gs.morphs.URDF(file='/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/assets/urdf/plane/plane.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [22:42:06] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <5012222>, morph: <gs.morphs.URDF(file='/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/assets/urdf/go2/urdf/go2.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [22:42:09] [INFO] Building scene <5e1339b>...
[Genesis] [22:42:14] [INFO] Compiling simulation kernels...
[Genesis] [22:42:21] [INFO] Building visualizer...
[Genesis] [22:42:48] [INFO] Viewer created. Resolution: 720×540, max_FPS: 25.
Actor MLP: Sequential(
  (0): Linear(in_features=45, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=12, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=45, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
)
Traceback (most recent call last):
  File "/home/ylxj/anaconda3/envs/Genesis/Genesis/examples/locomotion/go2_eval.py", line 47, in <module>
    main()
  File "/home/ylxj/anaconda3/envs/Genesis/Genesis/examples/locomotion/go2_eval.py", line 43, in main
    obs, _, rews, dones, infos = env.step(actions)
  File "/home/ylxj/anaconda3/envs/Genesis/Genesis/examples/locomotion/go2_env.py", line 125, in step
    self.scene.step()
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/utils/misc.py", line 49, in wrapper
    return method(self, *args, **kwargs)
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/engine/scene.py", line 730, in step
    self._visualizer.update(force=False)
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/vis/visualizer.py", line 141, in update
    self._viewer.update()
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/vis/viewer.py", line 132, in update
    buffer_updates = self.context.update()
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/vis/rasterizer_context.py", line 775, in update
    self.update_rigid(buffer_updates)
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/vis/rasterizer_context.py", line 308, in update_rigid
    buffer_updates[self._scene.get_buffer_id(self.rigid_nodes[geom.uid], "model")] = geom_T.transpose(
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/ext/pyrender/scene.py", line 490, in get_buffer_id
    return primitive.get_buffer_id(buffer_name)
  File "/home/ylxj/anaconda3/envs/Genesis/lib/python3.9/site-packages/genesis/ext/pyrender/primitive.py", line 310, in get_buffer_id
    raise ValueError(f"Buffer {buffer_name} does not exist")
ValueError: Buffer model does not exist

### Expected Behavior

How can I resolve this error?Thanks!

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 22.04, Windows 11 24H2]
- GPU [e.g. GTX1060]
- GPU-driver version (570.86.15)
- CUDA / CUDA-toolkit version (CUDA12.4)

### Release version or Commit ID

Genesis=0.2.1 
python=3.9

HPC-Ken, 2025-03-25:
Same issue here on examples/tutorials/visualization.py

duburcqa, 2025-03-30:
I think this issue has been fixed on the main branch of Genesis. Although it is not officially released yet, it can be installed via pip using this command:
```
pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git
```

Ping me on this issue if it is not resolved yet.

closed 2025-03-30 by duburcqa

S-Ali-Zaidi, 2025-04-02:
> I think this issue has been fixed on the main branch of Genesis. Although it is not officially released yet, it can be installed via pip using this command:
> 
> ```
> pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git
> ```
> 
> Ping me on this issue if not resolved and I will re-open.

Hi @duburcqa,

Hi there, I'm having the same issue on my linux machine (but not on my mac!) -- including after trying `pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git`

```
Building wheels for collected packages: genesis-world
  Building wheel for genesis-world (pyproject.toml) ... done
  Created wheel for genesis-world: filename=genesis_world-0.2.1-cp312-cp312-linux_x86_64.whl size=84551398 sha256=ed981c90438cca3bc231b1ce89b596bf483555619df01ac0565bf77fe8b125d9
  Stored in directory: /tmp/pip-ephem-wheel-cache-kjikqx2u/wheels/c5/c3/d6/57f983d778159744c5bcc116c76dfbf19adc16f33f8df0eb35
Successfully built genesis-world
Installing collected packages: genesis-world
  Attempting uninstall: genesis-world
    Found existing installation: genesis-world 0.2.1
    Uninstalling genesis-world-0.2.1:
      Successfully uninstalled genesis-world-0.2.1
Successfully installed genesis-world-0.2.1
(.venv) ali@TheTower:~/Projects/Genesis$ /home/ali/Projects/Genesis/.venv/bin/python /home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_eval.py
[Genesis] [20:49:40] [INFO] ╭───────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [20:49:40] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [20:49:40] [INFO] ╰───────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [20:49:40] [INFO] Running on [NVIDIA GeForce RTX 3050] with backend gs.cuda. Device memory: 5.68 GB.
[Genesis] [20:49:40] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [20:49:41] [INFO] Scene <88f49cf> created.
[Genesis] [20:49:41] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <24ac5d8>, morph: <gs.morphs.URDF(file='/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/assets/urdf/plane/plane.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [20:49:41] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <fd8c8a7>, morph: <gs.morphs.URDF(file='/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/assets/urdf/go2/urdf/go2.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [20:49:42] [INFO] Building scene <88f49cf>...
[Genesis] [20:49:43] [INFO] Compiling simulation kernels...
[Genesis] [20:49:47] [INFO] Building visualizer...
[Genesis] [20:49:58] [INFO] Viewer created. Resolution: 480×360, max_FPS: 25.
Actor MLP: Sequential(
  (0): Linear(in_features=45, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=12, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=45, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
)
Traceback (most recent call last):
  File "/home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_eval.py", line 47, in <module>
    main()
  File "/home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_eval.py", line 43, in main
    obs, _, rews, dones, infos = env.step(actions)
                                 ^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_env.py", line 125, in step
    self.scene.step()
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/utils/misc.py", line 73, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/engine/scene.py", line 723, in step
    self._visualizer.update(force=False, auto=refresh_visualizer)
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/vis/visualizer.py", line 148, in update
    self._viewer.update(auto_refresh=auto)
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/vis/viewer.py", line 126, in update
    buffer_updates = self.context.update()
                     ^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 776, in update
    self.update_rigid(buffer_updates)
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 309, in update_rigid
    buffer_updates[self._scene.get_buffer_id(self.rigid_nodes[geom.uid], "model")] = geom_T.transpose(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/ext/pyrender/scene.py", line 489, in get_buffer_id
    return primitive.get_buffer_id(buffer_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/.venv/lib/python3.12/site-packages/genesis/ext/pyrender/primitive.py", line 310, in get_buffer_id
    raise ValueError(f"Buffer {buffer_name} does not exist")
ValueError: Buffer model does not exist
[Genesis] [20:49:59] [INFO] 💤 Exiting Genesis and caching compiled kernels...
(.venv) ali@TheTower:~/Projects/Genesis$ 
```

### 🚧 Environment Info

- **OS**: Ubuntu 24.10
- **Kernel**: Linux 6.11.0-21-generic (PREEMPT_DYNAMIC)
- **GPU**: NVIDIA GeForce RTX 3050 (but using AMD Ryzen 9600x iGPU for display)
- **GPU Driver**: 570.124.04
- **Python**: 3.12.7
- **Genesis**: 0.2.1 *(installed in virtual environment `(.venv)`)*

S-Ali-Zaidi, 2025-04-06:
hi @lalalala999 and @HPC-Ken,

 I've come across a fix that allows for things to work on my system. [Ive documented everything here](https://github.com/Genesis-Embodied-AI/Genesis/issues/973#event-17136760259) on issue #973 , but I will copy paste my comment on it below. Will be curious to hear if this fixes the issue for you!

---
---

Please note that I've found a way to get the `show_viewer=True` renderer working on my machine, so long as I run a script with the command `PYOPENGL_PLATFORM=glx` prior to running it. 

This workaround was discovered by first running this script:

```python
import time
import traceback

import genesis as gs

# Try to import the Primitive class from the pyrender module.
# Adjust the import if your package structure is different.
try:
    from genesis.ext import pyrender
    import genesis.ext.pyrender.primitive as prim
except Exception as e:
    print("Could not import pyrender primitives. Ensure your Genesis package is installed in developer mode.", e)
    exit(1)

# Monkey-patch the get_buffer_id method to print debug info when "model" is missing.
original_get_buffer_id = prim.Primitive.get_buffer_id

def debug_get_buffer_id(self, buffer_name):
    if buffer_name not in self._buffers:
        print("[DEBUG] Buffer", buffer_name, "is missing for primitive with uid:",
              getattr(self, "uid", "unknown"))
        print("[DEBUG] Available buffers:", list(self._buffers.keys()))
    return original_get_buffer_id(self, buffer_name)

prim.Primitive.get_buffer_id = debug_get_buffer_id

# Initialize Genesis with your preferred settings.
gs.init(seed=0, precision="32", logging_level="debug", backend=gs.cuda)

# Create a minimal scene with the viewer enabled.
scene = gs.Scene(
    show_viewer=True,
    viewer_options=gs.options.ViewerOptions(
        res=(640, 480),
        camera_pos=(3.5, 0.0, 2.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=30,
        max_FPS=25,
    ),
    # Use Rasterizer or RayTracer as preferred.
    renderer=gs.renderers.Rasterizer(),
)

# Add a simple plane entity.
plane = scene.add_entity(gs.morphs.Plane())

# Build the scene (this compiles kernels, initializes the viewer, and registers buffers).
scene.build()

print("Scene built. Entering simulation loop to trigger viewer updates...")
print("If a missing 'model' buffer is encountered, debug info will be printed.")

# Run a simulation loop.
try:
    while True:
        scene.step()
        time.sleep(0.1)
except Exception as ex:
[... 165 lines cut ...]
OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.10.1

=== OpenGL Renderer (glxinfo) ===
OpenGL renderer string: AMD Radeon Graphics (radeonsi, raphael_mendocino, LLVM 19.1.1, DRM 3.59, 6.11.0-21-generic)

=== Xorg Version ===
Xorg version not available

=== Python Version ===
Python 3.12.7

=== PyOpenGL Version (pip freeze) ===
PyOpenGL==3.1.9

=== Taichi Version ===
[Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.12.7
Taichi version: (1, 7, 3)

=== Genesis Version ===
Genesis version: 0.2.1

=== PYOPENGL_PLATFORM Environment Variable ===
PYOPENGL_PLATFORM=
(.venv) ali@TheTower:~/Projects/Genesis$ 
```