Genesis-Embodied-AI/genesis-world #866: [Bug]:  The model from mujoco_menageri flybody/fruitfly.xml disappears unexpectedly

ale-amic1, 2025-03-14:
### Bug Description

The [fly model](https://github.com/google-deepmind/mujoco_menagerie/blob/main/flybody/fruitfly.xml) from mujoco_menagerie is added to Genesis, but after a few steps the model bends and then disappear. I would expect it to just fall to the ground, and maybe bounce on it. This video was created adding a sleep of 5 seconds after every `scene.step()` so the fly disappears quite quickly.
https://github.com/user-attachments/assets/76ba80e0-777f-49b0-a273-8d3b1b8c9375

### Steps to Reproduce

Python script for reproducing the video and logs
```
import genesis as gs
gs.init(backend=gs.cuda)

scene = gs.Scene(
        show_viewer=True,
    )
plane = scene.add_entity(gs.morphs.Plane())

fly = scene.add_entity(
    gs.morphs.MJCF(
# TODO PUT THE PATH TO YOUR MUJOCO_MENAGERIE FOLDER
        file=str("/path/to/mujoco_menagerie/flybody/fruitfly.xml"),
        pos=(0, 0.00, 0.6),
    ),
)

scene.build()
import time as t

for i in range(1000):
    scene.step()
    print(f"Fly pose: {fly.get_pos()}")
    t.sleep(5)
    

```

### Expected Behavior

The fly would just appear and drop to the ground, and maybe bounce.
Notes:
- the model contains tendons and adhesion actuators that are not supported yet
- the model contains some physics options for Mujoco
```
<option timestep="0.0001" gravity="0 0 -981" density="0.00128" viscosity="0.000185" cone="elliptic"
    noslip_iterations="3"/>
```
Without these options, the model gets crazy also in the Mujoco viewer, so maybe similar options should be set in Genesis.

### Relevant log output

```shell
[Genesis] [19:08:52] [INFO] ╭───────────────────────────────────────────────────────────────────╮
[Genesis] [19:08:52] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [19:08:52] [INFO] ╰───────────────────────────────────────────────────────────────────╯
[Genesis] [19:08:52] [INFO] Running on [Quadro T2000] with backend gs.cuda. Device memory: 3.63 GB.
[Genesis] [19:08:52] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [19:08:53] [INFO] Scene <877216a> created.
[Genesis] [19:08:53] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <98e38da>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [19:08:53] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <817eda8>, morph: <gs.morphs.MJCF(file='/home/amici/code/genesis-sim/mujoco_menagerie/flybody/fruitfly.xml')>, material: <gs.materials.Rigid>.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[Genesis] [19:08:54] [WARNING] Collision mesh in MJCF is not visualized by default. To visualize collision mesh, please use `vis_mode='collision'` when scene.add_entity.
[... 85 lines cut ...]
[Genesis] [19:09:44] [INFO] Running at 0.65 FPS.
Fly pose: tensor([ 5.8369e-05, -2.0522e-04,  5.6460e-01], device='cuda:0')
[Genesis] [19:09:49] [INFO] Running at 0.58 FPS.
Fly pose: tensor([-0.0084,  0.0108,  0.5675], device='cuda:0')
[Genesis] [19:09:54] [INFO] Running at 0.53 FPS.
Fly pose: tensor([-9.1547, -7.6279, 11.8824], device='cuda:0')
[Genesis] [19:09:59] [INFO] Running at 0.49 FPS.
Fly pose: tensor([ 37399.9492, -80907.7109,  20713.4082], device='cuda:0')
[Genesis] [19:10:04] [INFO] Running at 0.46 FPS.
Fly pose: tensor([2.4412e+09, 2.6660e+09, 2.8276e+08], device='cuda:0')
[Genesis] [19:10:09] [INFO] Running at 0.43 FPS.
Fly pose: tensor([nan, nan, nan], device='cuda:0')
[Genesis] [19:10:14] [INFO] Running at 0.41 FPS.
```

### Environment

- OS: e.g. Ubuntu 24.04
- GPU: Quadro T2000
- GPU-driver version:  570.124.06  
- CUDA / CUDA-toolkit version: 12.8 

### Release version or Commit ID

Using the main branch, commit `3be00216d94dfdd53d13163a2b9640058a8976a5`

duburcqa, 2025-03-14:
@ale-amic1 Could you check if it works better after this PR https://github.com/Genesis-Embodied-AI/Genesis/issues/866 ?

ale-amic1, 2025-03-17:
@duburcqa, the link is bringing me back to this issue 😅, could you please update the link? :)

duburcqa, 2025-03-21:
You should try on the current main branch.

ale-amic1, 2025-03-24:
Hello, I tried with the current main branch. I see different warnings about tendons and body not being supported, but I witness the same bug. The fly disappears after a few steps.
```
Genesis] [10:54:04] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [10:54:04] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [10:54:04] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [10:54:04] [INFO] Running on [NVIDIA GeForce GTX 1070] with backend gs.cuda. Device memory: 7.92 GB.
[Genesis] [10:54:05] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
length 1.0
radius 0.005
head_length 0.03
head_radius 0.01
l_ratio 0.03
r_ratio 2.0
[Genesis] [10:54:06] [INFO] Scene <a3d605b> created.
[Genesis] [10:54:06] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <85b9559>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [10:54:06] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <98f967e>, morph: <gs.morphs.MJCF(file='/home/amici/code/genesis-sim/mujoco_menagerie/flybody/fruitfly.xml')>, material: <gs.materials.Rigid>.
[Genesis] [10:54:09] [WARNING] (MJCF) Tendon not supported
[Genesis] [10:54:10] [WARNING] (MJCF) Actuator type 'mjtTrn.mjTRN_TENDON' not supported
[Genesis] [10:54:10] [WARNING] (MJCF) Actuator type 'mjtTrn.mjTRN_BODY' not supported
[Genesis] [10:54:10] [WARNING] (MJCF) Actuator type 'mjtTrn.mjTRN_BODY' not supported
[Genesis] [10:54:10] [WARNING] (MJCF) Actuator type 'mjtTrn.mjTRN_BODY' not supported
[Genesis] [10:54:10] [WARNING] Overriding base link's pos with user provided value in morph.
[Genesis] [10:54:10] [INFO] Building scene <a3d605b>...
[W 03/24/25 10:54:15.137 38729] [frontend_ir.cpp:begin_frontend_struct_for_on_external_tensor@1694] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [10:54:24] [INFO] Compiling simulation kernels...
[Genesis] [10:55:21] [INFO] Building visualizer...
[Genesis] [10:55:25] [INFO] Viewer created. Resolution: 3600×2700, max_FPS: 60.
[Genesis] [10:55:25] [INFO] Running at 62.34 FPS.
Fly pose: tensor([0.0000, 0.0000, 0.6000], device='cuda:0')
[Genesis] [10:55:31] [INFO] Running at 3.55 FPS.
Fly pose: tensor([ 4.2284e-12, -4.6430e-13,  5.9902e-01], device='cuda:0')
[Genesis] [10:55:36] [INFO] Running at 1.92 FPS.
Fly pose: tensor([ 1.4181e-11, -1.2906e-12,  5.9706e-01], device='cuda:0')
[Genesis] [10:55:41] [INFO] Running at 1.34 FPS.
Fly pose: tensor([ 3.4427e-11, -2.3717e-12,  5.9411e-01], device='cuda:0')
[Genesis] [10:55:46] [INFO] Running at 1.04 FPS.
Fly pose: tensor([ 6.0985e-11, -9.6467e-12,  5.9019e-01], device='cuda:0')
[Genesis] [10:55:51] [INFO] Running at 0.86 FPS.
Fly pose: tensor([4.0791e-11, 4.4820e-10, 5.8529e-01], device='cuda:0')
[Genesis] [10:55:56] [INFO] Running at 0.74 FPS.
Fly pose: tensor([ 3.9965e-09, -3.5781e-08,  5.7940e-01], device='cuda:0')
[Genesis] [10:56:01] [INFO] Running at 0.65 FPS.
Fly pose: tensor([-3.0252e-07,  2.8204e-06,  5.7253e-01], device='cuda:0')
[Genesis] [10:56:06] [INFO] Running at 0.58 FPS.
Fly pose: tensor([ 3.9608e-05, -2.1110e-04,  5.6463e-01], device='cuda:0')
[Genesis] [10:56:11] [INFO] Running at 0.53 FPS.
Fly pose: tensor([-0.0074, -0.0033,  0.5623], device='cuda:0')
[Genesis] [10:56:16] [INFO] Running at 0.49 FPS.
Fly pose: tensor([ 7.9986, -3.1435, -9.1547], device='cuda:0')
[Genesis] [10:56:21] [INFO] Running at 0.46 FPS.
Fly pose: tensor([-21033.3887, -44446.2969,  37011.3711], device='cuda:0')
```

duburcqa, 2025-03-24:
Ok thank you ! In that case I will have a look asap

duburcqa, 2025-03-27:
Probleme solved ! Just need to set the simulation to what is prescribed in the Mujoco MJCF file. For now Genesis is not handling this automatically.
```
scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            dt=0.0001,
        ),
        show_viewer=True,
    )
```

closed 2025-03-27 by duburcqa