Genesis-Embodied-AI/genesis-world #877: [Bug]: Multi-bodies with container collision simulation not stable.

RunqiuBao, 2025-03-17:
### Bug Description

## issue description
- when throwing multiple bodies into a container (The container is convexified by coacd), it often ends up the bodies pile will explode and show unstable movement.
 
## exampe:
- Simulation start: multiple bodies do free fall into a container. <br>
  [image]
- Simulation end: the bodies pile exploded. <br>
  [image]

## my environment:
- debian bullseye docker container.
- cpu: `12th Gen Intel® Core™ i7-12700 × 20`
- gpu: `NVIDIA GeForce RTX 3060`
- genesis version: 
```
commit 98b29b4a4e56bdaa55ca1d8a359c5935eb24f26c (origin/main, origin/HEAD, main)
Author: Alexis DUBURCQ <alexis.duburcq@gmail.com>
Date:   Mon Mar 3 09:18:13 2025 +0100

    Update 'IS_OPENGL_42_AVAILABLE' condition to stick on Public API. (#814)
```

### Steps to Reproduce

## how to reproduce:
- download the zip in a machine with genesis installed, and run `python3 reproduce_unstable_sim.py`.

[genesis_issue.zip](https://github.com/user-attachments/files/19280161/genesis_issue.zip)

### Expected Behavior

- when throwing multiple bodies into a container (The container is convexified by coacd), it often ends up the bodies pile will explode and show unstable movement.

### Relevant log output

```shell

```

### Environment

- debian bullseye docker container.
- cpu: `12th Gen Intel® Core™ i7-12700 × 20`
- gpu: `NVIDIA GeForce RTX 3060`

### Release version or Commit ID

- genesis version: 
```
commit 98b29b4a4e56bdaa55ca1d8a359c5935eb24f26c (origin/main, origin/HEAD, main)
Author: Alexis DUBURCQ <alexis.duburcq@gmail.com>
Date:   Mon Mar 3 09:18:13 2025 +0100

    Update 'IS_OPENGL_42_AVAILABLE' condition to stick on Public API. (#814)
```

YilingQiao, 2025-03-17:
There are many rigid bodies stacking together, making the scene quite sensitive to simulation. Can you try this configuration? if that's still not stable, we can look for other solutions.
```

genesis.init(backend=genesis.cpu, precision="64")
# renderer = genesis.renderers.RayTracer(
#     env_surface=genesis.surfaces.Emission(
#         emissive_texture=genesis.textures.ImageTexture(
#             image_path=join(dirname(genesis.__file__), "assets/textures/indoor_bright.png"),
#         ),
#     ),
# )

scene = genesis.Scene(
    show_viewer=True,
    vis_options = genesis.options.VisOptions(
        show_world_frame = False,
        segmentation_level='entity',
        show_link_frame  = False,
        show_cameras     = True,
        plane_reflection = True,
        shadow = False,
    ),
    rigid_options=genesis.options.RigidOptions(
        use_contact_island=True,  # Note: for multiple-contact
        constraint_solver=gs.constraint_solver.Newton,
        tolerance=1e-8,
        iterations=1000,
        ls_iterations=500,
        max_collision_pairs=10000,
        
    ),
    sim_options=genesis.options.SimOptions(
        dt=1e-2,
        substeps=32,
    ),
    # renderer=renderer,
)

```

YilingQiao, 2025-03-17:
If that's too conservative and slow, you might tune those parameters a bit.

RunqiuBao, 2025-03-17:
Thanks, it looks much better than using the default configuration.

Btw, I notice that in some cases some bodies will suddenly disappear during the simulation. Is it a known issue?

YilingQiao, 2025-03-19:
> Thanks, it now looks much better than using the default configuration.
> 
> Only one problem, I notice that in some cases some bodies will suddenly disappear during the simulation. Is it a known issue? It does not reproduce every time though, occurs quite random
> 
>  simplescreenrecorder-2025-03-17_18.34.44.mp4

The object might have flown away due to issues during the substeps. We could tune more parameters to make the simulation more stable if needed.

closed 2025-03-19 by YilingQiao