Genesis-Embodied-AI/genesis-world #883: [Bug]: World link/joint is included in links/joints of rigid entities

bxtbold, 2025-03-18:
### Bug Description

While loading robot mjcf files, world link and joint are added to `rigid_entity._links` and `rigid_entity._joints`.

I have tracked back and found out that the issue started to appear after refactoring mjcf parsing in [this commit](https://github.com/Genesis-Embodied-AI/Genesis/commit/284fead4df750620ed452c54efee3c4885520a79).

### Steps to Reproduce

```python
import genesis as gs
gs.init(backend=gs.cpu)
scene = gs.Scene(show_viewer=False)
plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
)
scene.build()
print([link.name for link in franka.links])
```

### Expected Behavior

Expected:
`['link0', 'link1', 'link2', 'link3', 'link4', 'link5', 'link6', 'link7', 'hand', 'left_finger', 'right_finger']`

Current:
`['link0', 'link1', 'link2', 'link3', 'link4', 'link5', 'link6', 'link7', 'hand', 'left_finger', 'right_finger', 'world']`

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 22.04
- GPU/CPU: N/A
- GPU-driver version: N/A
- CUDA / CUDA-toolkit version: N/A

### Release version or Commit ID

049c95d4168afe71ba3e6f058ce9965ab24e4abf

duburcqa, 2025-03-18:
I thought it was supposed to be the case, since it is kind of a convention to exclude world. Is this a problem? If so I can fix it!

duburcqa, 2025-03-18:
Ok previously the world was only added if it was a geometry, but it was still added in this particular case. Could you confirm this? I don't know if this design choice is any better, but I can restore it.

duburcqa, 2025-03-19:
I do not think you can specify the pose of the world itself in Mujoco. It is by definition the parent of all bodies and is stuck at the origin of the entire universe. There is no `<body/>` tag associated with it. Still you can attach geometries to it with their own relative pose.