Genesis-Embodied-AI/genesis-world #885: [Bug]: Object Collision: the ball fall and pass through a tank

Leung555, 2025-03-18:
### Bug Description

There is an error running the example script in rigid/nonconvex_mesh.py
- the ball fall and pass through the tank (bath tube shape)

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
    tank = scene.add_entity(
        gs.morphs.Mesh(
            file="meshes/tank.obj",
            convexify=False,
            scale=5.0,
            fixed=True,
            euler=(90, 0, 0),
        ),
    )
    ball = scene.add_entity(
        gs.morphs.Sphere(
            radius=0.1,
            pos=(0.0, 0.0, 1.0),
        ),
    )

    scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

the ball should collide with the tank

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04,]
- GPU/CPU [e.g. RTX 4060]

### Release version or Commit ID

v0.2.1

duburcqa, 2025-03-18:
Which version of Genesis are you using ? The latest release available on pypi ?

Leung555, 2025-03-19:
> Which version of Genesis are you using ? The latest release available on pypi ?

Yes, I also just try to reinstall everything separately from user guide and github and still not work. Is this problem occur in the others release? I also try decompose the nonconvex part also with coad_option but still not work for me.

duburcqa, 2025-03-19:
Ok, I will try on main branch then!

duburcqa, 2025-03-19:
Oh by the way, did you try on CPU ? GPU is not working on MacOS.

Leung555, 2025-03-19:
the ball still passtrhough when try with both cpu and gpu

duburcqa, 2025-03-19:
I can reproduce the error, so I can fix it.

duburcqa, 2025-03-19:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/888!

closed 2025-03-19 by duburcqa

wangyian-me, 2025-03-19:
I think this is a use case that fast_simplification doesn't work as well as previous simplification method. (fast_simplification  in commit 963687e).