Genesis-Embodied-AI/genesis-world #890: [Bug]: License is wrong

cielavenir, 2025-03-19:
### Bug Description

Genesis depends on tetgen, which is distributed under AGPL. Hence Genesis license should be AGPL as well, or make tetgen dependency optional at least.

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.

### Expected Behavior

Either Genesis license should be AGPL as well or tetgen dependency should be optional.

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

master

cielavenir, 2025-03-19:
umm ok, it seems we can buy tetgen license: https://wias-berlin.de/software/tetgen/FAQ-license.jsp

I can close this, but the history will remain so that people can reference later

closed 2025-03-19 by cielavenir

duburcqa, 2026-11-10:
@sarooshsh I will investigate this issue.

duburcqa, 2026-11-10:
The plan is to write our own custom algorithm to get rid of tetgen dependency entirely.