Genesis-Embodied-AI/genesis-world #893: [Bug]: Primitive cube oscillate when stacking on top of another cube

Leung555, 2025-03-20:
### Bug Description

Primitive cube oscillate when stacking on top of another primitive cube.

I create a scene based on the IK example to let franka arm stack the cube on top of another cube.

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
import numpy as np
import genesis as gs

########################## init ##########################
gs.init(backend=gs.gpu, precision="32")

########################## create a scene ##########################
scene = gs.Scene(
    viewer_options = gs.options.ViewerOptions(
        camera_pos    = (3, -1, 1.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 30,
        max_FPS       = 60,
    ),
    sim_options=gs.options.SimOptions(
        dt=0.01,
    ),
    show_viewer = True,
)

########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)
cube = scene.add_entity(
    gs.morphs.Box(
        size = (0.04, 0.04, 0.04),
        pos  = (0.65, 0.0, 0.02),
    ),
    surface=gs.surfaces.Plastic(color=(1,0,0)),
)
cube_2 = scene.add_entity(
    gs.morphs.Box(
        size = (0.04, 0.04, 0.04),
        pos  = (0.4, 0.2, 0.02),
    ),
    surface=gs.surfaces.Plastic(color=(0,1,0)),
)
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)
########################## build ##########################
scene.build()

motors_dof = np.arange(7)
fingers_dof = np.arange(7, 9)

# set control gains
# Note: the following values are tuned for achieving best behavior with Franka
# Typically, each new robot would have a different set of parameters.
# Sometimes high-quality URDF or XML file would also provide this and will be parsed.
franka.set_dofs_kp(
    np.array([4500, 4500, 3500, 3500, 2000, 2000, 2000, 100, 100]),
)
franka.set_dofs_kv(
    np.array([450, 450, 350, 350, 200, 200, 200, 10, 10]),
)
franka.set_dofs_force_range(
    np.array([-87, -87, -87, -87, -12, -12, -12, -100, -100]),
    np.array([ 87,  87,  87,  87,  12,  12,  12,  100,  100]),
)

# get the end-effector link
end_effector = franka.get_link('hand')

# move to pre-grasp pose
qpos = franka.inverse_kinematics(
    link = end_effector,
    pos  = np.array([0.65, 0.0, 0.25]),
    quat = np.array([0, 1, 0, 0]),
)
# gripper open pos
qpos[-2:] = 0.04
path = franka.plan_path(
    qpos_goal     = qpos,
    num_waypoints = 200, # 2s duration
)
# execute the planned path
for waypoint in path:
    franka.control_dofs_position(waypoint)
    scene.step()

# allow robot to reach the last waypoint
for i in range(100):
    scene.step()

# reach
qpos = franka.inverse_kinematics(
    link = end_effector,
    pos  = np.array([0.65, 0.0, 0.130]),
    quat = np.array([0, 1, 0, 0]),
)
franka.control_dofs_position(qpos[:-2], motors_dof)
for i in range(100):
    scene.step()

# grasp
franka.control_dofs_position(qpos[:-2], motors_dof)
franka.control_dofs_force(np.array([-0.5, -0.5]), fingers_dof)

for i in range(100):
    scene.step()

# lift
qpos = franka.inverse_kinematics(
    link=end_effector,
    pos=np.array([0.65, 0.0, 0.28]),
    quat=np.array([0, 1, 0, 0]),
)
franka.control_dofs_position(qpos[:-2], motors_dof)
for i in range(200):
    scene.step()

# reach
qpos = franka.inverse_kinematics(
    link = end_effector,
    pos  = np.array([0.4, 0.2, 0.18]),
    quat = np.array([0, 1, 0, 0]),
)
franka.control_dofs_position(qpos[:-2], motors_dof)
for i in range(200):
    scene.step()

# release
franka.control_dofs_position(np.array([0.4, 0.4]), fingers_dof)

for i in range(1000):
    scene.step()```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

the cube should be stable when contact with another cube

### Screenshots/Videos

![Image](https://github.com/user-attachments/assets/0fd6cfca-b16a-466b-99ad-338038320c10)

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04,]
- GPU [e.g. RTX 4060)
- GPU-driver version (550)
- CUDA / CUDA-toolkit version (12.4)

### Release version or Commit ID

genesis v0.2.1

wangyian-me, 2025-03-20:
Might want to try adding `box_box_detection=True` to rigid solver options.

duburcqa, 2025-03-20:
Could you try on main branch ? I think it should be more stable. And yes, enabling `box_box_detection =True` may also help.

duburcqa, 2025-03-21:
I can confirm it works fine on main branch with `box_box_detection=True`. I will see why it is failing so badly without box-box collision detection (I cannot even catch the red box).

closed 2025-03-21 by duburcqa