Genesis-Embodied-AI/genesis-world #903: [Bug]: 如何在fem中定义新材料

lishilei-cmd, 2025-03-21:
### Bug Description

我在FEM中想仿照elastic的参数来定义混凝土材料，修改为E=3.45e9,  nu=0.2,  rho=2400.0,然后创建一个最简单的box实体，引用FEM.Concrete材料，运行能看到画面，但看不到box实体。然而我调小为E=3.45e6,能看到box实体一瞬间就消失，继续调小为E=1e6,就能一直看到Box实体没有问题。

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
我定义的混凝土材料，采用弹性
import taichi as ti
import genesis as gs
from .base import Base
@ti.data_oriented
class Concrete(Base):
    def __init__(
        self,
        E=1e6,  # Young's modulus
        nu=0.2,  # Poisson's ratio
        rho=2400.0,  # density (kg/m^3)
        model="linear",
    ):
        super().__init__(E, nu, rho)
        if model == "linear":
            self.update_stress = self.update_stress_linear
        elif model == "stable_neohooken":
            self.update_stress = self.update_stress_stable_neohooken
        else:
            gs.raise_exception(f"Unrecognized constitutive model: {model}")
        self._model = model
    @ti.func
    def update_stress_linear(self, mu, lam, J, F, actu, m_dir):
        I = ti.Matrix.identity(dt=gs.ti_float, n=3)
        stress = mu * (F + F.transpose() - 2 * I) + lam * (F - I).trace() * I
        return stress
    @ti.func
    def update_stress_stable_neohooken(self, mu, lam, J, F, actu, m_dir):
        IC = (F.transpose() @ F).trace()
        dJdF0 = F[:, 1].cross(F[:, 2])
        dJdF1 = F[:, 2].cross(F[:, 0])
        dJdF2 = F[:, 0].cross(F[:, 1])
        dJdF = ti.Matrix.cols([dJdF0, dJdF1, dJdF2])
        alpha = 1 + 0.75 * mu / lam
        stress = mu * (1 - 1 / (IC + 1)) * F + lam * (J - alpha) * dJdF
        return stress
    @property
    def model(self):
        return self._model
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

希望能实现混凝土的模拟功能，采用线弹性，进行有限元分析

### Relevant log output

```shell

```

### Environment

- OS: Windows 11 24H2]
- GPU/CPU [ RTX 4090,  Intel I9-14900kf,  (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

genesis version-0.2.1

wangyian-me, 2025-03-21:
Can you provide more information? I think it's because your substep_dt is too large and also your precision is better to be 64.

lishilei-cmd, 2025-03-22:
import genesis as gs
import numpy as np

gs.init(backend=gs.cuda)
scene = gs.Scene( 
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(3.5, 0.0, 2.5),  # 摄像机位置
        camera_lookat=(0.0, 0.0, 0.5),  # 摄像机焦点
        camera_fov=100,  # 摄像机视野
        max_FPS=60,
        ),
        show_viewer=True,  # 是否显示可视化窗口
    vis_options=gs.options.VisOptions(
        visualize_mpm_boundary=True,
    ),
    fem_options=gs.options.FEMOptions(
        dt=0.01,
        # gravity=9.8
    )
)
plane = scene.add_entity(
    gs.morphs.Plane(),
)
bridge_1 = scene.add_entity(
    gs.morphs.Box(
        size=(1, 1, 1),  # XYZ方向尺寸
        pos=(0, 0, 1),   # 中心位置（左端在x=0）
        euler=(0, 0, 0),    # 初始旋转角度
    ),
    material=gs.materials.FEM.Concrete(),
)
scene.build()
for i in range(1000):
    scene.step()
这是我的运行代码，其中我在FEM下仿照elastic设置了混凝土的材料特性（仅修改了E,nu,rho，通常在有限元软件中也采用弹性来代表混凝土），然后运行一个简单的box实体，

https://github.com/user-attachments/assets/1dd204f0-41ed-47f9-bac9-9730167250ed
就像视频画面中（设置的是E=3e6）一样实体一闪就消失，实际中E=3.45e9时，运行能看到画面，但看不到box实体。当调小为E=1e6，就能一直看到Box实体没有问题。我的环境是win11系统，i9-14900kf，rtx4090。

lishilei-cmd, 2025-03-22:
感谢您的回复，我的电脑环境是win11系统，i9-14900kf，rtx4090。

附件shujutiqu是我的代码运行情况，我希望的是通过genesis来模拟混凝土的一些情况做有限元分析，并与目前工程上主流的有限元软件进行对比。因此我在FEM下仿照elastic创建了适用于混凝土的材料参数（仅修改了E，nu，rho），附件concrete中创建的混凝土材料，运行时得到的画面就是，box实体一闪就消失了，然后将参数调小为E=1e6，就能一直看到Box实体没有问题。
非常感谢你的回复，以上就是我的一些情况，如果还需要任何信息，我都可以提供，只希望能帮我解决这个问题，共同促进genesis的发展，十分感谢您！

wangyian-me, 2025-03-22:
dt太大了吧，我觉得就算E=1e6这个dt也有点太大了。试试用dt=0.0001或更小。然后precision用64位的gs.init(backend=gs.cuda, precision="64")试一下？

lishilei-cmd, 2025-03-23:
对的，把dt调小以后就解决问题了，非常感谢
但现在我想提取box的有关数据，当不定义材料属性能正常提取速度数据，一旦定义为FEM材料时就遇到了一个新的问题：AttributeError: 'FEMEntity' object has no attribute 'get_vel'. Did you mean: 'set_vel'?  不仅是使用FEM材料时这样，使用MPM材料也遇到相同的问题：AttributeError: 'MPMEntity' object has no attribute 'get_vel'. Did you mean: 'set_vel'?  修改为set_vel也解决不了。是否有什么方法能在使用FEM或MPM材料情况下，可以提取实体数据的？

wangyian-me, 2025-03-23:
这是另外一个问题了，可以新开一个issue。(我觉得这个api在fem应该是get_state，如果不好使的话就新开一个issue吧)

closed 2025-03-23 by wangyian-me