Genesis-Embodied-AI/genesis-world #930: [Bug]: Loading MuJoCo model when Genesis installed via pip install genesis-world

DMackRus, 2025-03-26:
### Bug Description

When I install genesis via `pip install genesis-world` the following script does not visualize an acrobot model in the viewer.

I tried installing the most recent version using the following commands from the README

```
git clone https://github.com/Genesis-Embodied-AI/Genesis.git
cd Genesis
pip install -e .
```

But when I use these commands, it does not install genesis for me? I.e. I cant run any genesis related things. In case it is relevant, I am in a virtual python environment.

My current fix is I use `pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git` to install genesis, then my script works and I can view the acrobot model correctly.

Will there be a new release to genesis soon so that when it is installed via pip the issue is fixed, as it works with the latest version of Github?

Am I doing something wrong trying to install the latest version with the given commands? Do they not work when using a python virtual environment?

### Steps to Reproduce

```
import genesis as gs

gs.init()

# Gensis init and scene creation
scene = gs.Scene(
    show_viewer    = True,
    viewer_options = gs.options.ViewerOptions(
        camera_pos    = (3.5, -1.0, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 40,
    ),
    rigid_options = gs.options.RigidOptions(
        dt                = 0.01,
    ),
)

# Scene entities
plane = scene.add_entity(
    gs.morphs.Plane(),
)

pendulum = scene.add_entity(
    gs.morphs.MJCF(file='acrobot.xml'),
)

N = 25
scene.build(n_envs=N, env_spacing=(1.0, 1.0))

for i in range(1000):
        scene.step()
```

-------------------------------------------------- MuJoCo xml ---------------------------------------------------
```
<mujoco model="acrobot">
    <asset>
        <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="300" height="300" mark="edge" markrgb=".2 .3 .4"/>
        <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
        <material name="self" rgba=".7 .5 .3 1"/>
        <material name="self_default" rgba=".7 .5 .3 1"/>
        <material name="self_highlight" rgba="0 .5 .3 1"/>
        <material name="effector" rgba=".7 .4 .2 1"/>
        <material name="effector_default" rgba=".7 .4 .2 1"/>
        <material name="effector_highlight" rgba="0 .5 .3 1"/>
        <material name="decoration" rgba=".3 .5 .7 1"/>
        <material name="eye" rgba="0 .2 1 1"/>
        <material name="target" rgba=".6 .3 .3 1"/>
        <material name="target_default" rgba=".6 .3 .3 1"/>
        <material name="target_highlight" rgba=".6 .3 .3 .4"/>
        <material name="site" rgba=".5 .5 .5 .3"/>
    </asset>

    <default>
        <joint damping=".05"/>
        <geom type="capsule" mass="1"/>
    </default>

    <worldbody>
        <body name="upper_arm" pos="0 0 2.2">
            <joint name="shoulder" type="hinge" axis="0 1 0"/>
            <geom name="upper_arm_decoration" material="decoration" type="cylinder" fromto="0 -.06 0 0 .06 0" size="0.051" mass="0.1"/> <!-- material="decoration" -->
            <geom name="upper_arm" fromto="0 0 0 0 0 1" size="0.05" material="self"/>
            <body name="lower_arm" pos="0 0 1">
                <joint name="elbow" type="hinge" axis="0 1 0"/>
                <geom name="lower_arm" fromto="0 0 0 0 0 1" size="0.049" material="self"/>
            </body>
        </body>
    </worldbody>

    <actuator>
        <motor name="shoulder" joint="shoulder" gear="100" ctrllimited="true" ctrlrange="-1 1"/>
    </actuator>
</mujoco>
```

### Expected Behavior

I want the script I gave to load and visualize an acrobot model in the viewer when genesis is installed via `pip install genesis-world`

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

Version 0.2.1

Kashu7100, 2025-03-26:
Please refer to https://github.com/Genesis-Embodied-AI/Genesis/discussions/934. Hopefully, we can have better releasing cycle sooner.

duburcqa, 2025-03-27:
You need to uninstall `genesis-world` before doing `pip install -e .`, otherwise everything gets mixed up :/

closed 2025-03-27 by duburcqa