Genesis-Embodied-AI/genesis-world #939: [Feature]:

yomingjayism, 2025-03-27:
### What feature or enhancement are you proposing?

Is it possible we can have an API which can query/set the position/quaternion of a group of entities simultaneously?

### Motivation

As in a scene that frequently interact with many entities,
I want to have an API that can query and set the position/quaternion of multiple entities at same time instead of doing it in a for loop,
Because this would be more efficient.

### Potential Benefit

Speed up the query/set process if we have many entites in the scene, and we need to access them in every step.

### What is the expected outcome of the implementation work?

This API can not only get the position/quaternion of multiple entities, but also set the position/quaternion of multiple entities.

### Additional information

It would be similar to the API XformPrimView in omniverse isaac sim.
Reference: https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#xform-prim-view

duburcqa, 2025-03-27:
You can do this at solver level, i.e. `scene.sim.rigid_solver.get_*`. On main branch, the API has been unified because entity and solver-level.

duburcqa, 2025-03-27:
> all_qpos = solver_states.qpos.view(num_env, -1, 7)

This is not recommended, you should use `set_qpos` instead, and apply mask (first argument) to get only the entities you want to update. Note that specifying a slice for the `get_qpos` would be slightly more efficient, but it is the other way around for `set_qpos` (tricky).

yomingjayism, 2025-03-27:
Ah ha, thanks for the advice! I'll try using set_qpos with masks(qs_idx?) instead.

duburcqa, 2025-03-27:
yes!

closed 2025-03-28 by yomingjayism