Genesis-Embodied-AI/genesis-world #943: [Bug]: high friction make cube jump

milong26, 2025-03-27:
### Bug Description

set friction=5 to cube material then it jumps a liitle in video

### Steps to Reproduce

Code here: 
Using all assets from Genesis

```python
import genesis as gs
import numpy as np

########################## init ##########################
gs.init(backend=gs.gpu)

########################## create a scene ##########################
scene = gs.Scene(
    # viewer_options=gs.options.ViewerOptions(
    #     camera_pos=(2, 0, 1),
    #     camera_lookat=(0.06, 0.0, 0.5),
    #     camera_fov=30,
    #     max_FPS=60,
    # ),
    sim_options=gs.options.SimOptions(
        dt=0.01,
        substeps=4
    ),
    show_viewer=False,
)
########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)
cube = scene.add_entity(
    gs.morphs.Box(
        size=(0.4, 0.4, 0.4),
        pos=(0.8, 0.0, 0.2),
    ),
    material=gs.materials.Rigid(friction=5),
)
cube2 = scene.add_entity(
    gs.morphs.Box(
        size=(0.4, 0.4, 0.4),
        pos=(0.2, 0, 0.2),
    ),
    material=gs.materials.Rigid(friction=1),
)
cam = scene.add_camera(
    res=(640, 480),
    pos=(0.5, 2.5, 0.2),
    lookat=(0.5, 0, 0.2),
    fov=30,
    GUI=False,
)
scene.build()
cam.start_recording()
for i in range(200):
    cam.render()
    scene.step()
cam.stop_recording(save_to_filename="video1_sink.mp4", fps=60)
```

### Expected Behavior

Keep it static

### Screenshots/Videos

https://github.com/user-attachments/assets/09887531-cc0f-4cce-ba77-93e5e79c3151

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 18.04, link server via SSH 
- pytorch  pip3 install torch torchvision torchaudio --index-url https://download.pytorch.org/whl/cu118
- python==3.10 

Tell me if need more

### Release version or Commit ID

[da74990](https://github.com/Genesis-Embodied-AI/Genesis/commit/da74990b2cec90256db70bc38761dbbf491e2bb2)

YilingQiao, 2025-03-27:
It is kined expected that we are folling Mujoco's [soft contact model](https://mujoco.readthedocs.io/en/stable/computation/index.html#soft-contact-model). Large friction would make the contract softer. Does this cause bugs or make the simulation explode?

milong26, 2025-03-28:
> It is kined expected that we are following Mujoco's [soft contact model](https://mujoco.readthedocs.io/en/stable/computation/index.html#soft-contact-model). Large friction would make the contract softer. Does this cause bugs or make the simulation explode?

It hasm't made very big problem to my research, I'm just curious whether it changed the properties of cude

closed 2025-05-11 by duburcqa