Genesis-Embodied-AI/genesis-world #949: [Bug]: Not possible to destroy scene without memory leak

RunqiuBao, 2025-03-28:
### Bug Description

When simulation finished, it is not possible to destroy the scene and release all memories safely.

### Steps to Reproduce

- Before running the script, check memory usage by running `htop` in another terminal.
 - Run this script (`python3 report_issue_deleteScene.py`), when scene stopped building, check the memory usage again.
 - Press Enter to delete objects and do `genesis._gc_exit()`. Check memory usage. you will notice that not all the memories are released.
 - Now if you go `genesis_gc_exit()` again, there will be a segmentation fault happening.

script:
[report_issue_deleteScene.zip](https://github.com/user-attachments/files/19499244/report_issue_deleteScene.zip)

### Expected Behavior

- Before running script, my memory usage:
  - [image]
- After scene building, my memory usage:
  -  [image]
- After deleting objects and do  `genesis._gc_exit()`, my memory usage:
  -  [image]
 - Seg-fault if doing `genesis._gc_exit()` twice:
  -  [image]

### Relevant log output

```shell

```

### Environment

- OS: Debian 12
- CPU 12th Gen Intel® Core™ i7-12700

### Release version or Commit ID

98b29b4a4e56bdaa55ca1d8a359c5935eb24f26c

duburcqa, 2025-03-28:
`genesis._gc_exit()` is not supposed to be called manually (hence the `_` prefix). You should use `genesis.destroy` instead. Let me know if you still have any issue.

RunqiuBao, 2025-03-28:
@duburcqa The reason I had to use `_gc_exit` is because `destroy` does not really do anything:

[image]

duburcqa, 2025-03-28:
It is supposed to do something the first time it is called. But it does not do anything, then it is a bug and it must be fixed.

RunqiuBao, 2025-03-28:
I see. I will also look into this issue at my side

duburcqa, 2025-04-09:
I can confirm memory is not released. I think this is related to some circular reference issue. I will check if I cannot do something about it.

duburcqa, 2025-04-20:
I can confirm memory is not properly released on Linux, for this specific snippet:
```python
import genesis as gs
   ...: 
while True:
    gs.init(backend=gs.cpu, precision="64", logging_level='debug', debug=False)
   ...: 
    scene = gs.Scene(
        show_viewer=False,
        show_FPS=False
    )
    franka = scene.add_entity(
        gs.morphs.MJCF(
            file='xml/franka_emika_panda/panda.xml',
        ),
    )
    scene.build(n_envs=10000)
    for i in range(1000):
        scene.step()
    gs.destroy()
    import gc
    gc.collect()
```

Ping me if there is still a memory leak on your side and I will re-open.

closed 2025-04-20 by duburcqa