Genesis-Embodied-AI/genesis-world #951: [Bug]: PartNet-Mobility Dataset object don't have collision

milong26, 2025-03-28:
### Bug Description

I tried to use object model from [PartNet-Mobility Dataset](https://sapien.ucsd.edu/browse) then load its `mobility.urdf` to file. Then the object sink into ground . No collsion with plane.

### Steps to Reproduce

```python
import genesis as gs
import numpy as np

########################## init ##########################
gs.init(backend=gs.gpu)

########################## create a scene ##########################
scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt=0.01,
        # substeps=4,
        # gravity=(0, 0, -10.0),
    ),
    show_viewer=False,
)
plane = scene.add_entity(
    gs.morphs.Plane(),
)
obj=scene.add_entity(
    gs.morphs.Mesh(
        file="../assets/bottle/mobility.urdf",
    ),
)

cam = scene.add_camera(
    res=(640, 480),
    pos=(-1.5, -1.5, 1.5),
    lookat=(0, 0, 0.4),
    fov=30,
    GUI=False,
)
scene.build()
cam.start_recording()
for i in range(100):
    cam.render()
    scene.step()

cam.stop_recording(save_to_filename="video2.mp4", fps=60)
```

[bottle.zip](https://github.com/user-attachments/files/19507399/bottle.zip)

### Expected Behavior

The object has right collision

### Screenshots/Videos

### Relevant log output

```shell

```

### Environment

Ubuntu 18.04

### Release version or Commit ID

e3de23a (HEAD -> main, origin/main, origin/HEAD) Fix terrain generation. (#945)

[e3de23a](https://github.com/Genesis-Embodied-AI/Genesis/commit/e3de23a451cdf81af9e1e9be64bf7925cb52c8bd)

duburcqa, 2025-03-28:
On it.

duburcqa, 2025-03-28:
Which version of Genesis are you using ?

duburcqa, 2025-03-28:
Can you post a direct link to the asset so that I could reproduce the issue.

By the way, you could try adding `vis_mode="collision"` argument to `scene.add_entity`.

duburcqa, 2025-03-28:
I can reproduce the issue.

closed 2025-03-28 by duburcqa