Genesis-Embodied-AI/genesis-world #955: [Bug]: No texture for glb model.

Riften, 2025-03-29:
### Bug Description

No texture when loading glb file.
(glb file is not allowed in attachment. I put the test model in compressed file)

[table0001.glb.zip](https://github.com/user-attachments/files/19521488/table0001.glb.zip)

### Steps to Reproduce

```python
import genesis as gs
test_glb = 'table0001.glb'
gs.init(backend=gs.cuda)
scene = gs.Scene(
    show_viewer    = True,
)
glb_entity = scene.add_entity(
    gs.morphs.Mesh(
        file=test_glb,
        fixed=True,  # Make it static
    )
)
scene.build()
```

### Expected Behavior

Result from [online glb viewer](https://gltf-viewer.donmccurdy.com/)

[image]

### Screenshots/Videos

Result from genesis.

[image]

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 20.04

### Release version or Commit ID

commit f1ee64b590d8dad30b0805d575086b9e2fbb2ba8

Riften, 2025-03-29:
The model is from [HSSD dataset](https://huggingface.co/datasets/hssd/hssd-hab).

Riften, 2025-03-30:
> Do you know if it was working at some point ?

No. This is my first time trying to load glb model into genesis.

duburcqa, 2025-03-31:
After checking and spending some time trying to figure out what was the issue and how to solve it, it turns out the texture are compressed images of format ktx2. There is no Python parser available for this format at the moment when supercompression is enabled, which is basically the whole point of using it. Unfortunately nothing that be done at the moment to fix this issue, except logging a warning to inform the user of the issue.

closed 2025-05-24 by duburcqa