Genesis-Embodied-AI/genesis-world #968: [Bug]: Complex mesh explodes or falls apart during cloth simulation with PBD

GaoLF, 2025-04-02:
### Bug Description

[image]

[image]

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
    cloth_1 = scene.add_entity(
        material=gs.materials.PBD.Cloth(),
        morph=gs.morphs.Mesh(
            file="meshes/Cloth1.obj",
            pos=(0, 0.5, 0.3),
            euler=(-90, 0,-90.0),
            collision = True,
        ),
        surface=gs.surfaces.Default(
            color=(0.8, 0.4, 0.8, 1.0),
            #vis_mode="particle",
            vis_mode="visual",
        ),
    )
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

使用不同的mesh obj，能够有类似的物理效果

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

master branch

duburcqa, 2025-04-02:
Could you share the assets? Can you specify the exact Genesis commit that you are using?

duburcqa, 2025-05-22:
No reproduction script was provided. Closing.

closed 2025-05-22 by duburcqa