Genesis-Embodied-AI/genesis-world #973: [Bug]: `ValueError: Buffer model does not exist` when ShowViewer=True

S-Ali-Zaidi, 2025-04-04:
### Bug Description

After successfully running the go2_train.py locomotion example script, when I try to run go2_eval.py, i get the following error. This is the same error I get when I try to run virtually any script that activates the viewer. 

```
(.venv) ali@TheTower:~/Projects/Genesis$ /home/ali/Projects/Genesis/.venv/bin/python /home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_eval.py
[Genesis] [14:19:56] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [14:19:56] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [14:19:56] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [14:19:56] [INFO] Running on [NVIDIA GeForce RTX 3050] with backend gs.cuda. Device memory: 5.69 GB.
[Genesis] [14:19:56] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [14:19:57] [INFO] Scene <5193fe8> created.
[Genesis] [14:19:57] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <f2454eb>, morph: <gs.morphs.URDF(file='/home/ali/Projects/Genesis/Genesis/genesis/assets/urdf/plane/plane.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [14:19:57] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <6db62f4>, morph: <gs.morphs.URDF(file='/home/ali/Projects/Genesis/Genesis/genesis/assets/urdf/go2/urdf/go2.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [14:19:57] [INFO] Building scene <5193fe8>...
[Genesis] [14:20:00] [INFO] Compiling simulation kernels...
[Genesis] [14:20:21] [INFO] Building visualizer...
[Genesis] [14:20:32] [INFO] Viewer created. Resolution: 480×360, max_FPS: 25.
Actor MLP: Sequential(
  (0): Linear(in_features=45, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=12, bias=True)
)
Critic MLP: Sequential(
  (0): Linear(in_features=45, out_features=512, bias=True)
  (1): ELU(alpha=1.0)
  (2): Linear(in_features=512, out_features=256, bias=True)
  (3): ELU(alpha=1.0)
  (4): Linear(in_features=256, out_features=128, bias=True)
  (5): ELU(alpha=1.0)
  (6): Linear(in_features=128, out_features=1, bias=True)
)
Traceback (most recent call last):
  File "/home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_eval.py", line 46, in <module>
    main()
  File "/home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_eval.py", line 42, in main
    obs, rews, dones, infos = env.step(actions)
                              ^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/Genesis/examples/locomotion/go2_env.py", line 126, in step
    self.scene.step()
  File "/home/ali/Projects/Genesis/Genesis/genesis/utils/misc.py", line 73, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/Genesis/genesis/engine/scene.py", line 727, in step
    self._visualizer.update(force=False, auto=refresh_visualizer)
  File "/home/ali/Projects/Genesis/Genesis/genesis/vis/visualizer.py", line 148, in update
    self._viewer.update(auto_refresh=auto)
  File "/home/ali/Projects/Genesis/Genesis/genesis/vis/viewer.py", line 126, in update
    buffer_updates = self.context.update()
                     ^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/Genesis/genesis/vis/rasterizer_context.py", line 779, in update
    self.update_rigid(buffer_updates)
  File "/home/ali/Projects/Genesis/Genesis/genesis/vis/rasterizer_context.py", line 309, in update_rigid
    buffer_updates[self._scene.get_buffer_id(self.rigid_nodes[geom.uid], "model")] = geom_T.transpose(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/Genesis/genesis/ext/pyrender/scene.py", line 489, in get_buffer_id
    return primitive.get_buffer_id(buffer_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ali/Projects/Genesis/Genesis/genesis/ext/pyrender/primitive.py", line 310, in get_buffer_id
    raise ValueError(f"Buffer {buffer_name} does not exist")
ValueError: Buffer model does not exist
[Genesis] [14:20:33] [INFO] 💤 Exiting Genesis and caching compiled kernels...
(.venv) ali@TheTower:~/Projects/Genesis$ 
```

### Steps to Reproduce

Occurs when running any script where show_viewer=True, such as in go2_eval.py

```
import argparse
import os
import pickle

import torch
from go2_env import Go2Env
from rsl_rl.runners import OnPolicyRunner

import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-e", "--exp_name", type=str, default="go2-walking")
    parser.add_argument("--ckpt", type=int, default=100)
    args = parser.parse_args()

    gs.init()

    log_dir = f"logs/{args.exp_name}"
    env_cfg, obs_cfg, reward_cfg, command_cfg, train_cfg = pickle.load(open(f"logs/{args.exp_name}/cfgs.pkl", "rb"))
    reward_cfg["reward_scales"] = {}

    env = Go2Env(
        num_envs=1,
        env_cfg=env_cfg,
        obs_cfg=obs_cfg,
        reward_cfg=reward_cfg,
        command_cfg=command_cfg,
        show_viewer=True,
    )

    runner = OnPolicyRunner(env, train_cfg, log_dir, device="cuda:0")
    resume_path = os.path.join(log_dir, f"model_{args.ckpt}.pt")
    runner.load(resume_path)
    policy = runner.get_inference_policy(device="cuda:0")

    obs, _ = env.reset()
    with torch.no_grad():
        while True:
            actions = policy(obs)
            obs, rews, dones, infos = env.step(actions)

if __name__ == "__main__":
    main()

"""
# evaluation
python examples/locomotion/go2_eval.py -e go2-walking -v --ckpt 100
"""
```

### Expected Behavior

The Viewer Should not close out after a milisecond of being on screen. The error message about `ValueError: Buffer model does not exist` should not appear. This is an issue that does NOT occur when i run Genesis on my Macbook M1 Pro. Only on my linux machine. 

### Relevant log output

```shell

```

### Environment

- OS: Linux 6.11.0-21-generic #21-Ubuntu SMP PREEMPT_DYNAMIC x86_64 GNU/Linux
- Internal AMD Ryzen Zen5 9600x iGPU for Mobo-to-Monitor HDMI connection

- Discrete Nvidia GPUs: 
* Nvidia GeForce 3050 6gb
* Nvidia RTX A400 4gb
- NVIDIA-SMI 565.77
* Driver Version: 565.77
* CUDA Version: 12.7

### Release version or Commit ID

Commit Version 292df9f (and earlier)

Installed in Developer Mode with:
```
git clone https://github.com/Genesis-Embodied-AI/Genesis.git
cd Genesis
pip install -e ".[dev]"
```

### Additional Context

This is a confirmation that the error first logged [in this issue](https://github.com/Genesis-Embodied-AI/Genesis/issues/861) still persists. I have tried this under nvidia drivers 572.xx and 565.xx, with the issue persisting.

Kashu7100, 2025-04-04:
@bic4907 @ziyanx02 Could you check?

duburcqa, 2025-04-04:
This is a rendering issue. Very difficult to reproduce without direct access to the machine unfortunately... It has been reported multiple times but I don't know how to address it.

S-Ali-Zaidi, 2025-04-04:
> This is a rendering issue. Very difficult to reproduce without direct access to the machine unfortunately... It has been reported multiple times but I don't know how to address it.

Are there any steps you'd like me to take to collect more data on my system and how Genesis is interfacing with it? Any tests you would like me to run?

S-Ali-Zaidi, 2025-04-04:
> Are there any steps you'd like me to take to collect more data on my system and how Genesis is interfacing with it? Any tests you would like me to run?

Also, I accidentally closed the issue with my last comment. If someone could re-open it, that would be appreciated! 

@duburcqa @Kashu7100

S-Ali-Zaidi, 2025-04-06:
Hello, @duburcqa @Kashu7100 @bic4907 @ziyanx02:

Please note that I've found a way to get the `show_viewer=True` renderer working on my machine, so long as I run a script with the command `PYOPENGL_PLATFORM=glx` prior to running it. 

This workaround was discovered by first running this script:

```python
import time
import traceback

import genesis as gs

# Try to import the Primitive class from the pyrender module.
# Adjust the import if your package structure is different.
try:
    from genesis.ext import pyrender
    import genesis.ext.pyrender.primitive as prim
except Exception as e:
    print("Could not import pyrender primitives. Ensure your Genesis package is installed in developer mode.", e)
    exit(1)

# Monkey-patch the get_buffer_id method to print debug info when "model" is missing.
original_get_buffer_id = prim.Primitive.get_buffer_id

def debug_get_buffer_id(self, buffer_name):
    if buffer_name not in self._buffers:
        print("[DEBUG] Buffer", buffer_name, "is missing for primitive with uid:",
              getattr(self, "uid", "unknown"))
        print("[DEBUG] Available buffers:", list(self._buffers.keys()))
    return original_get_buffer_id(self, buffer_name)

prim.Primitive.get_buffer_id = debug_get_buffer_id

# Initialize Genesis with your preferred settings.
gs.init(seed=0, precision="32", logging_level="debug", backend=gs.cuda)

# Create a minimal scene with the viewer enabled.
scene = gs.Scene(
    show_viewer=True,
    viewer_options=gs.options.ViewerOptions(
        res=(640, 480),
        camera_pos=(3.5, 0.0, 2.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=30,
        max_FPS=25,
    ),
    # Use Rasterizer or RayTracer as preferred.
    renderer=gs.renderers.Rasterizer(),
)

# Add a simple plane entity.
plane = scene.add_entity(gs.morphs.Plane())

# Build the scene (this compiles kernels, initializes the viewer, and registers buffers).
scene.build()

print("Scene built. Entering simulation loop to trigger viewer updates...")
print("If a missing 'model' buffer is encountered, debug info will be printed.")

# Run a simulation loop.
try:
    while True:
        scene.step()
        time.sleep(0.1)
except Exception as ex:
    print("Error during scene.step():", ex)
    traceback.print_exc()
```

The results of which output the following when ran WITHOUT `PYOPENGL_PLATFORM=glx`:
[... 160 lines cut ...]
OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.10.1

=== OpenGL Renderer (glxinfo) ===
OpenGL renderer string: AMD Radeon Graphics (radeonsi, raphael_mendocino, LLVM 19.1.1, DRM 3.59, 6.11.0-21-generic)

=== Xorg Version ===
Xorg version not available

=== Python Version ===
Python 3.12.7

=== PyOpenGL Version (pip freeze) ===
PyOpenGL==3.1.9

=== Taichi Version ===
[Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.12.7
Taichi version: (1, 7, 3)

=== Genesis Version ===
Genesis version: 0.2.1

=== PYOPENGL_PLATFORM Environment Variable ===
PYOPENGL_PLATFORM=
(.venv) ali@TheTower:~/Projects/Genesis$ 
```

duburcqa, 2025-04-08:
This is very interesting. It already trying using PYOPENGL_PLATFORM='glx' after the other 2 options failed, but then you get a segfault instead of a valid OpenGL context. While if you enforce this option right from the start it works well. It seems that we are not destroying to old context properly before trying another one, which may get us into trouble...

duburcqa, 2025-04-08:
@S-Ali-Zaidi Please could you try installing this branch to see if it fixes your issue? Thank you!
```bash
pip install git+https://github.com/duburcqa/Genesis.git@viewer_robust
```

closed 2025-04-09 by duburcqa