Genesis-Embodied-AI/genesis-world #978: [Bug]: Black screen in visualization window when Franka robotic arm example is running

LiRWZ, 2025-04-08:
### Bug Description

I'm using docker to run the HELLO_GENESIS routine (loading the Franka robotic arm and letting it fall freely), and I can pop up the visualization window normally, and the terminal is outputting debugging information, but the visualization window doesn't show anything.

### Steps to Reproduce

import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()

### Expected Behavior

The visualization window shows something

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 20.04.01]
- GPU/CPU [11th Gen Intel(R) Core(TM) i9-11900K @ 3.50GHz}
- GPU-driver version (550.144.03)
- CUDA / CUDA-toolkit version (CUDA 12.4)

### Release version or Commit ID

version number: 0.2.1

LiRWZ, 2025-04-14:
> This issue has been fixed on Genesis main branch a while back but there is no new release available on PyPI for now.
> 
> Try installing the latest version of Genesis. You can follow Genesis README instructions for this.
> 
> Ping me of this issue if it is still failing after this. Closing for now.

I've tried installing the main branch, but the problem persists.

LiRWZ, 2025-04-14:
sorry but i don't know how to initialize genesis with "Debug" logging level and where the logging file is?

LiRWZ, 2025-04-14:
> Hum... ok strange. Other graphical apps are working? Can you share the logging messages after initialising genesis with "Debug" logging level?
 there it is:
[Genesis] [07:32:10] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [07:32:10] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [07:32:10] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [07:32:10] [INFO] Running on [NVIDIA GeForce RTX 4090] with backend gs.cuda. Device memory: 23.63 GB.
[Genesis] [07:32:10] [DEBUG] [Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.10.12
[Genesis] [07:32:10] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [07:32:10] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [07:32:11] [INFO] Scene <ab14c39> created.
[Genesis] [07:32:11] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <e043d32>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [07:32:11] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 0.
[Genesis] [07:32:11] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <00783fc>, morph: <gs.morphs.MJCF(file='/root/dockerShared/packages/Genesis-main/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [07:32:11] [WARNING] (MJCF) Tendon not supported
[Genesis] [07:32:11] [DEBUG] (MJCF) No actuator found for joint `world`
[Genesis] [07:32:11] [DEBUG] (MJCF) No actuator found for joint `link0`
[Genesis] [07:32:11] [DEBUG] (MJCF) No actuator found for joint `hand`
[Genesis] [07:32:11] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint1`
[Genesis] [07:32:11] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias.
[Genesis] [07:32:11] [WARNING] (MJCF) Approximating tendon by joint actuator for `finger_joint2`
[Genesis] [07:32:11] [WARNING] (MJCF) Actuator control gain and bias parameters cannot be reduced to a unique PD control position gain. Using max between gain and bias.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 1.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 2.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 3.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 4.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 5.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 6.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 7.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 8.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 9.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 10.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 11.
[Genesis] [07:32:12] [INFO] Convex hull is not accurate enough for collision detection (0.789). Falling back to more expensive convex decomposition (see FileMorph options).
[Genesis] [07:32:12] [DEBUG] Convex decomposition file (.cvx) found in cache.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 12.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 13.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 14.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 15.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 16.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 17.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 18.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 19.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 20.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 21.
[Genesis] [07:32:12] [INFO] Convex hull is not accurate enough for collision detection (0.789). Falling back to more expensive convex decomposition (see FileMorph options).
[Genesis] [07:32:12] [DEBUG] Convex decomposition file (.cvx) found in cache.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 22.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 23.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 24.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 25.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 26.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 27.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 28.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 29.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 30.
[Genesis] [07:32:12] [DEBUG] Preprocessed file (`.gsd`) found in cache for geom idx 31.
[... cut ...]
[Genesis] [07:32:13] [WARNING] Constraint solver time constant was increased to avoid numerical instability (from `0.005` to `0.02`). Decrease simulation timestep to avoid altering the original value.
[W 04/14/25 07:32:14.979 4145] [frontend_ir.cpp:begin_frontend_struct_for_on_external_tensor@1694] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [07:32:17] [INFO] Compiling simulation kernels...
[Genesis] [07:32:29] [INFO] Building visualizer...
[Genesis] [07:32:30] [DEBUG] Trying to create OpenGL Context for PYOPENGL_PLATFORM='native'...
[Genesis] [07:32:35] [INFO] Viewer created. Resolution: 960×720, max_FPS: 60.

duburcqa, 2025-04-14:
What about other graphical app such as Firefox ? Does it report that webgl hardware acceleration is working?

Try `export PYOPENGL_PLATFORM='glx'` or `export PYOPENGL_PLATFORM='egl'`. Maybe it will help.

duburcqa, 2025-04-20:
@LiRWZ Any update on your side?

trns1997, 2025-04-30:
hey @duburcqa @LiRWZ, 

[image]

I just faced a similar issue if not the same. I am running Genesis in a docker container via WSL 2 on windows and the visualization window is black. I conducted a simple gui test with `glxgears` and the display is forwarded correctly, so there is another underlying issue. Note I built the docker image using the main branch. Lemme know how i can help.

trns1997, 2025-05-07:
hey again,
I found this old issue https://github.com/Genesis-Embodied-AI/Genesis/issues/102 which is linked to the issue i mentioned

I basically did the following:
```
docker run --gpus all --rm -it -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix -v $PWD:/workspace -v /usr/lib/wsl:/usr/lib/wsl genesis
export LD_LIBRARY_PATH=/usr/lib/wsl/lib:LD_LIBRARY_PATH
python examples/tutorials/hello_genesis.py
```

This has nothing to do with @LiRWZ  issue. @LiRWZ there were suggestions that reducing the max FPS value might help.

duburcqa, 2025-05-07:
Just to clarify, is it working as expected with the update of the environment variable `LD_LIBRARY_PATH` ? If so, this should definitely be added to the troubleshooting documentation of Genesis !

trns1997, 2025-05-08:
@duburcqa yes the volume mount from wsl and adding the path to `LD_LIBRARY_PATH` resolves the issue. I'll create an PR in the doc repo adding this subtilty for users using genesis in a docker on wsl2.

closed 2025-05-11 by duburcqa