Genesis-Embodied-AI/genesis-world #979: [Feature]: How to use forward kinematics for robotic arms

QYS01, 2025-04-08:
### What feature or enhancement are you proposing?

I am using a Panda robotic arm and want to input the angles of each joint to obtain the end effector pose. Is there a related function for this? I only saw the inverse kinematics function

### Motivation

I am using a Panda robotic arm and want to input the angles of each joint to obtain the end effector pose. Is there a related function for this? I only saw the inverse kinematics function

### Potential Benefit

I am using a Panda robotic arm and want to input the angles of each joint to obtain the end effector pose. Is there a related function for this? I only saw the inverse kinematics function

### What is the expected outcome of the implementation work?

Find the forward kinematics function

Kashu7100, 2025-04-08:
There is `forward_kinematics()` in RigidEntity

duburcqa, 2025-04-09:
Just to make sure, you may without any simulation running?

QYS01, 2025-04-10:
Thank you, the local code hasn't been updated for 2 months, so we couldn't find the forward kinematics function. After updating to the latest code in the repository, I have found the function（forward_kinematics()） you mentioned.