Genesis-Embodied-AI/genesis-world #980: [Bug]: After replacing the Genesis with the new source code, a problem of circular import occurred in the Python code

JoXIA, 2025-04-08:
### Bug Description

After replacing the Genesis with the new source code from the official website, a problem of circular import occurred in the Python code. If the official website source code is not used to overwrite, although hellogenesis.py still cannot run on Windows, running parallel simulations is possible. Once the genesis folder is overwritten with the official website source code, even running parallel simulations will result in a circular import.

 I guess that is because the code of genesis which will run in windows is still not stable until now. So genesis official website may need to update these code to solve the circle import problems.

### Steps to Reproduce

replay.py
```python

import numpy as np

from . import _replay
from .utils import ascontiguous

def _map_isolated_points(points, edges, triangles, return_outliers=False):
    r"""Map the isolated points to the triangles.

    (points, edges, triangles) represents a structure. The goal of this function
    is to compute a mapping array such that the points that are not in the triangles
    but are in the edges are merged into the points that are in the triangles, with
    respect to the edges. An example is given below.

          (1)
         / | \\
      (0)  |  (2)-3
         \ | /  \\
          (4)    6-9
           |
           5     8-7

    In this example, the points 5, 3, 4, 7, 8, 9 are not connected to any triangle.
    The expected mapping is:

    0 -> 0
    1 -> 1
    2 -> 2
    3 -> 2
    4 -> 4
    5 -> 4
    6 -> 2
    7 -> 7 (7 cannot be merged into any point in the triangles)
    8 -> 8 (8 cannot be merged into any point in the triangles)
    9 -> 2

    The output will be the mapping array and the merged points array. In this example,
    the mapping array is [0, 1, 2, 2, 4, 4, 2, 7, 8, 2] and the merged points array is
    [3, 5, 6, 9]. The points 7 and 8 are outliers. If return_outliers is True,
    the function will return the mapping array, the merged points array and the
    isolated points array. Else, the function will return the mapping array and the
    merged points array.

    Parameters
    ----------
        points : sequence
            array of points
        edges : sequence
            array of edges
        triangles : sequence
            array of triangles
        return_outsider : bool
            if True, return the outliers

    Returns
    -------
        np.ndarray
            mapping array
        np.ndarray
            merged points array
[... 188 lines cut ...]
    from ..rigid_entity import RigidEntity
  File "D:\A_New_software_anaconda\envs\genesisL\Lib\site-packages\genesis\engine\entities\rigid_entity\__init__.py", line 1, in <module>
    from .rigid_entity import RigidEntity
  File "D:\A_New_software_anaconda\envs\genesisL\Lib\site-packages\genesis\engine\entities\rigid_entity\rigid_entity.py", line 12, in <module>
    from genesis.utils import mesh as mu
  File "D:\A_New_software_anaconda\envs\genesisL\Lib\site-packages\genesis\utils\mesh.py", line 18, in <module>
    from genesis.ext import fast_simplification
  File "D:\A_New_software_anaconda\envs\genesisL\Lib\site-packages\genesis\ext\fast_simplification\__init__.py", line 2, in <module>
    from .replay import _map_isolated_points, replay_simplification  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\A_New_software_anaconda\envs\genesisL\Lib\site-packages\genesis\ext\fast_simplification\replay.py", line 3, in <module>
    from . import _replay
ImportError: cannot import name '_replay' from partially initialized module 'genesis.ext.fast_simplification' (most likely due to a circular import) (D:\A_New_software_anaconda\envs\genesisL\Lib\site-packages\genesis\ext\fast_simplification\__init__.py)
```

### Environment

- OS: Windows 11
- GPU/CPU： Nvidia RTX 4060
- GPU-driver： Geforce Experience 3.27.0.120
- CUDA / CUDA-toolkit：CUDA 12.4

### Release version or Commit ID

genesis-world   0.2.1

JoXIA, 2025-04-08:
Based on the results of attempts by fellow enthusiasts around me, it is highly likely that using the source code of an older version can solve this problem. The source code of the older version is not directly available on the official website at present. You can visit the guide below, where the source code of the older version is provided at the end of the article. Of course, we still hope that the official website can fix the problem of the new version as soon as possible. Wish Genesis a smoother progress.

https://blog.csdn.net/xiangshangdemayi/article/details/145285887?ops_request_misc=&request_id=&biz_id=102&utm_term=windows%20genesis&utm_medium=distribute.pc_search_result.none-task-blog-2~all~sobaiduweb~default-0-145285887.142^v102^pc_search_result_base7&spm=1018.2226.3001.4187

duburcqa, 2025-04-09:
`hellogenesis.py` runs fine on Windows on main branch. Do you have a script that I could use to check parallel simulation?

duburcqa, 2025-04-09:
I'm waiting for more information on your side, but at this point there is nothing I can do to help you :/

duburcqa, 2025-04-09:
I will re-open this issue if you have any additional information to share. I suggest to completely uninstall genesis via `pip`, and ideally start again with a fresh venv.

closed 2025-04-09 by duburcqa

ifueko, 2025-05-07:
I have the same issue on a ubuntu 22.04 headless server. do you have any suggestions to fix the issue? a fresh install did not work, and all installation steps were successful.