Genesis-Embodied-AI/genesis-world #986: [Bug]: pip installed the 'UNKNOWN' package. pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git

devemin, 2025-04-09:
### Bug Description

Hello.

I did pip install from git+ URL of repository.
But It made installing 'UNKNOWN' package. and no attribute gs.init() .

I tried to change the commit number. But same.

`
pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git
`

[image]

Is this depend on my environment? or bug?

Thanks.
devemin

### Steps to Reproduce

import genesis as gs
from itertools import chain

gs.init(backend=gs.cpu)

### Expected Behavior

masa@L4THOST2204:~/Genesis/examples/tutorials$ python3 myhello.py 
Traceback (most recent call last):
  File "/home/masa/Genesis/examples/tutorials/myhello.py", line 6, in <module>
    gs.init(backend=gs.cpu)
AttributeError: module 'genesis' has no attribute 'init'

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 22.04,
- GPU/CPU Core i7 13700F
- GPU-driver version | NVIDIA-SMI 555.58.02              Driver Version: 555.58.02      CUDA Version: 12.5     |

### Release version or Commit ID

today recently
ca0c604627f71795a7b80b4c31ed411faa16410c

duburcqa, 2025-04-09:
Weird. let me have a look, this is surprising.

duburcqa, 2025-04-09:
I cannot reproduce the issue. On my machine with freshly spawned ubuntu container it works just fine.

duburcqa, 2025-04-09:
Actually I don't even have the line `Build wheels for collected packages: [...]`. I tried with Python 3.12. Let me check with 3.10.

devemin, 2025-04-09:
@duburcqa 
Many thanks!

I will try it with fresh Ubuntu 22.04 environment tomorrow.

duburcqa, 2025-04-09:
No issue with python 3.10 either :/

devemin, 2025-04-09:
@duburcqa 

Woooooow!!  It succeed!
`
pip install --upgrade pip
`

I am enjoying Genesis very much in this year. Next I would like to try various sim2real with Go1/Go2 :-)
Thanks for your support!

Ref to my jumping dog post:
https://x.com/devemin/status/1905293253903315310

<br>

<br>

[image]

[image]

duburcqa, 2025-04-09:
Glad to hear it fixed it for you and you are enjoying Genesis :) Great robot policy! Looking forward to see how far you can push the simulator. Keep in mind that main branch may be unstable. If you detect any issue, do not hesitate to open an issue, it is very helpful.