Genesis-Embodied-AI/genesis-world #991: [Bug]: No decomposition of mesh for single link URDF in convexify

jgleyze, 2025-04-09:
### Bug Description

Hello;

I have a small problem loading the urdf of a storage box, from what I understand in #950 URDFs can now be decomposed in mulitple convex parts. However in my case (a urdf containing a single object) coACD doesn't seem to be launched and the returned collision mesh is not at all fitting the original collision mesh even though i set decompose_error_threshold to 0; which from what i understand forces the decomposition.

Here is a photo of my original box :

[image]

and the result convex hull : 

[image]

### Steps to Reproduce

Here is a small git containing everything you need to recreate the problem :
https://github.com/jgleyze/gs_convex_example/tree/master

### Expected Behavior

I'd like the urdf object to be decomposed into convexed subparts

### Relevant log output

```shell

```

### Environment

N/A

### Release version or Commit ID

77d339de69425397da7376bc731384c1fe21cf29

duburcqa, 2025-04-10:
Can you provide your terminal output with debug logging ideally ?

jgleyze, 2025-04-10:
This is my output in debug mode after I ran a gs clean : 

[Genesis] [10:30:50] [INFO] Beware running Genesis in debug mode dramatically reduces runtime speed.
[Genesis] [10:30:51] [INFO] Running on [Intel(R) Xeon(R) Silver 4214 CPU @ 2.20GHz] with backend gs.cpu. Device memory: 93.05 GB.
[Genesis] [10:30:51] [DEBUG] [Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.10.16
[Genesis] [10:30:51] [DEBUG] [Taichi] Starting on arch=x64
[Genesis] [10:30:51] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: True, 🎨 theme: 'dark'.
[Genesis] [10:30:52] [INFO] Scene <d88f7c7> created.
[Genesis] [10:30:52] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <736766e>, morph: <gs.morphs.URDF(file='/home/gleyze/gs_scene_generator/gs_convex_example/blue_box/model.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [10:30:52] [INFO] Preprocessing geom idx 0.
[Genesis] [10:30:52] [INFO] Building scene <d88f7c7>...
[W 04/10/25 10:30:54.545 913944] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 04/10/25 10:30:54.978 913944] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 04/10/25 10:30:56.881 913944] [frontend_ir.cpp:begin_frontend_struct_for_on_external_tensor@1694] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 04/10/25 10:31:09.039 913944] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [10:31:09] [INFO] Compiling simulation kernels...
[Genesis] [10:32:39] [INFO] Building visualizer...
[Genesis] [10:32:40] [DEBUG] Trying to create OpenGL Context for PYOPENGL_PLATFORM='native'...
[Genesis] [10:32:41] [INFO] Viewer created. Resolution: 720×540, max_FPS: 60.
[Genesis] [10:32:41] [INFO] 💤 Exiting Genesis and caching compiled kernels...

duburcqa, 2025-04-10:
Perfect! Let me check

duburcqa, 2025-04-10:
I can reproduce the issue. I will fix it ASAP.

jgleyze, 2025-04-10:
Thank you very much for the fast fix !