Genesis-Embodied-AI/genesis-world #992: [Feature]:

Richal13Yu, 2025-04-10:
### What feature or enhancement are you proposing?

The ability to dynamically update or modify meshes after they have been imported into Genesis, such as altering vertex positions, adding or removing faces, and changing overall geometry.

### Motivation

As a robotics researcher working on surgical simulation and real-time interaction scenarios,
I want to dynamically modify the geometry of imported meshes during simulation runtime,
Because this capability is essential for realistic simulations involving deformable tissues and environment modifications.

### Potential Benefit

Implementing dynamic mesh modification would significantly enhance simulation realism and interactivity. This feature is particularly beneficial in surgical training, interactive robotics research, and scenarios involving real-time environmental changes.

### What is the expected outcome of the implementation work?

Provide runtime API access to modify vertex positions of imported meshes.

 Enable the addition and removal of faces dynamically.

 Allow real-time geometric updates without needing to re-import entire mesh files.

duburcqa, 2025-04-10:
Actually it is already possible to modify the geometry at. runtime has long as the number of vertices does not change. Do you have any high-API in mind, or some pseudo-code of a script? It would be easier for me to understand what you need exactly.

duburcqa, 2025-04-10:
Could you provide a complete self-contained example with all the assets to check on my side.

There will be some issues related to collision detection (pre-computed AABB bounding boxes, and support function for MPR algorithm). Besides, visual geometries is NOT the same as the collision geometries. `vverts_info` is for visual.

Richal13Yu, 2025-04-10:
> Could you provide a complete self-contained example with all the assets to check on my side.
>
> There will be some issues related to collision detection (pre-computed AABB bounding boxes, and support [...]

Sure, thanks for the clarification. I’ve attached a compressed file containing the STL mesh and a self-contained Python example demonstrating the runtime deformation. Let me know if there’s anything else you need to test on your side!

```
import genesis as gs
import numpy as np
import time

gs.init(backend=gs.gpu)

dt = 3e-3

scene = gs.Scene(
    show_viewer=True,
    viewer_options=gs.options.ViewerOptions(
        res=(1280, 960),
        # camera_pos=(0.0, 0.0, 0.0),
        # camera_lookat=(0, 0.001, 4.60),
        camera_fov=50,
        max_FPS=60,
    ),
    vis_options=gs.options.VisOptions(
        show_world_frame=True,  # visualize the coordinate frame of `world` at its origin
        world_frame_size=2.0,  # length of the world frame in meter
        show_link_frame=False,  # do not visualize coordinate frames of entity links
        show_cameras=False,  # do not visualize mesh and frustum of the cameras added
        plane_reflection=True,  # turn on plane reflection
        ambient_light=(0.1, 0.1, 0.1),  # ambient light setting
        lights=[
            # First directional light
            {
                "type": "directional",
                "dir": (1, 1, 1),
                "color": (1.0, 1.0, 1.0),
                "intensity": 5.0,
            },
            {
                "type": "directional",
                "dir": (-1, -1, -1),
                "color": (1.0, 0.8, 0.8),
                "intensity": 3.0,
            },
        ]
    ),
    renderer=gs.renderers.Rasterizer(),  # using rasterizer for camera rendering
    sim_options=gs.options.SimOptions(
        dt=dt,
        gravity=(0, 0, 0),
    ),
    rigid_options=gs.options.RigidOptions(
        dt=dt,
        gravity=(0, 0, 0),
        enable_collision=False,
        enable_self_collision=False,
    ),
    mpm_options=gs.options.MPMOptions(
        dt=dt,
        lower_bound=(0.0, 0.0, -0.2),
        upper_bound=(1.0, 1.0, 1.0),
        gravity=(0, 0, 0),  # mimic gravity compensation
        enable_CPIC=True,
    ),
)

plane = scene.add_entity(
    gs.morphs.Plane(),
)

sphere_surface = gs.surfaces.Glass(color=(0.86, 0.08, 0.24))

sphere = scene.add_entity(
    gs.morphs.Mesh(file='/home/richal/genesis_code/mesh/sphere.stl',
                   fixed=True,
                   pos=[0,0,0],
                   ),
    surface=sphere_surface,
)

rigid_solver = scene.sim.rigid_solver
verts = rigid_solver.verts_info.init_pos

verts_sphere = sphere.verts_info

scene.build()

scene.step()

def get_scaling_factors(t):
    a = 1 + 0.5 * np.sin(t)
    b = 1 + 0.3 * np.cos(t)
    c = 1 + 0.2 * np.sin(2 * t)
    return a, b, c

def deform_mesh(vertices, t):
    a, b, c = get_scaling_factors(t)
    deformed = np.zeros_like(vertices)
    deformed[:, 0] = vertices[:, 0] / a
    deformed[:, 1] = vertices[:, 1] / b
    deformed[:, 2] = vertices[:, 2] / c
    return deformed

t = 0

while True:

    new_positions = deform_mesh(verts,t)
    sphere.verts_info[:] = new_positions

    t += dt

    scene.step()

```

[sphere.zip](https://github.com/user-attachments/files/19682052/sphere.zip)

duburcqa, 2025-04-10:
I was writing an example script when I realised what you are actually looking may already be implemented, since Genesis features cloth and deformable object physics. Please have a look at `examples/tutorials/advanced_worm.py` or `examples/tutorials/pbd_cloth.py`.

What are you trying to do exactly ? Because changing the geometry of a mesh seems to contract the initial purpose of Rigid Body Dynamics, which is the solver you are currently using.

Richal13Yu, 2025-04-10:
Thanks! I’ll check out the examples you mentioned.

Later on, I may need to modify the mesh surface dynamically at runtime — for example, generating new surfaces when simulating something like a scalpel cutting through soft tissue. In such cases, new faces might be generated during the simulation. I’d also need support for surface stretching, like skin deformation under tension.