Genesis-Embodied-AI/genesis-world #1000: [Bug]: Using the URDF robotic arm model, the robotic arm can move through itself

QYS01, 2025-04-11:
### Bug Description

When I use the URDF model to load the environment and control the movement of the robotic arm, it can pass through itself without colliding with itself. Although I have set collision=True to confirm that the robotic arm can collide with the ground and other objects, it will not collide with itself. This URDF can collide with itself in ROS2 Gazebo.

### Steps to Reproduce

self.robot = self.scene.add_entity(
            gs.morphs.URDF(
                file="~/Genesis/rm_description/urdf/rm_75.urdf",
                fixed=True,
                collision=True,
            ),
        )

### Expected Behavior

The robotic arm will collide with itself

### Screenshots/Videos

https://github.com/user-attachments/assets/78c0482d-b194-42f1-b6ab-af2b60e4e2c2  
The self crossing occurred around the 26th second of the video

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 22.04]

### Release version or Commit ID

current version

duburcqa, 2025-04-11:
You need to manually enable rigid options ‘enable_self_collision’ and ‘enable_adjacent_collision’. In this case, adjacent refers to successive links in the kinematic chain. This is disabled by default as the collision geometries are buggy in many URDF files.

duburcqa, 2025-04-11:
OK so I think it may be time to enable self collision by default!

closed 2025-04-11 by duburcqa