Genesis-Embodied-AI/genesis-world #1015: [Bug]: Extreme Memory Usage When Loading Complex URDF with convexify=False (Up to 200GB RAM)

coghks625, 2025-04-14:
### Bug Description

I'm loading a complex object into the scene using a URDF file. When I set convexify=False, the RAM usage spikes up to 200GB, which eventually causes the system to freeze or crash.

Even though Genesis is initialized with GPU backend:
gs.init(backend=gs.gpu)

…it seems that the mesh processing—especially with convexify=False—still relies heavily on CPU memory, possibly due to the way trimesh handles the geometry under the hood.

Is this expected behavior from the trimesh pipeline?
Are there any recommended workarounds or preprocessing steps for loading high-poly meshes without convexification, but also without such extreme memory demands?

Any advice would be greatly appreciated—thank you!

### Steps to Reproduce

block = scene.add_entity(
    morph=gs.morphs.URDF(
        file="/workspace/genesis/gene/lib/python3.10/site-packages/genesis/assets/urdf/sam/urdf/block.urdf", 
        pos=block_pos,
        euler=(0.0, 0.0, 0.0),
        fixed=True,
        convexify=False,  # Causes memory usage to explode
    )
)

### Expected Behavior

Loading a complex-shaped URDF-based object with convexify=False to perform collision simulation based on its actual geometry.

### Relevant log output

```shell

```

### Environment

- OS: [Ubuntu 22.04]
- GPU/CPU [RTX 3090, Intel I9-9900k] 

### Release version or Commit ID

0.2.1

duburcqa, 2025-04-14:
Are you sure that the issue happens during mesh loading ? Yes mesh processing is expected to run on CPU no matter what… I don’t think we can do anything about it.

Why do you want to disable convexify ? Can you share you URDF for debugging purpose ? Because 200Gb looks like a memory leak and you setup seems perfect to debug it.

coghks625, 2025-04-15:
@duburcqa 

[image]

[image]

Hi, I'm a robotics developer working on generating spray painting motions for industrial robots.
In my application, collision checking is essential to ensure safe and feasible motion generation.

To accurately detect collisions, I need the simulated blocks to match their real-world shapes as closely as possible. For this reason, I disabled convexify, since convexified meshes are too simplified and do not reflect the true geometry of the object.

However, once convexify is disabled, the simulation consumes up to 200GB of RAM, which is far beyond the memory capacity of most desktop PCs.

The first image shows the simplified shape when convexify is enabled.

The second image shows the actual block geometry, which is critical for accurate collision checking.

My URDF setup is very simple — just a single block, nothing special or complex.
Is there any known workaround or best practice to reduce memory usage in this case?

Any advice or insights would be greatly appreciated. Thank you!

<?xml version="1.0"?>
<robot name="sam">
   
    
    <link name="base">
    </link>

     
    <joint name="block_joint" type="fixed">
        <origin rpy="0 0 0" xyz="0 0 0"/>
        <parent link="base"/>
        <child link="block"/>
    </joint>

    <link name="block">
      <visual>
        <geometry>
          <mesh filename="/workspace/Genesis/genesis/assets/urdf/sam/meshes/visual/block.dae" />
        </geometry>
      </visual>
      <collision>
        <geometry>
          <mesh filename="/workspace/Genesis/genesis/assets/urdf/sam/meshes/collision/block.stl" />
        </geometry>
      </collision>
      <inertial>
        <origin rpy="0 0 0" xyz="0 0 0"/>
        <mass value="10"/>
        <inertia ixx="0.002706437" ixy="0.000011205" ixz="-0.000008016" iyy="0.002153592" iyz="0.000141528" izz="0.002712272"/>
      </inertial>
    </link>

</robot>

duburcqa, 2025-04-15:
Let me investigate this.

duburcqa, 2025-04-15:
First, your mesh is not watertight. This is an issue for processing it.

duburcqa, 2025-04-15:
Your mesh is definitely "broken": 

[image]

From what I see, half of the vertices are flipped around.

duburcqa, 2025-04-15:
Once your mesh is fixed I could start investigating the actual issue.

duburcqa, 2025-04-15:
I created a PR to partially address your issue. Here is the final collision geometry. Is it fine for you?

[image]

duburcqa, 2025-04-15:
By the way, using the main branch, it does not take 200GB if `convexify=False`, but rather 2GB. However, mesh processing is never ending. I will investigate the issue.

closed 2025-04-15 by duburcqa

duburcqa, 2025-04-15:
[This PR](https://github.com/Genesis-Embodied-AI/Genesis/pull/1026) should the user to diagnose such issue.

coghks625, 2025-04-16:
Thank you again for helping to partially resolve this issue — I truly appreciate it.

I was aware that the original modeling file might have been damaged during the conversion process, but I didn’t expect that this would lead to such extreme memory usage.

I tested the mesh again using the main branch with the convexify=False option on a high-performance server. The mesh processing took around 40 minutes to an hour, and the viewer did launch successfully. However, during the process, memory usage continued to rise and reached nearly 200GB, with the frame rate hovering around 1–5 FPS.

I plan to replace the mesh with a clean, undamaged version later and run another test.

The attached image shows the model as visualized on the server machine.

[image]

duburcqa, 2025-04-16:
Thank you for testing this. I’m surprised it is taking so much memory. Maybe it is related to the number of votes of your machine, which is likely very high ? 

Did you tried with decimation enabled when convex if is disabled ? On my side it helps tremendously.

coghks625, 2025-04-16:
@duburcqa Yes, I also noticed that memory usage has significantly decreased with the latest version—thank you for that improvement.

However, I have one additional question. In the previous version of Genesis, collisions between links within the same URDF were not detected, possibly due to the convexification method.

With the latest update and the introduction of the CoACD-based decomposition, it seems that links within the same URDF are now treated as separate entities for collision detection. As a result, I can now observe collisions between parts of the same URDF.

This becomes problematic for me because I need to load not only the block but also a painting robot. When self-collision is enabled within the same URDF, the frame rate drops significantly.

Is there a way to disable collision detection between links within the same URDF?
I would like to maintain collision detection between the robot and the environment, but not within the robot itself.

coghks625, 2025-04-16:
```python
scene = gs.Scene(
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(0, -3.5, 2.5),
        camera_lookat=(0.0, 0.0, 0.5),
        camera_fov=30,
        res=(960, 640),
        max_FPS=60,
    ),
    sim_options=gs.options.SimOptions(
        dt=0.05,
    ),
    rigid_options=gs.options.RigidOptions(
        enable_self_collision=False,
        enable_collision=True
    ),
    show_viewer=True,
    show_FPS=False
)

Setting enable_self_collision=False like this successfully disabled self-collision detection within the URDF.
Thank you so much for your time and help! 🙏

duburcqa, 2025-04-16:
> Yes, I also noticed that memory usage has significantly decreased with the latest version—thank you for that improvement.

OK, in that case I'm very surprised it took so long... It takes at most 5min on my laptop with decimation enabled.

> With the latest update and the introduction of the CoACD-based decomposition, it seems that links within the same URDF are now treated as separate entities for collision detection. As a result, I can now observe collisions between parts of the same URDF.

Actually self collision is not enabled by default. But this is completely unrelated to convex decomposition. It was just to align with Mujoco default behaviour. And also because people are expected robots to self-collide more often than not.

duburcqa, 2025-04-16:
Keep in mind that true mesh collision is pretty unreliable and slow. This is true for Genesis but also any other simulator I think. Not being able to rely on convex geometry properties makes the problem extremely challenging to solve mathematically. Convex decomposition is vastly superior whenever it can be used.