Genesis-Embodied-AI/genesis-world #1016: [Feature]: How to add camera(s) to any morph

Pinnh, 2025-04-14:
### What feature or enhancement are you proposing?

The api which function is adding camera(s) to **any morph**, there is only method **add_camera** of the scene

### Motivation

There is no document which describle the method to add cameras to the morphs(box、sphere、drone...) that like the Isaac Sim which add cameras to the morphs by prim paths. The camera is a basic component in drones and vehicles, the autopilot system is heavily rely on the first person view of the visual data (RGB\Infrared)

### Potential Benefit

It's obvious

### What is the expected outcome of the implementation work?

- [ ]

### Additional information

reference:
https://docs.isaacsim.omniverse.nvidia.com/latest/sensors/isaacsim_sensors_camera.html#

AliBehdar, 2025-04-14:
I have same issue

YilingQiao, 2025-04-15:
Here is an example
https://github.com/Genesis-Embodied-AI/Genesis/blob/0d29e4987564106f35a7d17335955882471ef27b/examples/drone/interactive_drone.py#L147

More details can be found in PR #611

Pinnh, 2025-04-15:
If there are more than 2 cameras in a morph, how to code, or if there are two morphs both own one camera?  @YilingQiao 
The scene's camera follow a entity is not a good idea, the camera is more likely a part of morph , not scene.

jgleyze, 2025-04-17:
hello do you think the pr from #618 could help you ? I did the #611 and it was more designed to follow an agent to monitor its training not really something to attach to you robot

Pinnh, 2025-04-18:
I want to attach cameras to the robot, we have different purposes. I propose this issue is aim to mention the problem.

jgleyze, 2025-04-18:
That's why I am poiting you to #618 which does exactly that :)

NicolasKuske, 2025-05-15:
Hi @jgleyze,

That does indeed sound like the solution for @Pinnh and for me as well, as I want to attach a camera to a robot arm. 

Would it be possible to give an example how to use the mount function (for the last joint of the Franka panda arm)?

jgleyze, 2025-05-15:
Hey @NicolasKuske 

Here a quick example, i used [this](https://github.com/Genesis-Embodied-AI/Genesis/pull/618/files), it attaches a camera to the end effector of the panda and show the camera render in the GUI : 
```python
import genesis as gs
import numpy as np

def main():
    
    gs.init()
    
    scene = gs.Scene(show_viewer=True)
    scene.add_entity(gs.morphs.Plane())
    franka = scene.add_entity(gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'))
    
    attached_cam = scene.add_camera(GUI=True)
    
    T = np.eye(4)    
    
    T[:3, 3] = np.array([0.1, 0.0, 0.1])
    
    scene.build()
    
    attached_cam.attach(franka.get_link("hand"), T)
    
    for _ in range(1000):
        scene.step()
        attached_cam.render()
        
if __name__ == "__main__":
    main()
```

You have to tweak `T`which is the offset from the attached link, hope it helps :)

NicolasKuske, 2025-05-15:
That's perfect @jgleyze !

I'll try out some stuff with T. Currently it's structure is rather opaque to me. Maybe you can say one or two more words about it? :)

Thank you a lot already, that's quite important for our project!

duburcqa, 2025-05-15:
T is a so-called "homogeneous transform". It describes the relative pose of the camera wrt link it is attached to. It gathers the rotation matrix `T[:3, :3]` and the translation vector `T[3; :3]`.

closed 2025-05-22 by duburcqa

NicolasKuske, 2025-05-24:
For completeness, this is an example configuration for the T-matrix of the attached cam, to look out of the hand link, forward into the world:

```
T = np.eye(4)    

# Define Rx 180° rotation matrix - rotates around the x-axis of the hand link in the direction away from the robot
Rx_180 = np.array([
    [1,  0,  0],
    [0, -1,  0],
    [0,  0, -1]
])

T[:3, :3] = Rx_180

T[:3, 3] = np.array([0.0, 0.0, 0.08]) # each link has a different wire frame which can be observed in the scene view window (press l)
# for the hand link the z axis goes out upward, and the x axis orthogonal to arm direction outward (foward in x direction)

```

NicolasKuske, 2025-05-24:
Extension for the Tutorial "Inverse Kinematics & Motion Planning Example in Genesis", saving recordings from the point of view of the camera attached to the Franka hand link and in addition from a fixed camera observing the movement of the complete Franka arm.

``` python
import genesis as gs
import numpy as np

gs.init(backend=gs.cpu,
       theme='light'#,
#       #logging_level = 'warning'
       )

scene = gs.Scene(show_viewer=True, show_FPS=False, vis_options = gs.options.VisOptions(show_cameras=True))

# 3. Add static and dynamic entities to the scene
# 3.1 Ground plane at z=0
plane = scene.add_entity(gs.morphs.Plane())

# 3.2 Small cube to grasp: a 4cm^3 box at (0.65, 0, 0.02)
cube = scene.add_entity(
    gs.morphs.Box(
        size=(0.04, 0.04, 0.04),       # x,y,z dimensions in meters
        pos=(0.65, 0.0, 0.02),         # initial position in world coords
    )
)

# 3.3 Franka Panda robot loaded from MJCF file
franka = scene.add_entity(
    gs.morphs.MJCF(
        file='xml/franka_emika_panda/panda.xml'  # Path to MJCF/URDF file
    )
)

##### attach cam #### my thing :)
attached_cam = scene.add_camera(GUI=False)
    
   
T = np.eye(4)    

# Define Rx 180° rotation matrix - rotates around the x-axis in the direction away from the robot
Rx_180 = np.array([
    [1,  0,  0],
    [0, -1,  0],
    [0,  0, -1]
])

T[:3, :3] = Rx_180

T[:3, 3] = np.array([0.05, 0.0, 0.0]) # each link has a different wire frame which can be observed in the scene view window (press l)
# for the hand link the z axis goes out upward, and the x axis orthogonal to arm direction outward (foward in x direction)

##### fixed cam 

cam = scene.add_camera(
    res    = (640, 480),
    pos    = (3.5, 0.0, 3.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = False,
)

# 4. Build the scene (allocates bodies, joints, viewer, etc.)
scene.build()

attached_cam.attach(franka.get_link("hand"), T)

attached_cam.start_recording()

cam.start_recording()

# 5. Identify DOF indices for arm motors (7) and gripper fingers (2)
motors_dof = np.arange(7)       # indices 0..6
fingers_dof = np.arange(7, 9)    # indices 7..8

# 6. Tune PD control gains and force limits for Franka
#    These values yield stable, responsive behavior on the Panda arm.
kp = np.array([4500,4500,3500,3500,2000,2000,2000,100,100])
kv = np.array([ 450, 450, 350, 350, 200, 200, 200, 10, 10])
low = np.array([-87,-87,-87,-87,-12,-12,-12,-100,-100])
high= np.array([ 87, 87, 87, 87, 12, 12, 12, 100, 100])

franka.set_dofs_kp(kp)
franka.set_dofs_kv(kv)
franka.set_dofs_force_range(lower=low, upper=high)

# 7. Retrieve the end-effector link for IK
#    The link name 'hand' matches the MJCF definition of the Panda gripper palm.
end_effector = franka.get_link('hand')

# 8. Move to a pre-grasp pose using IK + motion planning
# 8.1 Solve IK for a pose 25cm above the cube, gripper oriented downwards
q_pre_grasp = franka.inverse_kinematics(
    link=end_effector,
    pos = np.array([0.65, 0.0, 0.25]),   # target position in world coords (m)
    quat= np.array([0.0, 1.0, 0.0, 0.0])  # target orientation as (x,y,z,w) quaternion
)
# Open the gripper fingers slightly
q_pre_grasp[-2:] = 0.04

# 8.2 Plan a collision-free path to the IK solution
#      num_waypoints controls the smoothness & duration (~2s at dt=0.01)
path = franka.plan_path(
    qpos_goal     = q_pre_grasp,
    num_waypoints = 200,
)

# 8.3 Execute the planned trajectory
for waypoint in path:
    attached_cam.render()
    cam.render()
    # PD-controller will apply torques to follow `waypoint` joint targets
    franka.control_dofs_position(waypoint)
    scene.step()
        
    
# Wait a bit for the controller to converge on final pose
for _ in range(100):
    attached_cam.render()
    cam.render()
    scene.step()
  

# 9. Grasp the cube: descend, close fingers, lift
# 9.1 Descend to grasp height (13cm above ground)
q_grasp = franka.inverse_kinematics(
    link=end_effector,
    pos = np.array([0.65, 0.0, 0.13]),
    quat= np.array([0.0, 1.0, 0.0, 0.0])
)
# Keep arm joints fixed, only move the 7 arm motors
franka.control_dofs_position(q_grasp[:-2], motors_dof)
for _ in range(100): 
    attached_cam.render()
    cam.render()
    scene.step()

# 9.2 Close gripper with force control: apply -0.5N on each finger
#      Negative sign to pull inward
actions = np.array([-0.5, -0.5])
franka.control_dofs_force(actions, fingers_dof)
for _ in range(100):
    attached_cam.render()
    cam.render()
    scene.step()

# 9.3 Lift the cube up to 28cm height
desired_lift = np.array([0.65, 0.0, 0.28])
q_lift = franka.inverse_kinematics(
    link=end_effector,
    pos = desired_lift,
    quat= np.array([0.0, 1.0, 0.0, 0.0])
)
# Only move the arm motors to the lift pose
franka.control_dofs_position(q_lift[:-2], motors_dof)
for _ in range(200):
    attached_cam.render()
    cam.render()
    scene.step()

print("✅ Grasp and lift sequence complete")

attached_cam.stop_recording(save_to_filename='grasp_cube_attached_video.mp4', fps=60)

cam.stop_recording(save_to_filename='observe-grasp_cube_attached_video.mp4', fps=60)

```

duburcqa, 2025-05-24:
@NicolasKuske What about opening a PR adding your example ?! It sounds like a nice addition

NicolasKuske, 2025-06-02:
@duburcqa cool, yes, I would love to!

How would that look like? Would I have to fork the repository and then extend the tutorial section in my fork and then request a pull?

duburcqa, 2025-06-02:
Yes !