Genesis-Embodied-AI/genesis-world #1020: [Bug]:

MahiRanka, 2025-04-14:
### Bug Description

I keep trying to run examples/locomotion/go2_train.py. I'm running into the following error -  [Genesis] [13:13:28] [ERROR] Unrecognized attribute: rendered_envs_idx

### Steps to Reproduce

```
import argparse
import os
import pickle
import shutil
from importlib import metadata
# import sys
# print(sys.path)
# import genesis 
# print(genesis.__file__)

try:
    try:
        if metadata.version("rsl-rl"):
            raise ImportError
    except metadata.PackageNotFoundError:
        if metadata.version("rsl-rl-lib") != "2.2.4":
            raise ImportError
except (metadata.PackageNotFoundError, ImportError) as e:
    raise ImportError("Please uninstall 'rsl_rl' and install 'rsl-rl-lib==2.2.4'.") from e
from rsl_rl.runners import OnPolicyRunner

import genesis as gs

from go2_env import Go2Env

def get_train_cfg(exp_name, max_iterations):
    train_cfg_dict = {
        "algorithm": {
            "class_name": "PPO",
            "clip_param": 0.2,
            "desired_kl": 0.01,
            "entropy_coef": 0.01,
            "gamma": 0.99,
            "lam": 0.95,
            "learning_rate": 0.001,
            "max_grad_norm": 1.0,
            "num_learning_epochs": 5,
            "num_mini_batches": 4,
            "schedule": "adaptive",
            "use_clipped_value_loss": True,
            "value_loss_coef": 1.0,
        },
        "init_member_classes": {},
        "policy": {
            "activation": "elu",
            "actor_hidden_dims": [512, 256, 128],
            "critic_hidden_dims": [512, 256, 128],
            "init_noise_std": 1.0,
            "class_name": "ActorCritic",
        },
        "runner": {
            "checkpoint": -1,
            "experiment_name": exp_name,
            "load_run": -1,
            "log_interval": 1,
            "max_iterations": max_iterations,
            "record_interval": -1,
            "resume": False,
            "resume_path": None,
            "run_name": "",
        },
        "runner_class_name": "OnPolicyRunner",
        "num_steps_per_env": 24,
        "save_interval": 100,
        "empirical_normalization": None,
        "seed": 1,
    }

    return train_cfg_dict

def get_cfgs():
    env_cfg = {
        "num_actions": 12,
        # joint/link names
        "default_joint_angles": {  # [rad]
            "FL_hip_joint": 0.0,
            "FR_hip_joint": 0.0,
            "RL_hip_joint": 0.0,
            "RR_hip_joint": 0.0,
            "FL_thigh_joint": 0.8,
            "FR_thigh_joint": 0.8,
            "RL_thigh_joint": 1.0,
            "RR_thigh_joint": 1.0,
            "FL_calf_joint": -1.5,
            "FR_calf_joint": -1.5,
            "RL_calf_joint": -1.5,
            "RR_calf_joint": -1.5,
        },
        "dof_names": [
            "FR_hip_joint",
            "FR_thigh_joint",
            "FR_calf_joint",
            "FL_hip_joint",
            "FL_thigh_joint",
            "FL_calf_joint",
            "RR_hip_joint",
            "RR_thigh_joint",
            "RR_calf_joint",
            "RL_hip_joint",
            "RL_thigh_joint",
            "RL_calf_joint",
        ],
        # PD
        "kp": 20.0,
        "kd": 0.5,
        # termination
        "termination_if_roll_greater_than": 10,  # degree
        "termination_if_pitch_greater_than": 10,
        # base pose
        "base_init_pos": [0.0, 0.0, 0.42],
        "base_init_quat": [1.0, 0.0, 0.0, 0.0],
        "episode_length_s": 20.0,
        "resampling_time_s": 4.0,
        "action_scale": 0.25,
        "simulate_action_latency": True,
        "clip_actions": 100.0,
    }
    obs_cfg = {
        "num_obs": 45,
        "obs_scales": {
            "lin_vel": 2.0,
            "ang_vel": 0.25,
            "dof_pos": 1.0,
            "dof_vel": 0.05,
        },
    }
    reward_cfg = {
        "tracking_sigma": 0.25,
        "base_height_target": 0.3,
        "feet_height_target": 0.075,
        "reward_scales": {
            "tracking_lin_vel": 1.0,
            "tracking_ang_vel": 0.2,
            "lin_vel_z": -1.0,
            "base_height": -50.0,
            "action_rate": -0.005,
            "similar_to_default": -0.1,
        },
    }
    command_cfg = {
        "num_commands": 3,
        "lin_vel_x_range": [0.5, 0.5],
        "lin_vel_y_range": [0, 0],
        "ang_vel_range": [0, 0],
    }

    return env_cfg, obs_cfg, reward_cfg, command_cfg

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-e", "--exp_name", type=str, default="go2-walking")
    parser.add_argument("-B", "--num_envs", type=int, default=4096)
    parser.add_argument("--max_iterations", type=int, default=101)
    args = parser.parse_args()

    gs.init(backend=gs.gpu, logging_level="warning")

    log_dir = f"logs/{args.exp_name}"
    env_cfg, obs_cfg, reward_cfg, command_cfg = get_cfgs()
    train_cfg = get_train_cfg(args.exp_name, args.max_iterations)

    if os.path.exists(log_dir):
        shutil.rmtree(log_dir)
    os.makedirs(log_dir, exist_ok=True)

    pickle.dump(
        [env_cfg, obs_cfg, reward_cfg, command_cfg, train_cfg],
        open(f"{log_dir}/cfgs.pkl", "wb"),
    )

    env = Go2Env(
        num_envs=args.num_envs, env_cfg=env_cfg, obs_cfg=obs_cfg, reward_cfg=reward_cfg, command_cfg=command_cfg
    )

    runner = OnPolicyRunner(env, train_cfg, log_dir, device=gs.device)

    runner.learn(num_learning_iterations=args.max_iterations, init_at_random_ep_len=True)

if __name__ == "__main__":
    main()

"""
# training
python examples/locomotion/go2_train.py
"""

```

### Expected Behavior

I expected to be able to train the RL policy however I'm not sure what's going wrong.

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/mahi/genesis_embodied/examples/locomotion/go2_train.py", line 184, in <module>
    main()
  File "/home/mahi/genesis_embodied/examples/locomotion/go2_train.py", line 174, in main
    env = Go2Env(
  File "/home/mahi/genesis_embodied/examples/locomotion/go2_env.py", line 40, in __init__
    vis_options=gs.options.VisOptions(rendered_envs_idx=list(range(1))),
  File "/home/mahi/.local/lib/python3.10/site-packages/genesis/options/vis.py", line 116, in __init__
    super().__init__(**data)
  File "/home/mahi/.local/lib/python3.10/site-packages/genesis/options/options.py", line 27, in __init__
    gs.raise_exception(f"Unrecognized attribute: {key}")

[Genesis] [13:13:28] [ERROR] Unrecognized attribute: rendered_envs_idx
```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

-

YilingQiao, 2025-04-15:
Can you pull the latest commit, run `pip install -e .`, and try again?

MahiRanka, 2025-04-15:
I always get this particular issue when I run in developer mode. This time I ran into the same issue, It's not able to find the proper directory for some reason. I dont understand why this would happen - 
[Taichi] version 1.7.3, llvm 15.0.4, commit 5ec301be, linux, python 3.10.12
Traceback (most recent call last):
  File "/home/mahi/Genesis/examples/locomotion/go2_train.py", line 180, in <module>
    main()
  File "/home/mahi/Genesis/examples/locomotion/go2_train.py", line 155, in main
    gs.init(logging_level="warning")
AttributeError: module 'genesis' has no attribute 'init'

MahiRanka, 2025-04-15:
Tried it multiple times. Same issue. The only time it works is when I do ```pip install genesis-world```

duburcqa, 2025-04-15:
This is very strange. Never faced this issue. In this case, you should do this instead:
`pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git`

MahiRanka, 2025-04-15:
It's pretty strange. It started working after I did what you recommended. Thank you!

duburcqa, 2025-04-15:
yes, you basically installed the latest version of Genesis via pip, building the wheel yourself from source. Before you were either installing the source directly without any wheel being involved in the process, or the wheels available on PyPI corresponding the "old" version 0.2.1.

closed 2025-04-15 by duburcqa