Genesis-Embodied-AI/genesis-world #1024: [Bug]: Vulkan is only supported on Intel?

gpu-daal, 2025-04-15:
### Bug Description

Not sure if it's my issue, but it seems like device selection logic with Vulkan backend only works if you have Intel XPU. Why is that? What's wrong with regular AMD GPU? PyTorch doesn't support Vulkan AFAIK, but Taichi does, so I'm not sure why Vulkan backend is not supported.

### Steps to Reproduce

import argparse
import numpy as np
import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(backend=gs.vulkan, seed=0, precision="32", logging_level="debug")

if __name__ == "__main__":
    main()

### Expected Behavior

Results in:
[Genesis] [12:19:56] [WARNING] Vulkan support only available on Intel XPU device. Falling back to CPU.
[Genesis] [12:19:56] [INFO] Running on [AMD Ryzen 7 7800X3D 8-Core Processor] with backend gs.cpu. Device memory: 30.46 GB.

Should result in choosing Vulkan regardless of Intel support.

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 24
- GPU/CPU: AMD AMD Radeon RX 7800 XT (RADV NAVI32)
- GPU-driver version: Mesa 24

### Release version or Commit ID

Genesis 0.2.1, commit 91450b2dc70ac5c073a3b78281ebc268cf021b3c

duburcqa, 2025-04-15:
Makes sense! [here](https://github.com/Genesis-Embodied-AI/Genesis/pull/1025) it is! Would you be able to test it?

gpu-daal, 2025-04-15:
That kind of works?

```
[Genesis] [12:44:57] [WARNING] Vulkan support only available on Intel XPU device. Falling back to CPU for torch device.
[Genesis] [12:44:57] [INFO] Running on [AMD Ryzen 7 7800X3D 8-Core Processor] with backend gs.vulkan. Device memory: 30.46 GB.
```

It still selects my CPU with vulkan backend, but from amdgpu_top I can see that my GPU is working when testing on water_wheel.py example (graphics queue mostly even without visualization). The FPS improved with visualization though, so that's good, but it only improved by 10-11 FPS - from 4 FPS on CPU to 15 on GPU.

duburcqa, 2025-04-15:
Yes, I updated the warning that was now misleading.

> so that's good, but it only improved by 10-11 FPS - from 4 FPS on CPU to 15 on GPU.

Yeah... I guess this is disappointing. Do you have AMD ROCm installed ? PyTorch is compatible with it (identified as a CUDA device). See install procedure here: https://pytorch.org/get-started/locally/. Can you check if torch CUDA works for you doing this? If so, then I could try to update the device selection logic to use torch CUDA + Taichi Vulkan and see if it runs faster after that, but I'm not very confident about that...

gpu-daal, 2025-04-15:
@duburcqa I only have AMD GPU, so CUDA Torch is definitely out of options - my plan is to use Vulkan for both simulation and rendering. I'll try to install ROCm though, it's a bit of a pain to work with drivers for it.
Is there a reason CPU-based PyTorch extracts such a heavy toll on performance? water_wheel.py example doesn't have any machine learning algorithms inside, so I'm confused, unless Genesis uses pytorch tensor arrays for everything.

duburcqa, 2025-04-15:
> I only have AMD GPU, so CUDA Torch is definitely out of options

Not it is not. As I said, torch is supporting AMD GPU as if they were CUDA devices as long as ROCm is properly configured.

> Is there a reason CPU-based PyTorch extracts such a heavy toll on performance?

I'm not sure if it is actually the case or not. That is why I said I was not very confident regarding whether or not it would improve performance :/

duburcqa, 2025-04-15:
> If the issue is purely my GPU, then that's pretty annoying, but not critical

Sorry I cannot help you on that. I have no idea what is supposed to be the performance. In [this PR](https://github.com/Genesis-Embodied-AI/Genesis/pull/1005), 35FPS is reported but the GPU being used is not mentioned.

> Unless Genesis already has a working version of differentiable rigid-body solver?

Sorry, I don't know this part of Genesis well enough to answer this question. @YilingQiao ?

closed 2025-04-15 by duburcqa

YilingQiao, 2025-04-16:
Rigib body solver's differentiation is still work in progress. Will hopefully be available this summer.

gpu-daal, 2025-04-22:
@YilingQiao I see. Do you guys already have an idea on how to make rigid-body solver differentiable? I've looked through the code, and it seems like Taichi doesn't allow to differentiate through kernels with mixed for-loops and regular statements. Are you guys splitting the rigid-body solver kernels or changing Taichi source code?

YilingQiao, 2025-04-22:
For example, the kernel for the constraint solver in rigid body simulation cannot use auto-diff.

gpu-daal, 2025-04-22:
I see what you mean, but does it not mean that forward kinematics kernel will probably need some rework? Since Taichi doesn't allow to differentiate through a kernel that has both for-loops and regular statements.

YilingQiao, 2025-04-22:
I remember someone mentioned they fixed this at some point. Might need to double-check.

gpu-daal, 2025-04-22:
Fixed Taichi or kernels? Fixing Taichi would require fixing the compiler I suppose. Also, checking for hibernation also messes up with Taichi's auto-diff afaik.

dwihandi-work, 2025-04-22:
@YilingQiao Thank you for your work. I am also very interested in a differentiable rigid body solver, is there currently no conrete timeline on when this will be available? I am also willing to contribute if required :)

gpu-daal, 2025-08-25:
@YilingQiao Hi, I saw you guys made lots of significant improvements to the project, as well as Taichi itself, looks cool! I was wondering if there is any update on differentiable rigid-body solver?