Genesis-Embodied-AI/genesis-world #1027: Dependency conflict on Linux and Windows

LeoneCosti96, 2025-04-16:
### Bug Description

Upon running the installation command pip install genesis-world (both windows and linux), I get:

ERROR: Cannot install genesis-world==0.1.1, genesis-world==0.2.0 and genesis-world==0.2.1 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.1 depends on taichi==1.7.3
    genesis-world 0.2.0 depends on taichi==1.7.2
    genesis-world 0.1.1 depends on taichi==1.7.2

What should I run instead?

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

correct installation

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

[image]

duburcqa, 2025-04-16:
This is strange. Can you install `taichi==1.7.3` manually first ?

LeoneCosti96, 2025-04-16:
yes, that fixed it, thank you

closed 2025-04-16 by duburcqa