Genesis-Embodied-AI/genesis-world #1028: [Feature]: Improve Trackball Rotation with Azimuth/Elevation Control

acrlw, 2025-04-16:
### What feature or enhancement are you proposing?

I propose modifying the rotation logic to use an Azimuth/Elevation system, similar to MuJoCo or other standard 3D viewers.

### Motivation

The current `Trackball` rotation (`STATE_ROTATE`) can feel a bit unintuitive when trying to orbit the scene.

### Potential Benefit

1. More intuitive and predictable camera rotation.
2. Significantly easier to align the camera for standard orthogonal views (front, side, top).

### What is the expected outcome of the implementation work?

Before:

[image]

After:

[image]

duburcqa, 2025-04-16:
That would be amazing ! The current trackball implementation is a crap! Just do it already x)

duburcqa, 2025-04-20:
@acrlw Did you implement this? Could you share the code here? I could take care of the rest if you want!

acrlw, 2025-04-20:
Thank you! I'm trying to make as few changes as possible to the original code. I'll share my code in a few days.

acrlw, 2025-04-20:
> [@acrlw](https://github.com/acrlw) Did you implement this? Could you share the code here? I could take care of the rest if you want!

Thank you! I'm trying to make as few changes as possible to the original code. I'll share my code in a few days.

duburcqa, 2025-04-20:
Amazing! Looking for this 👍

duburcqa, 2025-04-22:
Amazing, I will try it and merge this today if it works! Do you have an email address to add you as contributor of this PR?

acrlw, 2025-04-22:
> Amazing, I will try it and merge this today if it works! Do you have an email address to add you as contributor of this PR?

[acrlwen@shu.edu.cn](mailto:acrlwen@shu.edu.cn)

Thanks ❤️!

closed 2025-04-22 by duburcqa

acrlw, 2025-04-22:
> Merged!

Thanks for testing and merging!