Genesis-Embodied-AI/genesis-world #1030: [Bug]: convex=False not work

RainbowNebula, 2025-04-16:
### Bug Description

When I loaded the .glb file of test tube holder with convex = false, I found that none of the holes on the test tube holder could not be separated during the collision visualization. But in sapien 3, the hole can be passed through normally.

### Steps to Reproduce

[image]

[image]

The glb file: https://pan.baidu.com/s/1k6dAQoqI4MFL2orQIoh3GQ?pwd=tcs5

### Expected Behavior

I hope this bug can be fixed as soon as possible， thank you for your help.

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

v1

RainbowNebula, 2025-04-17:
> Can you upload the file on GitHub directly? I don't have any Baidu account.

myglb = scene.add_entity(
        # material=gs.materials.Rigid(),
        morph=gs.morphs.Mesh(
            file="/home/haichao/Pictures/shiguanjia.glb",
            scale=0.5,
            pos=(0.55, 0.3, 0.05),
            parse_glb_with_trimesh=True,
            # fixed=True,
            #euler=(0, 0, 0),
            euler=(90, 0, 90),
            decimate=False,
            # collision=False,
            fixed=True,
            convexify=False, 
        ),
        surface=gs.surfaces.Default(
            color = (0.8, 0.8, 0.8),
            vis_mode = 'collision',
            # friction=0.5,
        )
    )

Thanks for your reply, “convexify = false" not work, Github doesn't allow me to upload .glb files directly, so I put it in a zip file.

[glb.zip](https://github.com/user-attachments/files/19794550/glb.zip)

duburcqa, 2025-04-20:
I can confirm it is now working on the main branch. Please follow the instructions available on the README to install the latest version (not the pre-compiled wheels available on PyPI).

By the way, here is the result I get with finer-grained convex decomposition than default. 

[image]

If this is fine for you, then it should be better than disabling convexify completely. It should improve physics stability and probably runtime speed:

```python
import genesis as gs

gs.init(backend=gs.cpu, precision="64", logging_level='debug', debug=False)

scene = gs.Scene(show_viewer=True, show_FPS=False)
myglb = scene.add_entity(
        # material=gs.materials.Rigid(),
        morph=gs.morphs.Mesh(
            file="./shiguanjia.glb",
            scale=0.5,
            pos=(0.55, 0.3, 0.05),
            parse_glb_with_trimesh=True,
            # fixed=True,
            #euler=(0, 0, 0),
            euler=(90, 0, 90),
            decimate=False,
            # collision=False,
            fixed=True,
            #convexify=False, 
            coacd_options=gs.options.CoacdOptions(
                threshold=0.02,
            )
        ),
        surface=gs.surfaces.Default(
            color = (0.8, 0.8, 0.8),
            vis_mode = 'collision',
            # friction=0.5,
        )
    )
scene.build()
```